cleanflight/cleanflight #774: What happens with new Cleanflight commits ?

Pierre-A, 2015-04-15:
New commits (I tried with 78a4476506 and 37e551db11) can be compiled as usual with no apparent issue as seen below for 78a4476506.
`..........
obj/main/NAZE/stm32f10x_dbgmcu.o obj/main/NAZE/stm32f10x_dma.o –lm nostartfiles --specs=nano.specs -lc -lnosys -mthumb -mcpu=cortex-m3 -flto -fuse-linker-plugin -Os -ggdb3 -static -Wl,-gc-sections,-Map,./obj/main/cleanflight_NAZE.map -T./src/main/target/stm32_flash_f103_128k.ld
arm-none-eabi-size ./obj/main/cleanflight_NAZE.elf
   text    data     bss     dec     hex filename 
114100     256   13288  127644   1f29c ./obj/main/cleanflight_NAZE.elf arm-none-eabi-objcopy -O ihex --set-start 0x8000000 obj/main/cleanflight_NAZE.elf obj/cleanflight_NAZE.hex

C:\Documents and Settings.....`

Flashing to the Naze is also successfull, but then the board is freezed (initialisation is not completed and green led remains ON) and of course cannot be connected to the GUI.

This happens both with cloned and downloaded Zip versions.
Any macro parameters change in the configuration I was not aware of ?

thenickdude, 2015-04-15:
No problems here building the latest head (37e551db1162daea303abc4b1214358b1ca93ca5) with GCC version "arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]" and flashing it to a Naze32 board.

Did you choose the option to do a full erase?

What compiler version are you using? (run arm-none-eabi-gcc --version). The GCC 4.8 series is used to build official firmwares. The 4.9 series creates builds that don't boot on other boards (though this would be the first time I've seen them fail on the Naze32 if you're using 4.9)

Pierre-A, 2015-04-15:
Strange, as I also use GCC version "arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.4 20140526 (release) [ARM/embedded-4_8-branch revision 211358]".
It used to work up to commit ≠8fcffa1454 (Cleanflight 1.8.0) and I did not change any parameter in that config for compiling/flashing new versions.
Which is the size of the file you flash (mine is 114204 for commit ≠37e551db11 , that is 113948 bytes for text and 256 for data) ? no problem with initialisation of the Naze ?
Probably too obvious for tired head_and_eyes...

thenickdude, 2015-04-15:
File size details:

```
arm-none-eabi-size ./obj/main/cleanflight_NAZE.elf
   text	   data	    bss	    dec	    hex	filename
 114028	    256	  12916	 127200	  1f0e0	./obj/main/cleanflight_NAZE.elf
```

No problems with initialisation and I can connect to the board with the configurator fine. If you aren't doing a "full chip erase" when flashing, maybe flash back to the version that works and do a "dump" on the CLI. Paste the output here. It may be that your configuration isn't compatible with the newer version due to a bug.

Pierre-A, 2015-04-15:
Full chip erase already done.
Before doing anything else, and as we both are with Naze, would you mine exchanging our hex files and see what happens  ?

thenickdude, 2015-04-15:
Sure, I've emailed you with my .hex file. Can you send me your hex file, and please also attach your `obj/main/cleanflight_NAZE.elf` file so I can debug it if it doesn't boot for me, thanks!

tricopterY, 2015-04-15:
Same issue - I run 4.8 2014-q2 and everything used to work fine. When I tried the latest i.e commit ≠37e551db11, I get exactly the same problem as reported by @Pierre-A.

Pierre-A, 2015-04-15:
thnks for that ! but same behaviour with your hex ; board freezed on initialisation.

Diging in .. in fact I thought this was a full Naze32 in the model I work with.. and after verification .. it’s only an acro-Naze32.
So I flashed your hex and mines on a full Naze32 (same red version) and it works quite fine as usual !

The last commit I worked with (and with no issue at all on the acro-Naze32 board) is ≠8fcffa1 (dated march 13th). This makes me think that something changed during the last weeks on the initial recognition of the board configuration (magnetometer and/or baro sensor). 
I did not check the few tens of commits for the last month, do you have any souvenir of that ?

tricopterY, 2015-04-15:
After the above mishap, to get the Naze Acro FC loading the stable version 1.8.1, I had to force bootloader (short the boot pads) otherwise it would not connect and Configurator-firmware flasher would complain with a message: something like 'bootloader not responding'.

hydra, 2015-04-15:
@tricopterY the build would fail if the flash was too big.  the build will also fail if there is not enough 
space to store the config at the end of the flash too.

@Pierre-A  @tricopterY Please use git bisect to find the exact commit that causes the problem and report your findings on this issue

http://git-scm.com/docs/git-bisect

ensure you do a make clean and flash each build and do a full-chip erase each time.

Maybe we can spot a problem, however if upgrading to a later compiler fixes it for you then it's unlikely we'd do a fix.

thenickdude, 2015-04-15:
Pierre suggested that the hex that I built (with the newest available 4.8 compiler) didn't boot on their acro board either.

hydra, 2015-04-15:
Someone please git biscet and we will be able to identify the commit.  I am working on another issue right now.

Pierre-A, 2015-04-15:
Ok I'll try asap, but first I have to open a new PR for hott mode text (maybe tomorrow).

SteveAmor, 2015-04-15:
Issue introduced at this point - c11c25514b4d4b71eb6affc6d7e01cf4014e91f3

tricopterY, 2015-04-15:
@SteveAmor - spot on. After commenting out the following in NAZE-target.h
```C
//#define USE_MPU_DATA_READY_SIGNAL
```
It works ok.

Pierre-A, 2015-04-15:
OK guys, thnks to you
I confirm both observations.

hydra, 2015-04-16:
Can you try this fix:

https://github.com/cleanflight/cleanflight/commit/c11c25514b4d4b71eb6affc6d7e01cf4014e91f3#commitcomment-10748284

hydra, 2015-04-16:
weirdly, i had no issues with the code on my Rev 5 naze32...

SteveAmor, 2015-04-16:
Unfortunately, c11c25514b4d4b71eb6affc6d7e01cf4014e91f3#commitcomment-10748284 didn't work

Pierre-A, 2015-04-16:
**@hydra**, I suppose your Naze32 rev5 is a full one..? 
Mines are both rev5 (full and acro), and the issue is only with the acro one.

hydra, 2015-04-16:
@Pierre-A ok, i will did out an acro board and do some additional testing.  Kinda weird though as they share the same MPU.

SteveAmor, 2015-04-16:
@hydra I only have an acro board which it is failing on.

Pierre-A, 2015-04-16:
**@hydra**, sure, that's why I first suspected an issue related to the initialisation process around baro and magnetometer.
If this can help, looking with osc, the green led is permanently on and not fastly blinking (permanent low signal).

SteveAmor, 2015-04-16:
Just buzzed out my rev5 acro board. Interrupt Pin 12 on the MPU6050 is connected to PC13 (pin 2) on the STM, not PB13 (pin 26). As well as looking at the pin 1 dot on the chip, I verified my chip orientation, checking the TX test point goes to pin 30 on the STM. Which it did.

ledvinap, 2015-04-16:
well ... 
```
#define EXTI15_10_CALLBACK_HANDLER_COUNT 1
```
both baro and MPU callbacks try to get registered, but each clears only own EXTI IRQ flag (EXTI_ClearITPendingBit) . Only one callback is registered, so after other EXTI bit is set, wrong callback gets called indefinitely ...

Pierre-A, 2015-04-16:
re-well..
**ledvinap**, think you got it !
when set to 2
`#define EXTI15_10_CALLBACK_HANDLER_COUNT 2`
(keeping  `#define USE_MPU_DATA_READY_SIGNAL` as it is in Naze target.h)

and no more issue after flashing on Acro-Naze32 rev5.
Any other consequences changing that value ?

hydra, 2015-04-16:
Hah, just fixed this myself in 6d5b44df7af32d194448c57a99f2d0af02d89f27.

hydra, 2015-04-16:
FYI: The Flip32+ Deluxe v5 has an 8mhz oscilator with MPU6050 MPU INT wired up to Pin 26 / PB13

[image]

hydra, 2015-04-16:
I've not got a Naze32 acro handy, closing this issue as I think it's fixed now. If not please re-open and say what board you're using.

closed 2015-04-16 by hydra

SteveAmor, 2015-04-16:
Tested ok on Afroflight rev5 acro only.  

Somthing for the future, I'm convinced the MPU6050 interrupt signal (Pin 12) is connected to Pin 2 / PC13 on the Afroflight rev5 acro only board as I physically tested the continuity.

hydra, 2015-04-16:
@SteveAmor  testing continuity of an unpowered populated board isn't the same as testing a powered populated board with the MPU generating INT signals.

To test the code uncomment https://github.com/cleanflight/cleanflight/blob/6d5b44df7af32d194448c57a99f2d0af02d89f27/src/main/drivers/accgyro_mpu6050.c#L36 compile and flash and check the sensors tab and verify the value of debug 0 hovers around 1000.

hydra, 2015-04-16:
See https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/accgyro_mpu6050.c#L202

SteveAmor, 2015-04-17:
Thanks for the debug info. The interrupt works as expected.

hydra, 2015-04-17:
@SteveAmor thanks for confirmation