cleanflight/cleanflight #775: Proximity detection

nebbian, 2015-04-15:
I've been asked to develop a system that will fly indoors, and lock its position relative to its surroundings.   It will also be possible to set minimum distances so that it will be impossible to crash into a wall.

To this end I've specced out a system that uses 6 proximity sensors (sonar or lidar) that are driven by an external arduino board.  This board will then communicate to the Flight Controller using a serial interface.

I've sketched out the following protocol for the serial communication:

SXRYYYYZ

Where X is the sensor number, from 0 to 9.
YYYY is the range value, from 0 to 9999.  This is unitless.
Z is a carriage return.

An example serial stream is below:
S0R0256
S1R0250
S2R0244
S3R0239
S4R0235
S5R0228

This is for a 6 sensor system (Up, Down, Left, Right, Forward, Backward).  The sample range values are decreasing from 256 down to 228 as the board goes through and reads each sensor.  I have already programmed the arduino sensor board, and it appears to be working properly.

Inside the flight controller, I imagine that there will be CLI variables to set the direction (0 -> 359) and elevation (90 -> -90) of each sensor, allowing the user to configure the system for the sensors that are installed.  

Now for my questions:

1. It appears as if every time a new serial protocol is added, new CLI variables need to be added (x4) for the baud rate.    Wouldn't it be more sensible just to set the baud rate for the serial port?
1. What would be the best feature name for this?  I'm thinking that PROXIMITY_DETECTION might be the best for now.
1. Does this seem like the best approach (with the external daughterboard that communicates via serial protocol)?.  
1. Can you think of other data that should be sent over the serial protocol?
1. File structure: I was thinking of basically replicating the GPS file structure, as this has similar functionality to what I'm after.  Is this the right approach?  Should I keep all the proximity stuff in the same folder, like the blackbox system does?

Any advice will be appreciated.

thenickdude, 2015-04-15:
The ports can be used for multiple different functions that have different customisable baud rates (notably blackbox and MSP) so there need to be multiple baud rate settings per port.

dkisselev, 2015-04-15:
This might be easier to accomplish with an Optical Flow sensor: #529 

That way it's only one sensor, less code, easier to write, etc.

As far as structure, we should put together an internal navigation system. Somewhat like GPS, it accepts position or deviation values from a sensor driver (optical flow, accellerometer, sonar, etc), and pushes out RC commands that correct for those deviations.

tricopterY, 2015-04-18:
Not so sure about Sonar (may be cheap Optical Mouse sensors). IMHO - a better approach would be with LiDAR (Light Detection And Ranging) modules using I2C to CF H/W (No Arduino) eg.
https://www.sparkfun.com/products/13167 (also expensive)
But, apparently there are bugs with I2C and lockups, if trying to read the device too fast.
If going serial with Arduino, how about something like:
AYYYYBYYYYCYYYYDYYYYEYYYYFYYYYZ
or even:
YYYYYYYYYYYYYYYYYYYYYYYYZ
On receive split into a data structure every four bytes, for sensor 0 to 5.

hydra, 2015-05-15:
@nebbian Someone was asking about this in IRC the other day.  Are you still working on it? Do you have a link to your sensor code?

Also it might be better to use I2C for this and have an I2C register that gives distance readings for each sensor.  That would greatly simplify any flight controller code, since no data stream processing and handling code would be needed and would be MUCH simpler.

Driver code aside, the next part would be using the readings in cleanflight itself, i think first getting the alt-hold code to work better would come first, then you could perhaps re-use some of the code for hold in other directions other than avoiding the ground.

nebbian, 2015-05-15:
I'm moving house this weekend, so can't give much in the way of links and code for a while.  But I have the six sonar modules all being read by the arduino, which is sending data to Cleanflight via a serial link.

I2C won't work as each sensor needs a different address, and you usually use multiple sensors of the same type.

More to come.

Cheers,
Ben

leothehuman, 2015-05-24:
I2C will work if you give sensors different register numbers. You're not talking to the sensors, you're talking to the board, so you use one I2C address. The board reads sensors and puts values into registers on its own. In CF, you request register values from the board. Serial ports can't be shared, unfortunately, and there's not a lot of them. If you use I2C, there will be a lot more use of your product.

tednv, 2017-02-20:
what about one of these?

CJMCU-530
http://www.ebay.com/itm/2-3-5Pcs-CJMCU-530-Laser-Distance-Measuring-Sensor-ToF-Module-For-Rangefinder/122261378027?_trksid=p2045573.c100505.m3226&_trkparms=aid%3D555014%26algo%3DPL.DEFAULT%26ao%3D1%26asc%3D41376%26meid%3D8c359deec7e54f9eac83c953b40ce8df%26pid%3D100505%26rk%3D1%26rkt%3D1%26

it's already i2c and has a programmable address

only works under 2M but with a laser rangefinder this will be a huge improvement over sonar or baro for low altitude hover and maybe even obstacle avoidance

lidars are now $150+ thanks to garmin stickers

closed 2017-02-25 by hydra