cleanflight/cleanflight #779: Motor tab testing, ignore Motor Min/Max in configuration page and use ~1000\2000 values

MassiveOverkill, 2015-04-15:
I don't recall when the change occurred but I've noticed first-hand as well as reports from customers trying to calibrate their ESCs that the default value of 1850 isn't high enough to calibrate the ESCs when using the motor testing tab.  I've had to adjust the max throttle to 1950 in order for the ESCs to go into calibration mode.  I suppose the alternative is to change the default max throttle value to 1950.  

This is on Naze target.

thenickdude, 2015-04-15:
It doesn't make sense to calibrate your ESCs against a range which is larger than your mincommand/maxthrottle, though. That would leave you with missing thrust potential at the top of your range.

MassiveOverkill, 2015-04-15:
So I guess just raise the default value to 1950 for max throttle in the configuration tab so ESCs can be calibrated with those default values.

thenickdude, 2015-04-15:
1850 is a nice maxthrottle default because it's similar to the default 1860 max throttle of SimonK and the 1830 max throttle of BLHeli. Which ESCs require higher?

borisbstyle, 2015-04-17:
This would be a stupid feature. This would also cause a lot of calibration issues :D

cmavr8, 2015-04-17:
I agree with Nicholas, and I think that a user that has read the manual would have the proper numbers in min/max throttle. Of course the manual doesn't describe those yet, but it soon will (https://github.com/cleanflight/cleanflight/pull/663).

linuxslate, 2015-04-17:
Perhaps what we are asking for in not clear.  I am one of those who had difficulty getting a successful throttle cal on 2 different multi-rotors (different brand ESC's).  I also helped another person with difficulty doing an ESC throttle cal using the CF configruator.  We are not suggesting any changes to max_throttle, or anything about how CF works in flight.  To get a sucessful throttle cal, I had to set max_throttle to 2000, then go to the motors tab, do the throttle cal, and then set max_throttle back where I wanted it.  I discovered this when I was able to do the throttle cal in MW_EZ_GUI (android).  In EZ_GUI, the slider in the motors screen goes from 1000 to 2000 reguardless of max_throttle.
Throttle cal is (should be) based on the old #do not fly setting in MultiWii's .conf file.  I have not looked at the code, but I would be willing to bet that the #do not fly ESC Cal code in MultiWii ignored max_throttle.
It should be clear that if max_throttle is set low enough, and the motor tab slider only goes to max_throttle, that an ESC cal using the slider would be impossible.  What that value is probably varies from ESC brand to brand, but if the ESC does not see full throttle when it is powered, it will not go into throttle cal mode.
MW_EZ_GUI and CF Configurator work differently for this same function.  They cannot both be optimal.  One must be wrong, or at least not ideal.

MassiveOverkill, 2015-04-17:
Thanks for expanding on the issue linuxslate.

linuxslate, 2015-04-17:
I would say that many CF users are not even doing an ESC throttle cal at all -- Either they do not understand it, or the need to do it, and do not even attempt it, or they "go through the motions" but due to this issue, or the timing of operating the slider, etc., do not actually get the ESC's in cal mode.
Symptoms of a failed ESC cal can be:  Motors not spinning when armed (reguradles of whether motor stop is checked or not) -- Motors starting late or with hesitation.  -- Individual motors starting at different throttle settings.
Multiple failed attempts at ESC cals can result in entering the ESC settings mode, and accidently changing settings.  The end result of any of these is a aircraft that does not fly, or flyies much worse than it could.
Frankly, the old MW #do not fly method, even though it was text-based, and made you re-flash your whole flight software, was actually more sophisticated in (only) the respect that it was automated.  As nice as the CF GUI is, in this one respect, it has made what used to be an automated task -- manual.
An ESC Cal "wizard" or sequencer in the configurator would be nice. Ideally it would 1.) be "noob friendly" - i.e. guide you through the processes (with safety warnings), and 2.) ensure a successful cal on most brands/types of ESC's.

borisbstyle, 2015-04-17:
That sounds like an issue of EZ GUI rather than cleanflight :D. You can make a request to change that to Bart (dev from EZ GUI).  He is really fast with implementation.

linuxslate, 2015-04-17:
Don't get why the slider should not go the full range.  Perhaps there could be a tick mark at max_throttle, or another check box that would allow you to pass max_throttle, but I cannot see any down side to allowing the user to pass max_throttle when connected to the configurator and using the slider.  Perhaps someone could explain why that is bad.

borisbstyle, 2015-04-17:
You can use HOME and END button on windows machines, which works easier than ticking.

varesa, 2015-04-17:
I think some "override switch" could be an answer.  For the actual calibration the max value should be limited to the actual throttle range, but there might be a need to temporarily go over that to enter Cal.

linuxslate, 2015-04-17:
If the ESC does go into cal mode with max_throttle set to 1850, then the ESC will think 1850 is the highest setting it will ever see.  It will then apply *it's* full throttle to the ESC at 1850 -- i.e.subsequently changing max_throttle to 2000 would not result in any increase in max RPM.  If you calibrate the ESC to the true full range (2000), and then set max_throttle to 1850, the actual RPM at full throttle should then be lower.

linuxslate, 2015-04-17:
I looked at the MultiWii code.  MultiWii 2.4 uses the value ESC_CALIB_HIGH to cal the ESC's.  This is set to 2000 by default regardless of MAXTHROTTLE.  (I do see that ESC_CALIB_LOW is set to MINCOMMAND, and not fixed to 1000) There is no reference to MAXTHROTTLE at all in the Throttle cal code:

--->  Output.cpp

/********  special version of MultiWii to calibrate all attached ESCs ************/
  #if defined(ESC_CALIB_CANNOT_FLY)
    writeAllMotors(ESC_CALIB_HIGH);
    blinkLED(2,20, 2);
    delay(4000);
    writeAllMotors(ESC_CALIB_LOW);
    blinkLED(3,20, 2);
    while (1) {
      delay(5000);
      blinkLED(4,20, 2);
      SET_ALARM_BUZZER(ALRM_FAC_CONFIRM, ALRM_LVL_CONFIRM_2);
    }
    exit; // statement never reached
  #endif

----> Config.h

/********************************************************************/
  /****           ESCs calibration                                 ****/
  /********************************************************************/

    /* to calibrate all ESCs connected to MWii at the same time (useful to avoid unplugging/re-plugging each ESC)
       Warning: this creates a special version of MultiWii Code
       You cannot fly with this special version. It is only to be used for calibrating ESCs
       Read How To at http://code.google.com/p/multiwii/wiki/ESCsCalibration */
    #define ESC_CALIB_LOW  MINCOMMAND
    #define ESC_CALIB_HIGH 2000
    //#define ESC_CALIB_CANNOT_FLY  // uncomment to activate

I have never seen anything in MW that says you must set ESC_CALIB_HI to the same as MAXTHROTTLE, and as I say, I think that would be non-sensical.

So both MultiWii and EZ_GUI use 2000 as the high throttle when doing a cal.

varesa, 2015-04-17:
@linuxslate ah that is true, I've stopped calibrating ESCs long ago. They work just fine without (with proper hw+fw)

linuxslate, 2015-04-17:
It is true that I am not using the latest ESC F/W, or Oneshot, etc., but I have found much better results after calibrating the ESC's I use.  Like I said - multiple brands, and everything from 7A to 30A.

borisbstyle, 2015-05-04:
The guy from ezgui told me that he is fixing this now so that calibrating from the ezgui will use the configured values instead of default 1000-2000

eziosoft, 2015-05-06:
Should be fixed now

hydra, 2015-05-06:
Closing, nothing to do based on feedback.

closed 2015-05-06 by hydra