cleanflight/cleanflight #780: Please add Flysky I-bus just like Baseflight has

ridgeracer, 2015-04-15:
Hello,

I'm not sure if I'm posting in the correct section but is it possible to add I-bus to cleanflight just like baseflight has? Thank you.

hydra, 2015-04-16:
This shouldn't take someone long as the changes are quite isolated.  Ensure RX failsafe state is detected when porting.

ridgeracer, 2015-04-16:
Ok great hydra. Do I just wait for a new release of cleanflight for the feature?

hydra, 2015-04-16:
It won't happen by magic, someone has to do the work.  Waiting won't help, doing will :smile:

digitalentity, 2015-06-03:
Got a FlySky FS-i6 in my mail. Will post a PR for I-Bus serial receiver support soon.

ridgeracer, 2015-06-03:
Hey man. Thx alot for the help.

digitalentity, 2015-06-03:
@ridgeracer, what controller and what transmitter are you flying? I can
compile binaries for you so you can test this and leave feedback if it
works or not.

ridgeracer, 2015-06-03:
I'm using a FlySky i10. Yeah I can test it if you want.

digitalentity, 2015-06-03:
Here is latest firmware compiled for NAZE: https://www.dropbox.com/sh/l0t3usj0thzrxqn/AADMTak9tFxrOhhjVvvh_-2Ta?dl=0

Baseflight code (and this code as well) handles only 8 channels though, so you won't be able to use channel 9 & 10 for now. I don't have a 10ch transmitter/receiver to test if i-bus code can be extended to 10 channels.

ridgeracer, 2015-06-03:
Ok great. Now if I have the board wired for ppm is it the same for i-bus?

digitalentity, 2015-06-03:
No, you should follow the docs on how to connect a serial receiver and connect "SERVO" port on your FlySky receiver to corresponding UART on your FC

ridgeracer, 2015-06-04:
Ok so I used the same docs for baseflight for connecting the Turnigy/Flysky IA6B to the Naze32 board in serial mode. I flashed the Naze with the firmware you sent however the FC is not receiving a signal from the rx. I noticed that every time I set feature rx_serial in the CLI and I set serialrx_provider = 7 the FC never saves the first setting of feature rx_serial.

ridgeracer, 2015-06-04:
https://github.com/multiwii/baseflight/wiki/serialrx--like-SpekSat,-Futaba-SBUS,-Graupner-SUMD

ridgeracer, 2015-06-04:
https://github.com/multiwii/baseflight/commit/f564d366c7b6263620e39ccb7a3ff212fba77ec2

digitalentity, 2015-06-04:
Did you enable a "Serial RX" checkbox on the "Ports" tab in the configurator? You should only enable it on the port you have connected your receiver to. After selecting a "Serial RX" on "Ports" tab you can reboot and go to CLI and enter "feature RX_SERIAL" and "set serialrx_provider = 7"

ridgeracer, 2015-06-04:
Ok. So I was able to set serial rx on UART2 only. Went back to CLI. Was able to save feature RX_SERIAL and set serialrx_provider = 7 but still doesn't work.

digitalentity, 2015-06-04:
Are you sure you have your receiver connected to UART2?

ridgeracer, 2015-06-04:
I'm not sure it is UART2. I connected the black wire to ground pad. Red to positive pad next to pad 1. Signal to pad 4 on the back.

digitalentity, 2015-06-04:
Can you take a photo of your connections? There are so many board around...

ridgeracer, 2015-06-04:
Tell me about it. I spent the better part of this morning looking for UART2 connections for various boards. I'm out of town right now but my Naze 32 looks exactly like the one in the baseflight connection link. I'll take a pic this Monday. Thx for all your help.

digitalentity, 2015-06-05:
Should be UART2 according to docs... Photos should clarify if the wiring is correct.

ridgeracer, 2015-06-09:
[image]

ridgeracer, 2015-06-09:
Hopefully this helps. I purchased the board from MassiveRC.

digitalentity, 2015-06-09:
I see PPM input connected to ground, +5V and RC1 (that's PPM input). Can you take a picture of the wire from your receiver to the board when trying to connect I-Bus SerialRX?

ridgeracer, 2015-06-09:
I resoldered the connections for I-bus. Here are 2 pics of the top and bottom of the board with the rx.
[image]
[image]

MrSatoV, 2015-08-25:
Is there a plan to have iBUS working for CC3D?

dixiak, 2015-08-30:
I recently merged these changes locally and built an iBUS capable version of Cleanflight. I connected the 'servo' port on the receiver to the USART2 RX pin on the naze32. Had to set port, serialRX and the serial type - this last using a cli command. 

Works well. Flies well. Any reason this hasn't made it into mainline yet? Is it because of the failsafe problem?

MrSatoV, 2015-08-30:
Will that work on a CC3D or Sparky 1?

ridgeracer, 2015-08-30:
Can you guys look at my picture of my Naze32 board and tell me how to connect my rx to the UART2 port? I think I connected it wrong and never got it working with Ibus.

dixiak, 2015-08-31:
@ridgeracer I took a close look at your pictures and that's exactly how I hooked it up. Perhaps double check your config sequence: port, then config, then set serial provider type in cli.

ridgeracer, 2015-08-31:
Ok thanks again. I'll give it another try and let you all know if it works. Thanks again.

MrSatoV, 2015-09-07:
Spaarrkyyy?  XD

digitalentity, 2015-09-07:
@MrSatoV 
Should work on Sparky. Don't have one to test :)

MrSatoV, 2015-09-07:
So the support is built into the 1.9 Sparky hex?  I see a lot of people have 1.10, but my configurator doesn't show it.  Sorry for my ignorance, I'm finally making the switch from OP and TL.  Main reason is for iBus support, but also, that's where the party is.  LOL

digitalentity, 2015-09-07:
@MrSatoV 
When #983 gets merged official hex files will have support for i-Bus. Currently official binaries do not support it.

MrSatoV, 2015-09-07:
Oops, weird.  My phone freaking out, sorry.  I'll try to delete it when I get to a PC.
Is there a beta I can use to test the sparky?

light24bulbs, 2015-10-22:
Not sure if that merge happened.  do we have ibus support in main now?

MrSatoV, 2015-10-31:
I'd like to know that answer as well.

digitalentity, 2015-10-31:
No, this is not merged yet.

thedutchviking, 2015-12-07:
Forgive this noob comment, but would this discussion have anything to do with me being able to connect my FlySky FS-T6 / FS-R6B with Cleanflight? Right now I'm getting really strange input responses in the "Receiver" tab. 

ROLL - no response: fixed at 1500
PITCH - no response: fixed at 1500
YAW - no response: fixed at 1500
THROTTLE - no response: stuck at 885
AUX 1 - responds properly to SWD on/off
AUX 2 - responds properly to SWB on/off
AUX 3 - responds correctly to yaw input
AUX 4 - responds correctly to throttle input

ridgeracer, 2015-12-07:
Well this thread was originally started to port i-bus for the new flysky tx/rx over to cleanflight. You have an older non i-bus rx. You should set it for rx_parallel_pwm. Make sure you find a good diagram online for connecting your board to your rx as a parallel pwm connection.

thedutchviking, 2015-12-07:
Thanks @ridgeracer - that alone is helpful! Will do.

MrSatoV, 2015-12-07:
Is i-BUS working in Cleanflight or Betaflight?  I actually have a couple FS-iA6B RXs that do actually have i-BUS on them that I'd like to use in that way.  I put a Futaba SBUS RX on my prototype frame (after using PPM) and the difference is really noticeable.  Hard to go back on the others, you know what I mean?  XD

ridgeracer, 2015-12-07:
I-bus still hasn't been released on the official hex files. I tried the beta a few months back and couldn't get it to work with my Naze32. I posted pics above of how I had the rx leads connected to the board and the rx. Never really got around to buying a new fcb to test again. Sorry.

dixiak, 2015-12-08:
Working perfectly for me. I'm using an iA6B connected to the Naze32 over ibus serial. The naze is flashed with a private build where I integrated naze32 rev6 support with support for rx over ibus.

MrSatoV, 2015-12-08:
Share!  LOL

ridgeracer, 2015-12-08:
Yes please share the beta so I can try again. Also please see the pics I posted above and confirm that the rx leads are connected to the correct solder pads on my Naze32 so everyone has that part correct too.

dixiak, 2015-12-12:
http://www.dixiak.com/ibus
I've uploaded builds for people to try out. Please read the details before flashing blindly! There are some odd quirks.

MrSatoV, 2015-12-12:
Sweet!  I'll try to test asap... although not exactly sure when that will
be.  :-\  I have CC3D, OZE32 (Naze derivative), as well as a Sparky 1.1.
Actually more than one of each, but whatever.  LOL

Thanks for doing this Mr. Narco!

Shawn Sato-Veillon
佐藤  ショーン

ridgeracer, 2015-12-12:
Just to confirm, how I wired it in the pic is correct right?

hydra, 2015-12-14:
PR is merged now.

closed 2015-12-14 by hydra

MrSatoV, 2015-12-14:
WOOT!

Shawn Sato-Veillon
佐藤  ショーン

hydra, 2015-12-15:
Binaries that contain I-Bus support for testing are here:

http://cleanflight.memoryleaks.org/builds/cleanflight_cleanflight/2015-12-15/2529/

There may be newer binaries, if there are use them!

Please report any issues using new issues and reference this issue.

MrSatoV, 2015-12-15:
Cool, thanks.  I'm going to try it on CC3D and Sparky v1.1.

Shawn Sato-Veillon
佐藤  ショーン

huston007, 2015-12-22:
Hello. What baud rate should be set to UART where receiver is connected to? I tried default, 115200, on CC3D and it works, but should it be equal 115200 or may be below or above?

cklam2, 2016-01-22:
Just FYI: in the [rcgroup forum](http://www.rcgroups.com/forums/showthread.php?t=2486545) a guy created custom [firmware](http://www.rcgroups.com/forums/showpost.php?p=33772227&postcount=282) for the FlySky FS-i6 (and TGY-i6) which turns the 6 channel into a 10 channel transmitter. 

With this modification I was able to receive 8 channels in cleanflight through i-BUS with my FS-iA6B :)

And if someone has a data scope they can check whether it's possible to extend the nr of channels to 10 in cleanflight...

goldjacket, 2016-01-28:
Just trying to set up a ibus with my FlySky on a naze 32 with clean flight. With the FF - IAB 6 receiver. Downloaded multiple clean flight versions 1.12. What settings in clean flight do I select to enable serial communication. I don't see any Ibus boxes. the FF  - IA 6b receiver it tells me to connect to channel 1 according to the transmitter or do I connect to  top right of the FF -IA 6B where it says Ibus

huston007, 2016-01-28:
@goldjacket you have to use a development version of cleanflight configurator to see that options in UI.
https://github.com/cleanflight/cleanflight-configurator#alternative-way

goldjacket, 2016-01-28:
I have used multiple dev release . I'll try again with some of different
releases maybe that's the reason. Thanks

goldjacket, 2016-01-28:
Oh my bad I have to use clean flight configurator that is a dev. Not the
hex  files. Thanks

goldjacket, 2016-02-19:
So how do I enable the last 2 channels in cleanflight as to only 8 show up?