cleanflight/cleanflight #782: Allow splitting a UART between different unidirectional functions

ae5au, 2015-04-16:
For example, it would be nice to be able to use a single UART for S.Bus input and SmartPort output.  Another example would be GPS input with telemetry or blackbox output.

ae5au, 2015-04-16:
I apologize.  Though I thought I searched before I created this, I noticed that #713 already requested it and was closed.  I ask that it be reconsidered, though, because running OSD, GPS, and SBus on a Sparky (and without software serial yet implemented for it), I'm out of ports for Smart Port and would find it useful.

digitalentity, 2015-04-16:
I second this. Using GPS + something else on a single serial port would be great. Not all telemetry requires bi-directional link like MSP. I.e. MAVLink telemetry. CF does not support MAVLink but this can also be easily done.

thenickdude, 2015-04-16:
I don't think s.bus can be shared with smartport on the same UART because they run at different baud rates (1000000 and 56700). For the same reason you can't share GPS and blackbox because the GPS tends to run too slow to be paired with the Blackbox. 

If you don't use a bidirectional port for GPS then its settings can't be automatically reconfigured which would be a pain in the ass. My GPS doesn't even have flash memory on board for remembering settings while unpowered. 

I do plan to add support for sharing blackbox and s.bus, since they are both truly unidirectional and the Blackbox can happily run at that baud rate.

ae5au, 2015-04-16:
@thenickdude Good point about baud rate.  I wasn't considering that.

There are plenty of inexpensive GPS units that will run 115,200 and have flash for settings storage, so I wouldn't write off sharing blackbox and GPS on a port.

ae5au, 2015-04-16:
I'll let you decide if you want to close this or not.  The baud issue makes it less interesting to me at this time.

hydra, 2015-04-16:
if the systems share the same baud rate then it can be done, otherwise not.  softserial is a candidate to use io pins at different baud rates but it's non trivial due to shared timer usage.

blackbox and sbus together would be good.
smartport and sbus not technically possible since smartport is bi-direction single-wire and sbus is rx only and both use different baud rates.

not to mention the inverson issue that also will get in the way.

I'll leave this open since we do want sbus and blackbox, as @thenickdude mentioned above.

ledvinap, 2015-04-16:
Technically you can plan TX part so that it does not collide with expected RX time and change baudrate on the fly. On 303 this should even allow Smartport + blackbox .. but implementation won't be that easy ...

thenickdude, 2015-09-16:
I just realised that S.Bus is 100,000 baud, not 1,000,000 baud. In this case it seems that with the recent trend toward short looptimes, and many people finding that 250,000 baud logging isn't fast enough for them, 100,000 baud will never be an attractive baudrate for sharing with Blackbox. 

Closing this accordingly but feel free to reopen if someone else wants to implement it.

closed 2015-09-16 by thenickdude