cleanflight/cleanflight #784: Maple Mini Support

bk79, 2015-04-16:
based on the initial work done by silpstream I've completed porting to maple mini board.
What's in:
Two serial ports;
PWM RC support up to 6 channel ( on quadcopter or less than 4 motor/servo configuration)
PPM RC with  9 motor/servo output
I2C support (GY-88, GY-85) tested
It's in my fork I''l try to make a PR

JesperA75, 2015-07-06:
Great work!

Is there a special way of flashing the firmware to the maple mini? I've built the firmware and used the configurator to upload it but I get the "No response from the bootloader, programming: FAILED". I'm on mac, connecting the maple through USB. Device shows up in list. Should I use the serial port instead?

bk79, 2015-07-06:
@JesperA75  use UART1 D25-D26 VCP doesn't work

JesperA75, 2015-07-06:
Thanks man - will try tonight - the maple mini clone + gy-88 could be a very powerful combo considering the price! If this pans out well, this will be my new reference platform for my endeavours in swarms and cloud based autopiloting.

bk79, 2015-07-06:
@JesperA75 have a look also at the code developed by digitalentity with his nav-rewrite I've installed that one on my cc3d and I'm thinking of putting maplemini on it
https://github.com/digitalentity/cleanflight/tree/navigation-rewrite

JesperA75, 2015-07-06:
Looks very interesting @bk79 - this will be my first experiences with cleanflight - looking foreward to get the firmware on the board.

JesperA75, 2015-07-06:
Sorry to bother you @bk79 again. I've connected a 3.3v ftdi to the maple mini, and tried various ways of getting the maple into bootload from serial mode - but no luck. How did you manage it?

JesperA75, 2015-07-07:
Hmm - think I bricked it. Used the STM32 Flash Loader and got the hex file loaded onto the maple. After that it blinks slowly 5 times then nothing.

bk79, 2015-07-07:
@JesperA75  you can't brick it, it's doing that because it didn't recognize the i2c sensors, without those it will never start up, whenever you start it keep them connected  remember to use D0 for SDA and D1 for SCL, anyway you don't need the stm32 flasher: compile the HEX open the cleanflight from chrome select the firmware flasher , load the local firmware, make sure that boot D3 pin is  connected to GND and keep pressed D32 "button" and then press reset button, after 2 seconds release also "but" D32, you will be in bootloader mode:
now from the firmware flasher select No reboot sequence and full flash erase...then FLASH Firmware

JesperA75, 2015-07-07:
Ahh I see - I did just as you described but with no luck in chrome - maybe it's my cheap FTDI that works bad on mac - it just worries me that I cant seem to get it in a state now, where I can flash another firmware onto it...
Anyways - I haven't connected the GY88 yet, so if it's required for a proper startup then that certainly explains it. 
Thanks again - sorry to bother you so much, but the documentation is a bit flimsy on these maple clones. I hope to get to test your work soon :-)

bk79, 2015-07-07:
it works it's just a matter of timing for but/reset pushes to put it in bootloader mode fdti cheap or not works only as bridge, if you want to have a try without GY-88 open target.h inside maplemini and uncomment fake gyro and fake accelerometer, make sure that D2 (sorry my fault I told d3) is connected to GND

JesperA75, 2015-07-07:
Will try - got the flashing from chrome to work by using a different ftdi!
After flashing (with the fake gyro & accl), do I then remove the jumper on D2-GND?

bk79, 2015-07-07:
Yes it's used for the beeper

JesperA75, 2015-07-14:
Hey @bk79 - just wanted to let you know that I got the GY-86 board working nicely with your fork - nice!
Looking foreward to get this merged into the mainbranch.

Thanks!

bk79, 2015-07-14:
If you want there is a branch with also digital write nav rewrite

JesperA75, 2015-07-14:
@bk79 What does digital write nav rewrite do?

bk79, 2015-07-14:
Autonomous navigation but it's experimental

digitalentity, 2015-07-14:
@JesperA75 
I think @bk79 means DigitalEntity's nav-rewrite. That would be my attempt to improve altitude hold, position hold, return-to-home, waypoint navigation. It's in a very experimental stage, but if you are willing to help - you are more than welcome!

bk79, 2015-07-14:
@JesperA75 yes sorry writing with mobile in rush is always a not easy task

bk79, 2015-07-14:
@digitalentity I've added maplemini support to your branch because it's cheap and gives one more serial compared to cc3d that i own

JesperA75, 2015-07-14:
Thanks @bk79  and @digitalentity - I'll give it a try once I get the basic functionality working on my wing

digitalentity, 2015-07-14:
@JesperA75 
My navigation-rewrite does not support fixed wing yet, and I would be happy if you'll help add it.

JesperA75, 2015-07-14:
@digitalentity 
I would be glad to help out - I guess the biggest difference is that pitch is controlled through the IMU pitch instead of throttle. I will take a look at your source :-)

digitalentity, 2015-07-14:
@JesperA75 
Let's continue talking about fixed wing support for nav-rewrite here: digitalentity#26

giacomo892, 2016-10-07:
Maple Mini is working with NAZE32 hex but can someone point out where to connect MPU INT and MAG DRY (HMC5883)?
Thanks.

ps. are those actually used by inav for something?

closed 2017-02-25 by hydra