cleanflight/cleanflight #785: No Tricopter Yaw control below 1100us  throttle

Redshifft, 2015-04-16:
Naze32 board 1.8.1 fw Tricopter

After investigation I have found Yaw control is deactivated at exactly 1100us or lower throttle values.
Servo is active from power on and appears to be correcting yaw (armed or Disarmed) as expected but does not respond to TX when throttle is below1100us throttle stick.

disabling 'pid at min throttle setting' has no effect

cmavr8, 2015-04-16:
So, would you prefer the yaw user input to be taken into account always/regardless of throttle/arm status?

I believe the current behavior is intended, and meant to prevent the tail blade from hitting the ground/objects on the ground while landed and trying to arm/disarm.

Redshifft, 2015-04-16:
I am sure it was intended, disable yaw servo until arming would be nice but irrelevant because if you disarm from sticks the rear prop can still clip the ground when disarming. Best leave it exactly as it was in fw1.7.1 Full time functional, I'm sure all this would have all been considered in the codes previous life. Imo You just can not have the Tricopters finest asset not functioning at low throttles. 

Possibly ~75% servo travel below 1100us would best of everything, sufficient for flying and reduced for arm/disarm prop strike clearance.

cmavr8, 2015-04-16:
Could you please format your posts in a way that it is easy for readers to understand what you are requesting?

For example, you didn't explicitly answer my question above.

(Note: this is mainly to aid developers in understanding what we're talking about. I think I get what you're saying, but your statements should be clear by themselves.)

Redshifft, 2015-04-16:
Naze32 board fw1.8.1  Tricopter mix
Unexpected behaviour, NO user rudder (servo yaw control)  until throttle is more than 1100us in any situation.
"yaw user input to be taken into account always/regardless of throttle/arm status?"    YES

timkostka, 2015-04-16:
This is by design.  As Redshifft said, there is no way to distinguish between the user wanting to arm and the user wanting to control yaw.  You can set arming to a switch if you wish to recover yaw control at minimum throttle.

In v1.7.1, the yaw was also ignored at min throttle.  The only change is that now (v1.8.0+), pitch and roll are no longer ignored.

Redshifft, 2015-04-16:
No Yaw control does not return when armed so an arming switch would not help.
        
Timkostka, this is Tricopter mix not Quad X

thenickdude, 2015-04-16:
This is a conflict between the `tri_unarmed_servo` setting and the code added for the new `pid_at_min_throttle` setting. Previously, setting `tri_unarmed_servo` to 1 allowed the tricopter servo to move even when disarmed, which allows pilots to check that their servo is working correctly before take-off. However, now the yaw RC input is zeroed out long before that:

```c
// If we're armed, at minimum throttle, and we do arming via the
// sticks, do not process yaw input from the rx.  We do this so the
// motors do not spin up while we are trying to arm or disarm.
if (isUsingSticksForArming() && rcData[THROTTLE] <= masterConfig.rxConfig.mincheck) {
    rcCommand[YAW] = 0;
}
```

Looks like this code should not be run for tricopters when `tri_unarmed_servo` is set to 1.

timkostka, 2015-04-17:
If you set up a switch to ARM, `isUsingSticksForArming()` will return false, and the code won't ignore yaw any more.

Redshifft, 2015-04-17:
@timkostka , Yup dead right I just tested it for you. control restored armed or unarmed.

Who changed the code without testing and any idea why it was changed ?
More importantly can you straighten it out ?

timkostka, 2015-04-17:
I still think you're not reading everything I've written here.  Particularly, this:

In v1.7.1, the yaw was also ignored at min throttle. The only change is that now (v1.8.0+), pitch and roll are no longer ignored. Well, that and PID feedback is no longer ignored so your craft won't drop out of the sky anymore at min throttle (pid_at_min_throttle =1).

Cleanflight NEVER processed yaw in v1.7.1 or before at min throttle.  Go reflash with an old version and try it.

Redshifft, 2015-04-17:
Wrong, could not comment on what was processed but I had full control at any throttle position...told you this before, so we are quits. I have flashed multiple times back to 1.7.1, there must be something else then that has introduced this problem.

Looks like you are on the right track and it has to be something starting with 1.8.0
Appreciate your efforts, thanks

timkostka, 2015-04-17:
Oh, I guess my comments only applies to quads.  There appears to be
different processing for tricopters.

t's not a mystery, the code shown by thenickdude is what's causing your
issue.  But undoing that will undo progress made for quads.  I'm not
familair enough with how tris process inputs to be of help, especially as I
don't own a tri.

Redshifft, 2015-04-17:
Yes I saw you opened it as a Quad problem :)
Do we know who changed the code ?

cmavr8, 2015-04-17:
Why does it matter?
Anyway, it was by hydra, here I believe (ctrl-F for "mw.c"): https://github.com/cleanflight/cleanflight/commit/5e3734946ebb3950848a9213537fc4f191d6a6af

Redshifft, 2015-04-17:
Thanks, it does not any further than I might be able to ask them on the forum if they could do a  corrected Tricopter build  that I could test.
Interesting...https://github.com/cleanflight/cleanflight/issues/772
Sadly I could not give enough information, re 1100us at the time or this would all be sorted I'm sure.

tricopterY, 2015-04-17:
Use the Receiver Tab in the Configurator and check your throttle reading when the stick is at its lowest. Then try to adjust down your min_check value to say 50us below the lowest throttle reading. The default for min_check is 1100us.

Redshifft, 2015-04-17:
If you put min_check below min throttle you will not be able to arm, I got down to 1005us testing .
I just can't see how we can do it while min_check has to be within throttle sweep.

Redshifft, 2015-04-17:
Interesting, Quads also..
https://github.com/cleanflight/cleanflight/issues/787

Redshifft, 2015-04-17:
Is this a fix you can do Nick ? or are we talking next release ?

thenickdude, 2015-04-17:
So, the old behaviour on 1.7.2 was:

<table>
<tr><th colspan="5">Is yaw control of the tricopter allowed? 1.7.2</th></tr>
<tr><th></th><th colspan="2">Disarmed</th><th colspan="2">Armed</th></tr>
<tr><th></th><th>Throttle low</th><th>Throttle normal</th><th>Throttle low</th><th>Throttle normal</th></tr>
<tr><td rowspan="2">tri_unarmed_servo = 0</td><td>No</td><td>No</td><td>Yes</td><td>Yes</td></tr>
<tr><td>No</td><td>No</td><td>Yes</td><td>Yes</td></tr>
<tr><td rowspan="2">tri_unarmed_servo = 1</td><td>Yes</td><td>Yes</td><td>Yes</td><td>Yes</td></tr>
<tr><td>Yes</td><td>Yes</td><td>Yes</td><td>Yes</td></tr>
</table>

The current behaviour on 1.8.1 is (tri_unarmed_servo is broken):

<table>
<tr><th colspan="5">Is yaw control of the tricopter allowed? 1.8.1</th></tr>
<tr><th></th><th colspan="2">Disarmed</th><th colspan="2">Armed</th></tr>
<tr><th></th><th>Throttle low</th><th>Throttle normal</th><th>Throttle low</th><th>Throttle normal</th></tr>
<tr><td rowspan="2">tri_unarmed_servo = 0</td><td>No</td><td>No</td><td>No</td><td>Yes</td></tr>
<tr><td>No</td><td>No</td><td>No</td><td>Yes</td></tr>
<tr><td rowspan="2">tri_unarmed_servo = 1</td><td>No</td><td>No</td><td>No</td><td>Yes</td></tr>
<tr><td>No</td><td>No</td><td>No</td><td>Yes</td></tr>
</table>

It's clear that tri_unarmed_servo needs to be fixed, so if its enabled then yaw control is always possible, just like before:

<table>
<tr><th colspan="5">Is yaw control of the tricopter allowed? Proposed</th></tr>
<tr><th></th><th colspan="2">Disarmed</th><th colspan="2">Armed</th></tr>
<tr><th></th><th>Throttle low</th><th>Throttle normal</th><th>Throttle low</th><th>Throttle normal</th></tr>
<tr><td rowspan="2">tri_unarmed_servo = 0</td><td>No</td><td>No</td><td>No</td><td>Yes</td></tr>
<tr><td>No</td><td>No</td><td>No</td><td>Yes</td></tr>
<tr><td rowspan="2">tri_unarmed_servo = 1</td><td>Yes</td><td>Yes</td><td>Yes</td><td>Yes</td></tr>
<tr><td>Yes</td><td>Yes</td><td>Yes</td><td>Yes</td></tr>
</table>

What's not clear to me though is if we should be allowing yaw control on tricopters when armed, throttle is low (i.e. below mincheck) and tri_unarmed_servo is set to 0. This was previously allowed on 1.7.2. But it seems to me that if you set tri_unarmed_servo to 0, it's because you don't want your tail servo to be active on the ground (because it might cause the tail prop to hit the ground). In that case, you probably don't want the tail servo to move while performing the disarming stick command.

Any comments from tricopter pilots?

Redshifft, 2015-04-17:
Proposed is great, perfection would be a 75% servo travel limit below min_check.
How could your proposal not keep everyone happy.....Thank you, I appreciate the work :)

You just can not odds the disarm unless we can add a kill switch which would be a great feature for us wanting the safety of "stick arm"
But we have to have that yaw control back.

ethomas997, 2015-04-17:
I noticed that change with the new version on my tri, but just kind of went with it.  Your proposed chart looks good me.  When MOTOR_STOP is enabled, the tail prop hitting the ground is not such an issue.  Being able to run the tail servo all the time can be handy during testing and setup.  I don't think its travel limit should be reduced; that could cause confusion.

--ET

ethomas997, 2015-04-17:
How about if we implement a "tri_unarmed_servo = 2" case that does "No  No  Yes  Yes" like it was in v1.7.2 (tail servo always active while armed).  In this case if arming via the rudder stick, we'd want the tail servo inactive until after the rudder stick is back to center.

--ET

Redshifft, 2015-04-17:
@ethomas997 , If you run motor stop I can't see any of this makes quite so much  difference. When you pull the throttle back your motors are slowing to a stop!, your yaw will not do anything.
I run a motor Idle so I still would have control, but as1.8.1 is, as the throttle stick is approaching idle at 1100us, bang, no yaw then another big bang and wobble on the servo as it starts again or a jittering mess if the stick hovers around 1100us as yaw kicks in and out.

Servo travel limit to 75% at lower than 1100us throttle would reduce the chance of a propeller strike arming or disarming but not significantly affect yaw authority at near closed throttle.

cmavr8, 2015-04-18:
Side note/question: does any of this affect whether CF corrects using the servo?
Does CF try to correct (move servo) while motor rpm is 0 (throttle < min throttle and motor_stop =1)?
If 2 x yes, I have a note to make.

thenickdude, 2015-04-18:
The tricopter mixer doesn't consider or alter the motor RPM when moving the
servo, it just tilts it based on how much yaw you're commanding (after it
gets fed through the PID loop).

cmavr8, 2015-04-18:
Ok then my comment is relevant:
I have a tricopter that's not running CF currently (APM). When on the ground and armed it starts oscillating because the FC is correcting but the correction has no effect (because of 0 RPM) and at the same time it wobbles the tail boom around (big motor is moved), which makes the FC correct yet again etc etc. It never stops oscillating unless I take off, or hold it by hand, or disarm.

It's not a tuning issue; no matter how much I lower gains it still does this.

The same frame, with the KK2 board, had no problems at all, as that FC always outputted center pwm to the servo, when throttle is at zero.

So although I do not use CF on a tricopter currently, I'd like to bring this issue to your attention, in case it's easy to take care of it while fixing the main one you've been discussing.

thenickdude, 2015-04-18:
Presently on CF I think you would need to set MOTOR_STOP and set tri_unarmed_servo to 0 in order to avoid that effect. These settings would mean that the motors start spinning at the same time that the servo PID loop begins controlling the tail servo, so hopefully the minthrottle spinning of the props cancels out that oscillation.

Redshifft, 2015-04-18:
@cmavr8 
That would be "noise" induced, better Servo, ferrite choke on the lead, wire runs, filter option on the apm etc..

@thenickdude  would you be doing a quick fix build for this or would it have to wait for the next release ?

thenickdude, 2015-04-18:
@Redshifft here's a Naze32 firmware you can try with this fix, it's based on the current development head:

https://github.com/sherlockflight/cleanflight-dev/releases/tag/v1.8.1-tricopter-yaw

I've never tested it, so please try it and let me know.

Redshifft, 2015-04-18:
@thenickdude Err, slight problem Nick, did you try to flash it to a board ? and it was for a Naze board ?
I flashed the hex all seemed ok but now I can not connect to the board. Tried a reflash but it does not see the Bootloader and the PC gets a bit emotional if I persist and Bluescreens with a USB controller problem.
Thoughts ?

Redshifft, 2015-04-18:
FYI the flasher reports loaded firmware as 114284 bytes

thenickdude, 2015-04-18:
No, I didn't bother to flash it as the change should have been trivial. I'm AFK at the moment so I can't give it a go. 

It's not possible for FC firmware to trigger bluescreens because the USB connection is provided by a separate chip which doesn't care what the FC is doing. 

You should be able to reflash by shorting the bootloader pads:

https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20Naze32.md

tricopterY, 2015-04-18:
Please see  cleanflight/cleanflight#774

Redshifft, 2015-04-19:
@thenickdude  Yes no problem, I have already dealt with the corrupted boot loader ;)
You would think that about the bluscreen but it did it twice.
When you return, pull down that hex you upped and crc to the original maybe it got corrupted without it being Rar or zip it's possible.

Redshifft, 2015-04-19:
@tricopterY  I have flashed back and forth about 10 times on this original problem with no issues, and that was 1.8.1 strange. Genuine Naze board here, actually never ever had any issues with flashing with any of them.

tricopterY, 2015-04-19:
If @thenickdude used the commit with cleanflight/cleanflight#774 issue, for your  fix. Naze32 wouldn't boot, as there was a bug in that particular commit - that's now fixed.
Developers normally grab the Master for their fixes/updates and not a stable version.

Redshifft, 2015-04-19:
@tricopterY , I doubt he made any sort of mistake like that, they don't call him the dude for nothing :D

tricopterY, 2015-04-19:
:smile:

thenickdude, 2015-04-19:
Ah yep sorry about that, looks like I forgot to merge the latest commits after I did a fetch. I'll post a fixed firmware later today.

Redshifft, 2015-04-19:
LOL, Correction, used to call him the dude :D

thenickdude, 2015-04-19:
Okay, here's the fixed firmware:

https://github.com/sherlockflight/cleanflight-dev/releases/tag/v1.8.1-dev-tricopter-yaw

I even tested it this time! :)

Redshifft, 2015-04-19:
Thanks Nick, you are a star...and returned to full Dude status :D
Appears to work fine so far.
Has me a little confused regarding min command on the build you based the fix on, Is this no longer processed in the same way ? you can put it below your RC and still arm (with sticks) indeed you have too,  to stop what looks like failsafe or self arm kicking in. I need to play a little more ;)

cmavr8, 2015-04-19:
@Redshifft what I described is not related to RFI. It happens if the frame's landing gear is not completely sturdy, and is avoided completely when an external force (my hand) keeps everything steady.

The last of thenickdude's comments disappeared somehow. Here it is: "Presently on CF I think you would need to set MOTOR_STOP and set tri_unarmed_servo to 0 in order to avoid that effect. These settings would mean that the motors start spinning at the same time that the servo PID loop begins controlling the tail servo, so hopefully the minthrottle spinning of the props cancels out that oscillation."

So I guess it's already taken care off... :+1:

Redshifft, 2015-04-19:
@cmavr8 Do not agree there, I've had it with cheap servos on long leads, disturb them and they jitter, difficult without a scope to see if it's noise or bad rejection, cheap servo normally bad rejection.

thanks for re posting Nicks comment, what happened there ?

cmavr8, 2015-04-19:
Yeah I don't have enough experience Redshifft, but the same setup worked with the kk2 and not APM. And it's not just jittering, the correction is directly proportional to the width of the oscillation. 
Anyway, could be both, but since it's solved it's fine :)

Redshifft, 2015-04-19:
@cmavr8  I know little about KK boards but I thought the APM servo PWM output rate was higher than the normal servo 50hz, something like 100hz. Unless your servo is digital that might not be helping anything !

Redshifft, 2015-04-19:
@thenickdude  Well I have been playing with the Tricopter & Cleanflight  for 3 hours and all is excellent, I'm sure it will fly fine :)
Thanks again to Nick and others.
Final question if I may, all the versions of CF I used with this TRI, the servo corrected outside yaw movement armed or disarmed, has the TRI mixer always had a live yaw control loop from power on ?

Feel free to close this bug report as you wish, I don't  know how it works around here.

ethomas997, 2015-04-20:
I tried out the "v1.8.1-dev-tricopter-yaw" version and it works as expected (like in the "proposed" table above).  It looks good to me.

--ET

nebbian, 2015-04-21:
As a tricopter pilot, the old behavior of Cleanflight bothered me.  I was running 12" props and they would strike the ground when arming and disarming, whether motors turning or not.  This puts excessive stress on the servo and dings up the prop, for no reason at all.
The proposed table above is excellent, I would run motor_stop and tri_unarmed_servo = 0, which would solve all my previous issues.

tricopterY, 2015-04-21:
I am happy to see more and more converts to enabling motor_stop. This feature has a lot of merits, but people object, having it enabled by default. Happy flying.
PS. please do not start another conversation for and against motor_stop. 
I couldn't help it myself - opologies.

Redshifft, 2015-04-21:
More important is we retain User choices in Cleanflight .

closed 2015-04-22 by hydra

Redshifft, 2015-04-23:
Link to yaw servo issue for you Tricopter guys.

https://github.com/cleanflight/cleanflight/issues/803