cleanflight/cleanflight #787: When throttle bellow Min Check , Yaw user input does not work.

Tabsy123, 2015-04-16:
Im on Naze32 with 1.8.1
Im flying on Quad X layout craft.
I have motors spin at idle.
When throttle is lowest point (bellow min check value)
I have control of Roll and Pitch
But not yaw
Would be good to have all controls working not just some.

Craft seems to be correcting itself for yaw ,  but Stick input does nothing.

timkostka, 2015-04-16:
This is by design.  By default, where the user arms and disarms using the sticks, the yaw input at minimum throttle is ignored.  This is because you don't want the craft to respond to yaw when you're trying to arm or disarm.

You can set arming to a switch if you want to control yaw at min throttle.

Tabsy123, 2015-04-16:
Ok thanks for clarifying, I didnt think of that,
So would i just set my tx lowest point in throttle to be just above min check value?
Then arm with switch & tick that box in gui that allows arm and disarm with switch regardless of throttle value?

timkostka, 2015-04-17:
Just set up an arming mode via a switch.  Then, the code knows you aren't using YAW to arm and it will not ignore it at min throttle.

thenickdude, 2015-04-17:
@Tabsy123 it's better to change the value of min_check on the CLI. If your throttle sits at 1000 at minimum, then the default min_check of 1100 uses 10% of your stick throw for "arming commands". You can probably reduce that setting to 1050 or even lower and still be able to perform stick command reliably.

Redshifft, 2015-04-17:
You are screwed because its essentially throttle stick value this problem is referenced to, so you aint gonna be able to set min_check lower than full back throttle stick.

I reported the problem for the Tricopter mix, surprised you are the first to pick this up on a Quad.

Tabsy123, 2015-04-17:
It's all good I have it arm and disarm by switch, and as said all works perfect now

Redshifft, 2015-04-17:
That works for the Tri also, sadly not something I will even consider.

thenickdude, 2015-04-17:
This can't be "fixed" on quadcopters because otherwise when arming or disarming with the sticks, the craft would try to rotate itself around on the ground. It can be fixed back to the old behaviour for tricopters since waggling the servo won't cause it to flip over on the ground by accident. 

You can set pid_at_min_throttle to 0 and have pitch and roll also disabled when the throttle stick is low, which was the behaviour on 1.7.2.

closed 2015-04-17 by thenickdude

Redshifft, 2015-04-17:
Even if the arming procedure was set to activate after the yaw stick was returned to center,  Yaw 3 sec, zero 2 sec, arm ?

thenickdude, 2015-04-17:
No, as that wouldn't do anything to fix the disarming procedure.

Redshifft, 2015-04-17:
Yes of course, disarming throttle position is irrelevant if the control loops are running, easy to overlook after so long with the old way.
Although a switch disarm is no big deal, I just would not like a switch arm from a safety angle and being different to every other quad system, kill switch no problem at all in my books and faster than a stick or combination stick input. would be a great feature/option to  have in CF anyway.