cleanflight/cleanflight #789: Motors arm when using looptime = 0

borisbstyle, 2015-04-17:
I discovered an issue with the current stable 1.8.1 build. I am not sure if it existed in the previous builds as I never used looptime 0.

Basically this happens:
When looptime is set to 0 some motors can arm and spin on idle speed when no TX is not turned on. It seems like some kind of bug in failsafe detection during initialization. The funny thing is that not all motors would arm, but mostly 1 or up to 3 what I have seen. It seems random.

Setup used:
Frsky D4R-II
SN20 Blheli ESC 13.1 and 13.2

When you set looptime to 1200 for example the problem is gone.

DominicCliftonNEKTAN, 2015-04-17:
See #771 and #546.

Thanks for the report but please search before opening new issues :smile:

DominicCliftonNEKTAN, 2015-04-17:
Also, a new configurator was released which was the cause of people accidentally setting their looptime to 0 when restoring backups.

hydra, 2015-04-17:
See #771 and #546.

Thanks for the report but please search before opening new issues :smile:

Also, a new configurator was released which was the cause of people accidentally setting their looptime to 0 when restoring backups.

closed 2015-04-17 by hydra

borisbstyle, 2015-04-17:
Thanks. I couldn't find it