cleanflight/cleanflight #791: Trim Drift Issue in Horizon and Angle Mode

rscarawa, 2015-04-17:
I have noticed in the various releases (now in 1.9.0) that the quad drifts back and to the right.  I had seem this before upgrading and it took awhole lot of trim to get it to stay still.  Even after re-calibrating the accelerometer, that did not fix the issue.  Any idea why it seems to do this?  When it drifts, the quad is noticeably tilted in the air.

Is there a tool readily available for me to read the black box logs?  SOmething that does not require compiling?

https://www.dropbox.com/home/rc/cleanflight/1.9.0/Trim_Drift

Redshifft, 2015-04-21:
Normally it's your radio drifting from 1500us center, give it a check. Somewhere there is a deadband setting if you can't  fix the radio.

thenickdude, 2015-04-22:
@rscarawa you can read the logs with the blackbox web viewer:

https://github.com/cleanflight/blackbox-log-viewer

Or if you're on Mac or Windows you can download the prebuilt binaries of blackbox_decode and blackbox_render:

https://github.com/cleanflight/blackbox-tools/releases

Your link to your log doesn't seem to work for me (you might need a special "share link" to get something you can link to other people).

KernoQuads, 2015-04-23:
Hiya i have also had this issue. Fly's fine to start with no drift issue. Then after doing a few flips/rolls i have noticed it will start drifting to the right, have checked my transmitter everything is centerd at 1500, Cleanflight also comfirms that they are all centerd,

Tabsy123, 2015-05-02:
yes ,  I have this issue also,  pitches back alot,   its intermittent, seems to be a bit dependant as to where the stick/craft is when angle mode is activated.  I have tried accel calib ,  trims,  channels centered, none of which helped for more than a minute maybe,  as it is intermittent.

Tabsy123, 2015-05-03:
Ok My Problem has now been temporarily solved, Even tho it does not make sense for this do effect this fault, It does,  more so on crafts with low weight and heaps of thrust. 
Turning of Damped Light/Active braking in ESC BLHELI Solved this angle mode problem.

hydra, 2015-05-06:
Please provide a blackbox flight log that highlights the issue, then we'll have some data to work with.

ctzsnooze, 2015-05-06:
The exact same thing happens to me, but in my case the offset is to the left.  I can trim Acc and then in flight it soon after enabling Acc will lean hard left.  If I trim it hard right, then before takeoff on half throttle it leans hard right.  Something is completely wrong.  I think the key factor seems to be OneShot and BLHeli as it doesn't seem to affect everyone.  Can data be logged via the CleanFlight Chrome App? - it happens in my hand every time but i have no logging chip.

ctzsnooze, 2015-05-06:
PS - what exactly does increasing or decreasing these parameters achieve?  I can't find any documentation.  Nothing I do to them changes the issues I face with unstable Acc offsets in flight.

set acc_lpf_factor = 4
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff = 5.000

ctzsnooze, 2015-05-06:
See this video post please:

http://www.rcgroups.com/forums/showpost.php?p=31545774&postcount=7347

ctzsnooze, 2015-05-07:
Made a blackbox log to the computer (tethered by USB).  The quad is in Horizon mode the whole time.  I was holding it in my hand and not touching the sticks other than to arm and throttle up (see attached video with overlay - must admit that is very cool!).  On arming in Horizon and throttling up, it first leant forward, then a bit to the left, then settled with about 10-15 degrees of left angle, not as much as usual, but definitely not staying level. I hope this is visible in the video. It looks like I was kind of pushing it in those directions, but I wasn't it was moving around slowly and a bit randomly all by itself.

 If the video and log is not good enough I'll make a better one.  Later, as you can see better in the video, I pushed actively forward and backward against the levelling of the quad, in separate distinct steps, and it responded by pushing back - and overshooting. That's easy to see. (I don't get overshoot in Acro).  Here is the the log file:

https://dl.dropboxusercontent.com/u/9505117/HorizonLogCT.txt.zip

and the video with overlay is here:

https://dl.dropboxusercontent.com/u/9505117/testing%20Horizon.mov

Hope someone can make sense of this.  It looks like I'm being excessively fussy but it won't fly.

PS - I could see initially a lot of fast oscillation in the gyros so I dropped gyro_lpf back from 256 to 42.  I couldn't really feel or see the oscillation at 256 but it was obvious in the log.  It did feel smoother at 42.  I also set the acc_lpf to zero (on the advice of someone who said that improved acc responsiveness) and I set both the acc deadbands to 0.  Neither had any affect.

ledvinap, 2015-05-07:
IMO the problem is caused by ACC noise that is aliased by internal ACC sampling. What do you get in motor test tab?
You should try balancing props and using some vibration isolation in FC attachment ... 

Here are heavily filtered ACC data from your log:
[image]

ctzsnooze, 2015-05-08:
Thanks so much for looking at the log, I am very grateful indeed.  I'm so grateful to Nick for making logging possible.

I agree there is quite a lot of noise (both acc and gyro).  Despite the noise, the filtered Acc data, showing significant absolute offsets in the green and blue, is showing the actual movement of the frame.  At 455..460..470 the frame was moving of its own accord with no stick input, I just allowed it to wander, as it does.  The magnitude of the frame angle changes in the first half of the filtered data is exactly what I could see in reality.  The red offsets show the loss of angle; it was sudden and a bit random, generally to the left.  That's exactly the issue for which I'm trying to discover the cause.  

The bigger offsets later in the trace - say at 490 and 495 - were deliberate manual perturbations that I generated to test the levelling capability.  Which is OK but for the excessive correction after release.  I pushed and held the quad off-level, the correct motors sped up opposing me, and I released the pressure, and there was a too-big overshoot correction when I released it.  This is well reflected in your filtered Acc log.  The manual offsets, and the overshooting behaviour, are being tracked properly, and again are correct reflections of the frame angle.  

So I'm glad that the log shows precisely what's happening.

I also looked at it and thought about the possibility that frame noise could be causing it.  There are a number of reasons why I think it's unlikely that this is the fundamental cause and one reason why it could be.

At first the noise was even worse.  I checked all the props and did dynamic balancing but it wasn't a prop/motor balance issue.  Mostly it comes from BLHeli just not running the motors smoothly - there are certain throttle points where some motors just aren't perfectly smooth, it has a harshness and that shows up as frame vibration.  If all motors are at a speed where they all tend to not be perfectly smooth there can be a bit of vibration I can feel.  Usually though I can't feel it.  I've tried, and I can't fix it no matter what the BLHeli settings.  And then when all motors run at approximately the same speed there is inevitably some vibration from motor frequency aliasing.

At this point I had a gyro_lpf of 256 and a loop time of 1250, and there was a heap of very fast noise in the log.  Strangely the frame felt OK - you couldn't feel it, and it flew really well.  Obviously the gyros had no problem.  So I dropped gyro_lpf to 42 and loop time up to 2500.  

This greatly reduced the frame noise - but the acc offset issue did not improve at all.  This suggests that noise isn't the primary cause.

Yes, the FC is hard mounted to the frame, like on all my other acro quads, and even my brushless quads which vibrate a lot.  I have no issues with acc not staying level like this on any of them.   Adding vibration isolation is really a kind of unpredictable LPF, and in my case not easy to do because of the wiring and mounting situation.  But it isn't needed on any other, and just generally this one vibrates much less than other quads that stay perfectly level.  

Finally vibration in acc data should be random.  Even with aliasing errors, there should be no specific offset arising from random noise.  The filtered data suggests that the offset can be seen despite the noise. In actual flight I get a very specific completely repeatable tendency for the quad to set itself to a left roll angle.  It is nearly impossible to trim it out.  If random noise was the cause, it makes no sense that the direction angle offset would be so predictable.  

The one factor implicating noise is that there is a relationship between angle offset and motor speed. At low throttle, it angles right, and at higher throttle it angles left.  I have excluded non-linearities in thrust by direct motor testing in the motors tab.  At no point in the throttle range do the motors appear to themselves want to push the frame to any particular angle.  Even if there were small variances in motor power, the FC should push the weaker motors a little harder and retain level anyway.  

So I can't really explain the angle dependence on motor speed.  The only possibility is that the motor induced noise is not being filtered evenly, inducing some apparent offset in the incoming acc data.  But it makes no sense that the much bigger actual frame offsets (huge, up to 45 degrees) do not overcome the very much smaller vibrations in the the frame.  If so then something fundamental is wrong with the sensor filtering.

I have found no change in behaviour whether the acc_lpf value is 0 or 250, it makes absolutely no difference.  I've put the other three GLI Acc tuning values to all extremes to no apparent effect.  

Also,  looked at several other users flight logs - and I see huge amounts of vibration while they fly; typically this is in Acro though.  I have no other logs to compare for this kind of issue.

And finally the gyros are very able to deal with this noise.  The same quad can hang perfectly level in Acro.  So it is definitely possible to maintain level with gyros - this implies some problem with the code managing the Acc data itself, perhaps how noise is managed, I don't know.  

I have come to the following conclusions:

[... cut ...] n the general Horizon behaviour in PIDC2.

Sorry this is such a long email.  I will try to look at the code base myself and see what I can change.  I flew this morning briefly with lots of Acc trim and it sort of works OK despite the left drift.  I think maybe with 150-200 of trim it might be OK (not on take-off).  I will also look at the physical mounting of the FC in case there is a possibility that one particular motor can somehow have more influence over the Acc noise.

Thanks for your help.

thenickdude, 2015-05-08:
`acc_lpf_factor` controls the smoothing applied to the accelerometer readings before they're used to estimate attitude. It's used to control a weighted sum between the previous smoothed value and the new raw ACC reading. The default of 4 means that the new smoothed reading will be 1/4 of the raw reading plus 3/4 of the previous smoothed reading.

If you set `acc_lpf_factor` to 0, smoothing is completely disabled and only raw ACC readings are used.

`accxy_deadband`, `accz_deadband` and `accz_lpf_cutoff` are used during integration of accelerometer data (to infer absolute position). The output of these calculations is only used by the altitude hold feature, so these parameters have no impact on ordinary LEVEL flight.

@ctzsnooze that's a great test video! I need to play around with that log. Keep an eye on that "acceleration" value at the bottom left corner of the screen, it's invisible most of the time because it's the same colour as your laptop. It starts out near zero but by the end of the video it seems to be sitting around 0.3G even when the craft is idle? Can you maybe do a test with the craft just at idle throttle on the carpet and not moving the whole time? It'd be interesting to see how the acc calibration drifts. It's known to drift as the system warms up for some users, so I wonder if that's what is happening to you.

I plotted your acc data on the Blackbox web viewer and noticed that your acc X & Y noise is almost -0.5 to +0.5 G, which seems ridiculously high. Here's what my craft looks like (in flight):

[image]

And here's yours handheld:

[image]

thenickdude, 2015-05-08:
Yeah, grab this tweaked version of the web viewer:

https://github.com/cleanflight/blackbox-log-viewer/tree/acc-ignore-marker

Open up your log and click the "graph setup" tab. Add a new custom graph which plots the field `usingAccForAttitude`. This field is zero when CF would ignore the acc when computing attitude and one when it is using the acc. Your acc is being ignored quite often:

[image]

Ideally it should be being used more often than that when you're in a stable hover.

ledvinap, 2015-05-08:
@ctzsnooze : I still think that ACC noise is main problem
ACC is sampled at 1kHz. Any acceleration signal over Nyquist frequency (500Hz) will be aliased into 0-500Hz range (see http://en.wikipedia.org/wiki/Nyquist%E2%80%93Shannon_sampling_theorem).
Motors generate quite narrow-band signal, probably with same harmonics, so there may easily be quite large frequency content near 1kHz. Sampling will wrap this near 0Hz, possibly even causing constant offset.
Also the accelerometer is mechanical device, signal near it's resonance frequency may cause constant offset due to it's structure/imperfections.

No filtering will help in this situation - the spectrum is wrapped during 1kHz sampling and there is no way to separate it again.
Mechanical filter (gel sticky tape may be enough) will help to reduce frequency content over Nyquist (500Hz), digital filtering will then attenuate remaining noise in sampled  signal.

Gyro is sampled at 8kHz internally(and it's resonant frequency is around 30kHz), so there is more headroom to filter it - lowpass filters can work in this case.

ctzsnooze, 2015-05-08:
Thanks so much for your advice!  

Levidnap, clearly I must do something about the vibration.  I's not not easily felt in the frame at all, so the frequency must be up in the audio range.  

Nick, sorry, I can't sample faster as I am using a wired connection to an oldish FTDI and it won't go much faster.  I don't have logging on the AfroMini itself.  PS can I congratulate you on the amazing work in the logger, absolutely amazing and brilliant!

I am just now testing the motors one by one and looking at the live Acc data. 

So much noise!  But finally a very clear clue as to the problem!  :-)

The quad itself was resting on the desk as per this photo:

[image]

Note that at all times the flight controller in the centre was held quite still by my hand on top of it.  There was no way it could move much at all.  It was possible for some rotation left/right about the axis of the test motor arm, but otherwise the battery and the arm under my hand stayed on the table the whole time, and the absolute angle of the FC could not have moved more than a degree at any time.  The arm itself is quite stiff and the central core where the FC is located is very solid indeed.  So its difficult to imagine kind of any movement whatsoever of the Acc sensor.

Despite being held almost completely still like this, the FC in the centre of the quad recorded nearly 1G of vertical movement at times during the motor test (!)  Is that really possible?  I checked the Acc and it seems perfect in static xyz measurements (1.0G in esc direction), so this must be some bizarre integration issue as Levidnap explained - thank you!  :-)

The movie linked below shows what I see in Configurator while slowly upping throttle.  You can hear the sound of the motors and see that at certain speeds there is an abrupt change in the sound of the motors, sometimes a whistling, and with it an abrupt onset of Acc noise.  At full rpm there is almost 1G vertical noise just from one motor!  I find it hard to believe this is possible when the Acc itself appears rock solid on the desk under my fingers and I can't feel hardly any vibration at all.

BUT the amazing thing is what can be seen in parts of the movie - and now, I realise ledvinap is probably entirely correct:

[image]

Note how when the motor vibration suddenly appears in the red (z) axis, suddenly the Acc reports almost 1/3 G offset!  At that moment, the FC has not moved at all, it still in reality has the same absolute angle, held firmly on the desk.  This 1/3G is completely false!  If an apparent offset like this can appear just due to motor induced vibrations, that would explain exactly what is happening.  

Something like this could definitely explain the motor speed related and highly variable drift tendency.  

So I think now we know the root cause... but what is the solution?  And are we sure the Acc filtering in in CF is working properly?  Could it be that the Acc filtering on this particular AfroMini FC is somehow dodgy? 

I have multiwii quads that have sometimes had appalling vibration (chunks off props that I didn't notice, trying to fly with my last bent prop, etc) but they could still hold level really well.  So on multiwii boards, the Acc can cope with heaps of vibration.  This infers an issue with Acc filtering in CF as the root cause (or a bad 6050, or...  ??).

Here is the full movie showing the motor sound and the various motor induced FC acc changes:

https://dl.dropboxusercontent.com/u/9505117/Motor1.mov

If there is something fundamental wrong in CF with Acc filtering this would well explain why a number of people have the same generic kind of problem.  With fast loop times OneShot and active damping the likelihood of very high frequency noise is much higher.  This is where we are all headed.  I do not want to resort to mechanical filtering because that is so unpredictable.  In contrast, I am running Kiss active braking ESC's with MW2.3, oneshot, and 1500 loop times with the same motors and ABSOLUTELY no issues like this.  Rock solid both in acro and level, even with out of balance props.   So I very strongly suspect something fundamental wrong in CF for acc filtering as the root problem. 

But what next???     Nick, thanks for letting me know what those acc filter settings do, I will see if playing with them can make a difference.  This was with acc_pf_filter = 0.

PS there is one other thing, the gyro_cmpf_factor.  Could it be that 600 is so high that the Acc data is somehow mixed badly with gyro data????

Thanks again you guys are amazing.

thenickdude, 2015-05-08:
A warpquad is not really a good comparison because it is carbon fibre while this frame looks to be wobbly plastic. The vibration characteristics will be completely different. 

The smoothing is just a weighted sum, so there's very little to go wrong with it. 

gyro_cmpf_factor is pretty critical to attitude estimation, but nothing will save the system if the accelerometer is shaking that much. Garbage in, garbage out, it needs useful readings to work with. No amount of setting change can massage that data, you need physical smoothing as Ledvinap says.

ctzsnooze, 2015-05-08:
This frame is not 'floppy plastic'.  It is within 20% of the stiffness of my WarpQuad 200.  I built it specifically with stiffness as a requirement.  It is a 4.5mm polycarbonate / 0.5mm carbon fibre laminate with 12-15mm wide arms, whereas the WQ is a 3mm thick 7-9mm wide CF arm.  The physics of the materials equated to nearly the same stiffness.  When I try to bend it the deformation for any given load is nearly the same as the WQ.  There is very little difference.  Polycarbonate is less springy and if anything would resonate less.

In addition to the WQ200 I have an AlienWii and a MWC board on two 120mm brushed quads both running MW2.3.  They are on very thin 3mm polycarbonate frames from picnic quads.  These frames vibrate like crazy - but stay completely level.  I also have a MW 2.3 Scarab Reconn and a Knife and they can vibrate badly at times yet stay perfectly level.  I have, until now, NEVER experienced an issue like this, and I've had lots of time in level modes in quads with vibration issues.

So there is a very fundamental difference between the way Acc vibration is handled in CF than in MultiWii.  I don't know why that is, but that's the root cause of this issue, I'll bet on it.

Anyhow I repeated the same test with acc_lpf_factor at 250 (max), not 4 as in the previous video.

https://dl.dropboxusercontent.com/u/9505117/MotorTestWAccLPF250.mov

Here is a still frame:

[image]

As you can see, the filtering shows no 'vibration' anymore, but what's left are the offsets, big offsets, easily enough to cause the problems I and others are seeing.  These are falsely indicating angulation of the FC despite the true angle of the FC held exactly constant the whole time during the test.  Errors like that would clearly cause the problems I and others are experiencing.

I'd be grateful to know if anyone else can replicate these findings with your quads on your desk.  It only takes a moment.

ledvinap, 2015-05-08:
@ctzsnooze : sampling is probably main problem in your case, not filtering. Mechanical filter should remove frequencies over 500Hz and keep frequencies under about 50Hz,  the design is probably not that critical. Double sided tape seems to work fine.

ctzsnooze, 2015-05-08:
Thanks Ledvinap.  I think I understand what you are saying.

I quickly unbolted the whole FC / PDB assembly from the frame.  Simply resting that on the frame by the legs of the LEDs at each corner reduced vibration pickup a lot, and the aliasing offset is zero.  So definitely a mechanical filter looks like it will fix the problem.

I looked up the 6050 - it has no internal digital LPF for the Acc (it does for the gyro).  So there's no way for it to internally filter out high frequency noise, any that gets in becomes aliased and becomes output as the the difference between the noise and the sampling frequency, if the noise and the sampling frequency are the same, bingo, you get a DC offset.  So the problem I and others have is that very high frequency noise at values close to or equal to the internal sampling frequency of the Acc gets aliased out as DC offset.  Meaningfully lower frequency noise - even if there is a lot of it - will not cause DC offset.  So the mechanical mounting must primarily block high frequency noise at all costs.

These frequencies are so high as to not even be felt.  Hence I didn't detect them.  Most likely what I see in Configurator and the logs as 'noise'  is actually the aliasing artefacts themselves - the actual high frequencies themselves cannot be seen, they are too fast. 

Once sampled as offsets, and because the ACC hpf must be zero, they cannot be filtered out.

Now to practicalities.  

OneShot, 800Hz loop times, fast revving motors, small props, ultralight frames, BLHeli motor timing noise, etc etc are all going to generate these frequencies.  So the FC must 'float' above a suitable damper or Acc won't work.  Hmm.   In this case, the AfroMini is such a small FC that even the wires joining it are relatively stiff!  Guys with Naze 32 and other similar hard mounted boards will have problems for sure.

It would be easier if the Acc sensor itself was nicely mechanically isolated on the FC board!  :-)  Maybe we can hope for boards with such construction.

Almost makes me want to put HF isolation in the motor mounts  :-)  

I will rebuild the mounting between FC and the rest of the quad and let you know how that goes.

I'm sure it will fix the problem entirely.

ctzsnooze, 2015-05-08:
Success!  

Mounting the FC on gel servo tape (3 x 1mm layers) and using only soft wires to connect it to the frame completely fixed my level mode driving issues.

Big THANK YOU to nickthedude and levdinap for putting me on the right path.

It now it hovers reliably and level in Horizon and Acc mode without any offset on the same settings as before.  In fact now I have all my 'fast' settings (low loop time, high gyro_lpf, etc) back on, and it is perfect.

Clearly the issue was due to:

1.  Very high frequency vibrations, can't be seen and barely felt, generated by the motors due to fast BLHeli ESCs (stuttering a tiny bit perhaps), OneShot, active braking, low loop time, high gyro_lpf, high kV motors, small low-inertia props, rigid but very light frame, etc

2.  Hard mounting of the FC/Acc to the frame allowing these high frequencies to get to the Acc.

3.  Digital aliasing of the incoming motor noise at frequencies close to or identical to the sampling frequency of the accelerometer, causing it to output DC offsets and other much lower intermodulation frequencies (difference between sampling frequency and noise frequency) to the FC.

4.  FC and software then simply gets 'wrong' incoming data, telling it that the quad has an offset from the horizontal, and reacts accordingly.  

Nothing wrong with CleanFlight at all.  

With modern hardware, and small fast quads, the only solution is mechanical isolation of the Acc, i.e. the whole FC board, from high frequencies in the frame.  

Thanks again...  now to have some fun!

Chris

thenickdude, 2015-05-08:
>So there is a very fundamental difference between the way Acc vibration is handled in CF than in MultiWii. I don't know why that is, but that's the root cause of this issue, I'll bet on it.

MultiWii and Cleanflight use the same acc filtering strategy (weighted sum). If you ignore rounding (use floating point variables), their results are precisely the same for the same weight, except that on MultiWii the output on accSmooth lags the input by one loop iteration due to the way MultiWii orders their computation, which is quite undesirable. Here's an Excel "simulation" of that algorithm:

[image]

The input is accADC and the output is accSmooth. Now if you change these variables to their actual types, rounding must be introduced. On MultiWii, all the variables are integers and the rounding is truncation (rounding towards zero). On Cleanflight, the accLPF is floating point so there's no rounding there, while the accSmooth variable is an integer which is simply accLPF rounded towards zero:

[image]

So it looks like the only differences in acc smoothing between MultiWii and CF are due to rounding, and it doesn't look like they differ in their final answer by more than +-1 unit. However, **note that the default value for the scale factor in MultiWii is 1/16, while in Cleanflight it's 1/4.** If you want the same smoothing as MultiWii (except for the +-1 rounding) you need to set `acc_lpf_factor` to 16. Here's the Google Doc version of the spreadsheet:

https://docs.google.com/spreadsheets/d/1GihNoY6b6g9ZCPAhGTV9aDJsvt-ijrJjLsEKG2PiWrc/edit?usp=sharing

Obviously no amount of smoothing will get rid of the DC offset you're experiencing though, so your MultiWii/CF differences are likely due to different board and craft construction transmitting different vibrations.

I haven't examined the use of GYR_CMPF_FACTOR which would be the other place where a difference could arise.

Trussroads, 2015-05-08:
Hi gents. Firstly thanks for your continued development on the software. It's great to see it forging on! Back to topic and my two cents.. Everyone I know has been isolating for this for some time. So from an outsiders point of view I don't believe this will be any serious issue for development going forward because I don't know anyone that doesn't run Kyosho-Gel (Zeal I think?) or something very similar on mounts for exactly this reason. Actually I think it may have been first highlighted by the 3DR crew on setting up their boards wasn't it? and with full autonomous flight the ramifications with dirty input signal was... well, interesting. Generally the best set-ups allow for low and high frequency and as the boards usually have very little weight it can be difficult. However, there are several products that do a really decent job such as the one i mentioned above, or the foam type product from 3DR. In fact they have a whole section in their text about how to log for this and what the limits should be measured at a maximum for safe flight etc. There are of course a few products for isolation I left out, but thought it worth giving it a really short mention here after reading these posts. Best of luck all and thanks again for your dev work. cheers.

ctzsnooze, 2015-05-09:
Strangely though despite that knowledge, most Acro FCs bolt directly to the frame.  Gel isn't used much.  Maybe because few Acro pilots bother with Acc mode, and they fly fine in Acro mode.

thenickdude, 2015-05-09:
Mine's bolted to the frame and that was the super smooth acc log I posted
earlier.

ctzsnooze, 2015-05-09:
Thanks again Nick for your follow-up on this, and for checking on the rounding.  

Interestingly, late last night, and purely empirically, I tried different values for acc_lpf_factor.  It seemed that 12-16 seemed to be the best compromise between responsiveness and visible smoothing, so that's what I ended locking in. I had no idea at the time that this would be the equivalent of the default MultiWii smoothing!  

I don't know if 16 vs 4 in CF will make much difference - maybe a bit smoother; seemed responsive enough.  Many MW quads are running loop times 3 times slower than what I now run, does that come into it?  Maybe at 3x faster loop time that smoothing of 4 translates into 1.3 in actual 'time constant'.  In which case possibly and even higher value for acc_lpf_factor might better translate to the classic MW feel in Level modes. 

Out of curiosity, do you know why, historically, CF uses 4 vs 16 as the default for acc_lpf_factor?

Finally I'm amazed that you could have such a super smooth Acc log with the FC being bolted to the frame.  Could I ask what the motor / esc / frame / AUW setup is of such a smooth quad?  

I've sort of moved to sn20a ESCs with BLHeli from Kiss.  I'm suspicious that most of the noise comes from non-smooth ESC operation, coupled with ultralight short arms etc etc etc.

Not so long ago I did a fair but of thrust testing.  Kiss ESCs ran super smooth, with a silky sort of feel.  In contrast BLHeli 13.2 on sn20a's was definitely not so smooth, even on settings that seemed 'best'.  I'm current running low demag, medium timing and damped light.  With the CF motors tab and very slow steps up, there are obvious audible tone changes at certain motor drive points, and with the noise change abrupt onset vibrations.  It can be clearly seen in the video.  It's not like prop out of balance stuff where there is gradual onset and dynamic balancing can help.  It's abrupt on and off and very RPM specific.  

I'd have stuck with the Kiss ESCs but for the cost and that several smoked randomly on power up and took a long time to get replacements. 

My earlier thrust testing was with a simple PPM tester.  I think I'll now rig up a better tester, with an ESC and an FC on a longish arm, and set it mechanically with some ability for the arm to vibrate, so that the FC can detect the vibration readily.  Then I can absolutely find the smoothest settings for any particular motor / ESC / prop / CleanFlight config combination.  It will take some time, but will I think be worth it if I can end up with a trace like yours.

The most brilliant thing about CleanFlight is the wonderful tools that people like you have made.  Without them we are in the dark - yet now it's possible to really optimise things very, very effectively.

Again, Thank you!

thenickdude, 2015-05-09:
The old code did a shift by 4 (equivalent to a division by 16) and the new code does a division by 4. It could have been an intentional choice to change the default smoothing but it's more likely to be a simple misunderstanding which kept the number 4 the same without regard to the change in meaning. 

Yeah, the time window effect of the smoothing depends on looptime. Faster looptimes will produce smaller smoothing windows for the same smoothing factor. 

My craft is an Armattan CF226, Cobra 2204 1960Kv, Kiss 18A ESC, 4S battery, Naze32 full.

ctzsnooze, 2015-05-09:
Cool - your build sounds similar to my WQ270 (though I use a lighter 3S battery with SS 2204 2300s - it comes in around 360g AUW).  It has a hard mounted FCB running MultiWii and has no Horizon issues.

The little quad that had problems is only 140g AUW and there's heaps of thrust.  It screams around the sky just like the WarpQuads, but rotates much faster and is much more reactive.  Like a goKart compared to a street car almost.  Requires huge concentration.  Having much less linear and rotational mass/inertia... including props... is probably why the frame is so sensitive to the slightest non-smoothness of the motors.  Interesting.

Will use 16 or so for acc smoothing, think it should be at least that value.

Am posting this outcome to others with similar issues, hopefully they will get sorted too.  

Now it flies properly in Horizon, I really appreciate some of the CleanFlight Horizon implementation details in PIDC2.

I really like the ability to modify the stick angle at which it transitions to Acro.  If the stick angle at which the quad does nice big power arcs is close to the transition point, control becomes very difficult.  Many people hate Horizon because of this.  I did.  It's because around that point the proportionality between stick angle and craft angle becomes unpredictable - a fraction more stick and acro predominates, starting  the quad rotating to greater and greater angles, and if you then go to fraction less stick, suddenly the levelling predominates and it realises it's now way past the intended angle and pulls back hard.  

In cleanflight it should be possible for a person to tune the crossover point so as to retain stick-to-angle proportionality even for very fast forward flight as in big power horizontal circles etc.

I made a horizon mode hack in multiwii that additionally attenuated levelling based on angle of the quad, so that when level it did nothing, but when fully inverted with centred sticks there was no levelling at all.  I then halved the stick-angle attenuation so that there was some stick input all the time.  This combination allowed some stick-proportional levelling to be retained nearly all the time - it was great!  It levelled 'normally' around mid-sticks (felt just like normal Horizon) but also gave amazing predictable flight control in high-angle fast forward fight.  Past vertical it would turn fast but still be stick controllable in the classical Horizon way even if your sticks were not hard out.  This allowed all kinds of near vertical or near past vertical flips loops and even inverted horizon control style flying, but retained levelling in emergencies or for pottering around.  Everyone who's used it loves it.  

I've made that code change in PIDC3 in CleanFlight - I hope I can add it to PIDC2 because it would be an even better fit there.

I only use Horizon in extremely small spaces LOS (e.g. inside bedrooms etc) or for indoor FPV.  The CF implementation where I can tune the degree of sort of Acro behaviour is great.

Once again a big thank you.  CleanFlight is amazing, because of people like you!

I've learned a lot from this!   :-)

ctzsnooze, 2015-05-11:
*Final* followup on this topic.  Gel mounting the FC made a huge difference, but I wanted to find where the noise was coming from and what else could be done.  Also wanted to see how a Kiss ESC went.  That's the fun bit!  = check the video at the end.

After extensive testing, I have some feedback on all kinds of software settings and their influence on acc aliasing.  I think that there may be one factor within CleanFlight that might be worth looking into - pls read on and let me know what you think.

The test rig was an AfroMini FC fairly firmly mounted with gel servo mount tape to the arm of my thrust testing rig, and used the motors tab to drive the test motor/esc.  The frequency response was deliberately higher than on my quad with only trivial high frequency damping.  This setup allowed me to detect acc noise in the CF motors tab, including aliasing artefacts, at different throttle values, and to precisely determine the influence of any of the myriad possible ESC or FC settings.

It was very interesting.

I tested all kinds of BLHeli 13.2 ESC settings.  If OneShot and Damped Light were enabled, for these little 1306 3100 motors with 4045s, and from an Acc aliasing noise perspective, DeMag off and medium or medium-high timing worked best.  

Disabling OneShot (disabling in Cleanflight) or disabling Damped Light (setting highPWM in the ESC instead) did not solve the problem.  It definitely shifted the throttle value at which the Acc aliasing occurred, but basically it happened in all modes, just to different amounts in different places.  Some users may find they have better results with OneShot on vs off, but that most likely is only because that may change the throttle value at which significant aliasing happens, and depending on their hover throttle value it will be more or less of an issue.  There is no specific badness about OneShot or Damped Light, that's for sure.

With OneShot and Damped Light on, and with a ESC cal range of 1000-2000, there were significant Acc aliasing points at 1340 and 1500 in the Z axis and around the 1700 for x/y.  The aliasing was abrupt in onset and offset over a very tight PPM range, i.e. on and off within 10-15 ppm - most of the time there was nothing at all, then suddenly there would be a big offset in one axis.  Clearly it arises from a non-smooth ESC drive issue.  BLHeli has a lot to answer for!  :-)

Having exhausted all ESC related issues, and only making a small improvement, but being absolutely sure the issue is fundamentally related to non-smooth ESC performance, I wondered about CleanFlight options... loop time, gyro_lpf, acc_lpf_factor, gyro_cmpf_factor... and was able to make some progress here.

Loop times between a silly low value (500) and a range of prime numbers up to 10,000 had no effect at all - worth knowing.  Likewise gyro_lpf 42 vs 256 no effect so I left it at 256.

I then tested 'real world' time constants for acc_lpf_factor.  I generated a very abrupt 0.5g deceleration and watched for overshoot etc in the acc trend.  A value of 64 gave optimal damping but still very fast response, so I used that.  It helped greatly at smoothing unwanted high frequency noise from the Acc trace - that was a big plus - the offsets were still there when the aliasing occurred, but overall it looked much smoother and the acc showed a nearly dead straight line except for aliasing points.  Very good.

Finally I wondered about gyro_cmpf_factor - could the aliasing be arising because of the mixing of gyro with acc data within cleanflight itself?  What if the gyro and acc sample rates happened to be coincident at some points?  That's beyond me as I don't know the details of the calculation of gyro_cmpf_factor. 

Theory aside, there was value in changing gyro_cmpf_factor!

A gyro_cmpf_factor value of 100 was definitely not as good as 1000.  Much worse overall at 100, and noticeably better at 1000 than 600!  :-)  

I figured that this result was because the gyro's were less affected than Acc by motor noise.  So I left it on 1000.  It actually made quite a difference.  But it is also possible that the mixing of gyro and acc data could somehow be contributing.  

If gyro_cmpf_factor could go higher than 1000, it might be even better!   I found a post where someone said their quad flew better like this so I thought, why not?  :-)

Test flying with gyro_cmpf_factor = 1000 and acc_lpf_factor = 64 seems very good.

I would be grateful if I could test more extreme values of gyro_cmpf_factor.  That may help discover if the best setting lies outside the current limitations.  A lot could be learned if the adjustment range went from 'uses entirely acc' (no gyro mixing at all) to a value higher than 1000 for the 'use mostly gyros' end of the scale.

Here's the 'before' video:  https://dl.dropboxusercontent.com/u/9505117/Motor1.mov

Here's an 'after' video showing the 'typical for an sn20a'  Z axis aliasing noise and with only a bit of residual aliasing in the Y axis around 1550-1650 - much better than previously:

https://dl.dropboxusercontent.com/u/9505117/OneShot_DL_DemagOff_MediumMT_CFopt.mov

FINALLY here's the fun bit... what about trying a Kiss 18a ESC?  I had one around, v1 type.  The good news - its way, way smoother, sounds smoother - no whistling - almost zero acc noise with the above settings until about 1600.  Even then, not too bad.  BUT, all of a sudden there was a desync at 3/4 throttle and out came the Magic Smoke!  Dead.  Compare this video to the others (until it died) - it has a totally different sound, much, much smoother:

https://dl.dropboxusercontent.com/u/9505117/Kiss_smoked.mov

Not much good if they smoke themselves though.

hydra, 2015-05-15:
@ctzsnooze can you please summarize if anything needs to be done for this issue or not?

thenickdude, 2015-05-15:
Software can't fix acc sensor aliasing, so there's nothing to be done on the CF end, the solution is to use vibration absorbing mounting for the FC.

ctzsnooze, 2015-05-17:
Hydra - 

Even though Nick is right, the problem is fundamentally the low sample rate of the hardware Acc, causing DC offset aliasing with vibrations at the acc sample rate.  Solution requires hardware vibration absorbtion for the FC.

BUT there are some software issues that arise, and I'd think it worthwhile to consider them.  Here is the summary, with detailed explanations below

1.  acc_lpf_factor should default to 32 or something like that, not 4.

2.  it would be great to be able to change acc_lpf_factor in the motors tab of the configuration with save option (and likewise for gyro_lpf)  

3  it would be great to see graphical gyro data in motors tab either as well as or instead of acc data.

reasons:

1. default acc_lpf factor:

 Nick noticed that acc_lpf_factor in multiwii defaults to 2^4 (i.e. 16) but in cleanflight it is 4.  Also it relates to loop time, and often in CF we are running loop times twice as fast as multiwii.  So a time-adjusted default acc factor equivalent to multiwii would be 32.  AND.... in my testing I found much better smoothing with 32 or 64, in the end I used 64.  AND.... the sample rate of the Acc is 256.  With a fast loop time of 1000, the acc data does not change for 4 loops, so a smoothing for 4 loops kind of has no effect.  There will be problematic aliasing at 1/2, 1/3 and 1/4 of the acc sample rate that can be filtered only if we run acc lpf two times lower than the lowest.  I found 64 dealt with most 'false' aliasing errors but retained more than adequate response time for real movements that the Acc needs to deal with.  Finally, the acc is not intended to be super fast.  If there are say 640 cycles per second of CPU then if we filter 64 then we are still having time constant around 1/10th second which is fast enough for most acc related responses (certainly flies well like this).  So a good compromise default value would be 32, I personally think that would be best.

2 : It would be great to be able to change acc_pf_factor in the motors tab of the configurator... this would be helpful because what you see in the acc graph depends on acc lpf.  If one is trying to tune out acc noise by varying acc lpf factor, or if you want to see the 'raw' acc signal, you need to change acc lpf factor.  It would be quicker and easier if it was possible to do so on the motors tab itself rather than going to CLI and rebooting.

3.  Likewise if we use motors tab to assist with vibration identification or prop balancing, which it's really great for, it could be very useful to see gyro data (ideally unfiltered gyro data).  So if in the configurator we could choose to see gyro vs acc (or both) as waveforms, that could be great.... and likewise if we could tune the gyro_lpf there also (with a save button) so we don't have to go to CLI, that makes sense also.

Finally:  a cool configurator idea, maybe too much of a problem.... I wondered if it would be possible to do a 'realtime' fast fourier transform of the gyro data in the motors tab, showing the amplitude vertically vs frequency horizontally of all ac components to 256 Hz - how cool would that be for identifying noise, its cause, and whether or not you fixed it....  I use this kind of FFT display at work to display brain waves it's the *only* way to see what's going on...  :-)

Thanks!

C

ledvinap, 2015-05-17:
just some thoughts:
- acc is IMO sampled at 1kHz
- too high acc_lpf_factor may delay the signal too much, degrading performance (but this is just guess ... there may be problems in filter delay is comparable to quad natural frequency)
- gyro_cmpf_factor is main filtering constant for ACC data (acc signal is rotated by gyro signal and then filtered). There is not much effect of acc_lpf_factor, but it may be useful to remove high frequency noise from acc data (noise shaping).

- FFT may be possible, but not much useful unless done on 1kHz. Code needs some changes to make 
1kHz acc + gyro sampling practical (see my local branch, it does do that)

ctzsnooze, 2015-05-18:
Hi levidnap

What you say goes against what I saw, nearly totally.

I looked at sensors.p but had trouble being certain of what it was instructing the 6050 to do for acc sample rate.  I'm fairly sure it was to 256.  I looked up the 6050 spec sheet.  I do know, for sure, that there is no hardware filtering on the 6050 for acc.  There is hardware filtering in the 6050 for gyro.  The gyro on the other hand is sampled at very high frequency and smoothed in hardware in the 6050 at the specified gyro_lpf before going to the FC.

My experience was that there was a huge influence acc_lpf_factor on what I saw in the motors tab acc graphs.  This would be easy to demonstrate for your self, put props on your quad, hold it down, hook up to motors tab, set acc_pf_factor to 1, spin one motor, observe vibrations and waveforms at different RPM, then set to 256 - totally different appearance, very smooth now.  Clearly post-filter data.  And that filtering can only be done in the FC. 

I did change gyro_lpf - 48 or 256 - just quickly - it had no effect at what I saw in motors tab acc data.

I am surprised to hear that acc and gyro are cross-filtered.  I had wondered if that was the cause of the problem.  If the acc is sampled at 256 and the gyro filter is set to 256 that may be partly why the acc data is aliasing in such strange ways.  

An FFT display in the motors tab of the configurator could be generated from gyro data, even if only up to 256 that would be very helpful anyway.

I flew level and horizon quite a bit with acc_pf_factor at 64 and loop time 1250 (approx).  It was fine like that.  Given that 32 is basically the same (at a loop time of 1750) as the original multiwii 2^4 i.e. 16 at 3500, and everyone was 'happy' with the multiwii default, and that it worked better for me empirically at 64 than 32, I think there are many good reasons to change acc_lpf_factor to 32 as the default.  It would be wrong to leave it as is.

MasterZap, 2015-05-18:
Isn't the acc_lpf_factor fundamentally flawed? And the gyro_cmpf_factor is what SHOULD be used?

Basically, the two things do the same thing, but one on the raw data (so too much filtering will kill any legitemate high-rate accelerometer changes), but the second one does it AFTER accelerometer data has been rotated by the gyro date.

To me, filtering the accelerometer vector AFTER gyro rotation makes sense; if both vectors move in tandem, that means your attitude change was legitemate and shouldn't be filtered out. So filtering against the "accelerometer rotated  by gyro" must be the right thing to do!

Also, the code includes accelerometer data within a band around 1G magnitude, and just tosses away data outside this band, with a hard edged stop. This feels wrong to me. I would have weighted accelerometer data relative to how close to magnitude 1.0 it was. Now, some of the strange "bucking" effects in angle and horizon mode can probably be attributed to the effect off accelerometer "snapping" in an out as we go inside/outside this band.

I'll tinker in my branch with fixes to this, and report back.

/Z

hydra, 2015-05-18:
Looking forwards to hearing about your results.

ctzsnooze, 2015-05-18:
One fundamental thing concerns me.  In Horizon or Level mode, the FC gets inputs from gyros via their PIDs and from the acc via its PID algorithms.  

I had always assumed that in principle the Acc data was independent of the gyro data, completely.  In other words, that you could filter the Acc data and manipulate it with independent PIDs (in multiwii at least), and the influence of the Acc would be independent of the gyro influence, maybe at times 'internally' fighting or being of opposite sign to the gyro influence, and of different time constants etc, but fundamentally independent.  So to me it made sense to have independent filtering arrangements for gyro and acc, so that one could optimise the validity of the signal before the FC had to deal with it.

Now if the Acc data is somehow mixed - rotated? - with some modification of the gyro data, and then sent to the FC, then the gyro data now has two input lines into the FC, one direct via gyro PIDs and the other via Acc PIDs and filters.  It seems likely that this will cause problems of some kind given the different time constants and gains for the same data.

It would be OK if both acc and gyro had the same frequency response at hardware level but they don't.  

I would prefer to keep my gyro frequency response at 256 (or as high as possible) so that in Acro I have the most responsive behaviour possible.

At the same time, given that when I (Rarely, but sometimes) fly Horizon or Level modes, I want Acc modes to be smooth and work well when active.  But the Acc has aliasing issues at frequencies that don't trouble the gyros at all.  So I may find it flies great in Acro with a high gyro_lpf, but needs a lower Acc filter value.  

So then I would want a *different* filter for Acc than for gyros, not the same value for both.

I found in multiwii that once gyro I was tuned properly, it was better to have zero level I.  Adding level I over good gyro I caused low frequency oscillation nearly all the time.  This makes sense because the gyros alone are very good at holding angle; the Acc P and D will get that angle to zero.  It seems bad to have two things fighting each other to do the same thing.

The point of this being that if we assume the gyros are doing the fast work of getting the quad in shape and keeping the angle where the FC (i.e. the sticks) want it, then the Acc input to the FC can be considered a secondary set of sticks, in a sense, independent of the gyros entirely - and therefore with independent time constants and behaviours, just as our stick inputs have all kinds of crazy independent behaviour, and it doesn't matter to the gyros at all.  That conceptual model makes the most sense to me.  

The time constants for the Acc to get the quad level don't need to be super fast, but the gyro time constants do.  Its definitely better to have gyro_lpf at 256 imho.  But the Acc is aliasing anything at 256 Hz, it is not a high frequency acc (I understand that they exist but the 6050 is not).  So at that sample rate it would be better to use less than 64Hz filtering on the Acc (64Hz would be an acc_pf_factor of 12 at 1250 loop time).  In practice filtering down to 25Hz (acc_lpf of 32 is only 1/2 the rate of stick inputs at 50Hz, still quite fast).  Even as low as 12Hz in my experience work fine.  I think it works because the gyros are doing the hard work of moving the quad to where it should be, not the Acc, and with smoothed acc data there is little tendency to oscillate as the quad follows the smooth incoming acc data.

What I see as being very problematic is mixing the gyro data into the acc signal and then re-inputting it, after different filtering, back into the FC.  That seems a recipe for problems. 

I am not a technical expert on this but whatever we do for now could we keep the filtering independent?  We can then test making the values the same, or different.

ctzsnooze, 2015-05-18:
PS thanks MasterZap I can see you understand the coding well - very much look forward to how you perceive the best way forward.  I currently use gyro_cmpf_factor = 1000 (max allowed) - it made no difference to acc aliasing issues but seemed sensible to use the gyros to allow faster 'virtual acc'  behaviour.  

I will be happy with whatever works best and is least likely to be internally problematic.

thenickdude, 2015-05-18:
The accelerometer is mixed with the gyros using a complementary filter. This allows the attitude to be estimated using the accelerometer to estimate low frequency, long term changes (i.e. to fix the gyro's long term drift issues), while the gyro's high frequency movements can take immediate effect on the attitude. The `gyro_cmpf_factor` setting tunes this mixing. If the gyro didn't have long-term drift from integrating rotations then we wouldn't even need an accelerometer at all, that's the only task that the acc needs to do.

ledvinap, 2015-05-18:
@ctzsnooze : see page 13 here  http://store.invensense.com/Datasheets/invensense/RM-MPU-6000A.pdf

Gyro is sampled at 8kHz, ACC at 1kHz and both can be filtered (LPF bandwidth is set to 42/44Hz by default, filter setting is common to both channels). This seems to be first order filter (from introduced delay).

Gyro (and acc) filtering is done in digital domain, 'hardware filtering' is a bit confusing .. 

Motor graph is indeed filtered by `acc_lpf`. But the signal is then filtered again by `gyro_cmpf_factor` (using gyro rotation signal to rotate filter history). 
This filter has much lower bandwidth and craft rotation influence is compensated for. (when filtering with acc_lpf, craft rotation is seen as disturbance). Resulting data are not visible in configurator, but it should be quire trivial to change.

My feeling for good value for acc_lpf is to get same bandwidth as filtering selected on mpu6050 - 42Hz by default, so 1000/(42*looptime) may be reasonable (roughly 8 - 16)

@MasterZap : Yes ...
In theory you can choose 'ideal' complimentary filter factor based on incoming data quality. Kalman filter is based on this principle (keeping track of variance of filtered data and assuming some input data variance. There are methods to estimate it).
It will need some experimenting. Flight controllers with full Kalman filter implementation are probably upper bound of possible gain.

MasterZap, 2015-05-18:
So in my thinking, what should happen is roughly this:

(Warning, barely coherent rambling, off the top of a fritzed brain while waiting for a compiler...)

- We should keep a running estimate of what we think is down (this exists already in EstG in imu.c)
- We should also keep the previous value read from the acc's, just the raw value

Then the logic should se something
- read the gyros
- rotate the estimated down vector based on the gyros
- check if:
-- acc value is different from the last acc value. If not, do nothing (we have no new data, change nothing)
-- compute a "usefulness value" from the accelerometer value is, by computing how close to "1.0" it is. Something like  0% at 0.5G, 100% at 1G, 0% at 1.5G (or whatever)
-- mix this new value weigted by the "usefulness value" with the existing "estimated down vector". 

Assuming ideal state and the copter not actually accelerating, the new accelerometer vector read should be close or identical to this rotated "old value"....  so if we filter the new data against this "rotated old data", filtering could in principle be EXTREMELY heavy. Because the only thing we are filtering out is the difference, yet we loose no nibleness.

But today, the pre-filtering on the accelerometer (acc_lpf_filter) causes the vectors movement to turn to mush BEFORE the rotation. So a fast move will not be reflected in the acc vector, making level modes slow to respond.

Secondly, the whole idea of having different PID's for level and gyro is insane.

Here's how it should work:

There is only one single attitude of the craft. Then, there is a *desired* attitude. 

In Acro mode, this desired attitude is directly affected by the sticks, and nothing else.

In LEVEL/Horizon, this desired attitude is computed differently (it involves which way is up)

However, HOW we arrive to that desired attiude is the same. We have a desired attitude, we have an actual attitude, between them is an error, and we apply a SINGLE PID controller to correct that error.

The difference in ACRO and LEVEL should not have anything to with PID's, it's just how the "desired attitude" is computed. In once case it's stick values, in the other case, it's stick relative to what is up.

The only thing you want is:
- an attitude estimate that takes all sensors into account
- a single PID controller that corrects between actual and desired attitude
- for level modes, a single "speed" setting (no PID nonsense) which is how quickly centering the stick actually takes you level. You kinda need this because otherwise, moving the stick super-quickly basically commands the craft to do completely instant attitude changes, which isn't possible due to inertia. So a minor smoothing of the "desired attitude" vector will calm down the error rates sent to the actual PID controllers.

But hey, that's just me thinking out loud.

/Z

ledvinap, 2015-05-18:
@MasterZap 
I mostly agree with first part:
- acc value change is expected on every loop iteration, no need to test that
- the distance from 1g downwards metric may be good one, but it should be analyzed. There may be better estimates of data quality (data variance may be one)

`acc_lpf_filter` may help if it is complementing/improving internal MPU filter (you will get second order response instead of internal first order). Motor noise may be in this band, so heavier filtering may help. Filter delay may is relatively low (additional 5ms for 42Hz).

gyro_cmpf_factor is 600 in current implementation, time constant is in order of seconds, so acc data filtering is not that important unless same bias is introduced.

As for the PID:
Main problem in your argument is that in ACRO mode, the stick sets desired rate of change (angular speed). 
Gyro is measuring this quantity directly, so there is no integration to introduce drift. And the system is (approximately) first order, making the PID much easier to tune.

In LEVEL mode the setpoint represents desired angle. The system is second order, measured rotation rate must be integrated ( => drift) and it is harder to control this system. For example it's impossible to get stable Level mode without using D term (derivative of angle -> it is P term in Gyro mode).

ctzsnooze, 2015-05-18:
Hi - I have some clarification on the 6050:
>The DLPF_CFG parameter sets the digital low pass filter configuration. It also determines the internal sampling rate used by the device as shown in the table below.
Note: The accelerometer output rate is 1kHz. This means that for a Sample Rate greater than 1kHz, the same accelerometer sample may be output to the FIFO, DMP, and sensor registers more than once.

          |   ACCELEROMETER    |           GYROSCOPE
 DLPF_CFG | Bandwidth | Delay  | Bandwidth | Delay  | Sample Rate
 ---------+-----------+--------+-----------+--------+-------------
 0        | 260Hz     | 0ms    | 256Hz     | 0.98ms | 8kHz
 1        | 184Hz     | 2.0ms  | 188Hz     | 1.9ms  | 1kHz
 2        | 94Hz      | 3.0ms  | 98Hz      | 2.8ms  | 1kHz
 3        | 44Hz      | 4.9ms  | 42Hz      | 4.8ms  | 1kHz
 4        | 21Hz      | 8.5ms  | 20Hz      | 8.3ms  | 1kHz
 5        | 10Hz      | 13.8ms | 10Hz      | 13.4ms | 1kHz
 6        | 5Hz       | 19.0ms | 5Hz       | 18.6ms | 1kHz
 7        |   -- Reserved --   |   -- Reserved --   | Reserved
 
Returns:
DLFP configuration

MasterZap, 2015-05-19:
That's an interesting point @ctzsnooze ,,, if we only care about acc in a narrow band around 1G, can the 6050 be set to a smaller G sampling range, to get more accuracy?

/Z

ctzsnooze, 2015-05-19:
Yep the 6050 can be narrowed down for greater sensitivity.  As far as I can tell needs a change in the bottom of drivers/accgyro_mpu6050.c  ... here:

    // ACC Init stuff. Moved into gyro init because the reset above would screw up accel config. Oops.
    // Accel scale 8g (4096 LSB/g)
    i2cWrite(MPU6050_ADDRESS, MPU_RA_ACCEL_CONFIG, INV_FSR_8G << 3);

I've been looking at IMU.c which seems to be the only place that acc_pf_factor or gyro_cmpf_factor get used, first acc_pf_factor:

        if (imuRuntimeConfig->acc_lpf_factor > 0) {
            accLPF[axis] = accLPF[axis] * (1.0f - (1.0f / imuRuntimeConfig->acc_lpf_factor)) + accADC[axis] * (1.0f / imuRuntimeConfig->acc_lpf_factor);
            accSmooth[axis] = accLPF[axis];
        } else {
            accSmooth[axis] = accADC[axis];
        }

Not sure exactly how the smoothing works because I don't know what accADC[axis] is, if you could explain it a bit that would be great!. :-)

and second gyro_cmpf_factor:

    // Apply complimentary filter (Gyro drift correction)
    // If accel magnitude >1.15G or <0.85G and ACC vector outside of the limit range => we neutralize the effect of accelerometers in the angle estimation.
    // To do that, we just skip filter, as EstV already rotated by Gyro

    float invGyroComplimentaryFilterFactor = (1.0f / (imuRuntimeConfig->gyro_cmpf_factor + 1.0f));

    if (72 < (uint16_t)accMag && (uint16_t)accMag < 133) {
        for (axis = 0; axis < 3; axis++)
            EstG.A[axis] = (EstG.A[axis] * imuRuntimeConfig->gyro_cmpf_factor + accSmooth[axis]) * invGyroComplimentaryFilterFactor;
    }

Here I can figure that if accSmooth[axis] holds a smoothed value for acc, then this is divided linearly by the value of the gyro_cmpf_factor, sort of.  So in my case where gyro_cmpf_factor was 1000, only 1/1000th of the smoothed acc data was getting into the EstG.A[axis] value per cycle.... which implies that any frequencies below 1Hz are being derived basically entirely from gyro, not acc data.  Am I interpreting this correctly?

Do you know what value is actually being reported in the motors tab of the CLI?  It would certainly be interesting in the motors tab to see the incoming acc and gyro data (filtered only in the 6050 itself as per the gyro_lpf as indicated above), and the derived acc and gyro data post filtering (the values that actually end up going into the FC PID algorithm).

Maybe also we need a flight controller attached to a vibration generator (loudspeaker with test tone input) to actually figure out what's going on... my motor generates vibration but isn't very controllable :-)

ledvinap, 2015-05-19:
@ctzsnooze : there are two sample rated in mpu6050 - internal sampling of sensors (1kHz for ACC, 8kHz for gyro) and sample rate  used for FIFO, motion engine etc. Second rate is configured according to DLPF_CFG and sample rate divider. But sensor data are downsampled to this rate, so only high-frequency components are lost.

The filtering is a bit more complicated - the filters used are first order, so signal at frequency that is double the cutoff is halved (lookup some theory, please). Cascading two filters (with same cutoff)  will reduce this frequency by four, so it is sometimes reasonable to use more filters.

Setting gyro LPF to higher frequency results in larger signal in frequency range that will be over Nyquist for gyro readout (1/looptime), so it makes aliasing problems worse.
You can read gyro data at 8kHz and filter them yourself, but that is impractical. Other option is to set LPF as low as practical (so that it does not affect frequencies you are interested in) and read on slower rate. (http://en.wikipedia.org/wiki/Decimation_%28signal_processing%29)

gyro_cmpf_factor is filter that affects leveling, acc_lpf is order of magnitude lower and has minor effect. Also leveling is mainly gyro based, these filter won't affect leveling speed (at least not directly).

Increasing acc resolution (changing measurement scale) won't help - noise is much larger that resolution, S/N ration will stay the same.

`accADC` is raw data from accelerometer. First order IIR filter is used to do the filtering : http://en.wikipedia.org/wiki/Low-pass_filter#Simple_infinite_impulse_response_filter.
Fc is approximately `1/(2*pi*looptime*acc_lpf)` (and cutoff frequency is 2*pi smaller that what I used before, sorry ... )

Configurator displays acc_smooth

The problem with high freuency in ACC may be caused by mechanical and other nonlinearities. Resulting signal has DC offset that gets through `gyro_cmpf_factor`

ctzsnooze, 2015-05-19:
Thanks for those clarifications, much appreciated.

I'm now very clear on the internal filtering of the mpu6050 and how it is affected by gyro_lpf.  I w

I run about 1250 loop times or about 800Hz, my Nyqvist is 400Hz.  What gyro_lpf would you recommend in that case?  To me it seems reasonable to be maybe 2 times slower than my Nyqvist frequency - say 188Hz?  Even with some a little loop time jitter I should get a good representation of the incoming data from both acc and gyro.  

Using your formula - thank you, I remember that stuff now! - for my 1250ms loop time, with acc_lpf_factor is 1, Fc = 1/(6*0.00125*1) = 1/.00725 or 133Hz. With acc_lpf of 4, Fc becomes 33Hz, at 16 becomes 8Hz.  So when I set 64 I was around 2Hz!!??

It is true that with acc_lpf_factor set to 64, my acc_smooth trace in motors was very smooth.  It looked like 2Hz.  But when I made an abrupt 0.5G deceleration, the frequency response was much faster than 2Hz - it was very satisfactory.  And when the DC offset appeared, there was noise well above 2Hz.  And I could fly the quad quite OK in level mode as well, it levelled out quite quickly in Horizon mode when needed, and flew quite nicely.  This still seems strange.

I'm guessing the gyro complementary factor calculation has something to do with it, and that is controlled by accMag.   accMag is the sum of the smoothed (by acc_lpf_factor) acc values (accSmooth) for each axis.  So accMag is 1 when the sum of the smoothed acc data indicates that the quad is not accelerating in any direction.  If acc_lpf_factor is 0, it is sum of the instantaneous acc values.  That bit I get.

If accMag is close to 1, the EstG.A value includes a proportion of the accSmooth values depending on gyro_cmpf_factor; I also get that bit. If accMag is not close to 1 (the quad is accelerating or decelerating, that EstG.A calculation is bypassed.  The code notes say  "If acc magnitude >1.15G or <0.85G... we neutralize the effect of accelerometers in the angle estimation... we just skip filter, as EstV already rotated by Gyro".  

But then I can't figure out what happen if accMag is not close to 1, i.e. they skip the smoothed acc data altogether an use Est.V.  EstG.A is never mentioned again.  I guess I don't understand the code at this point.  What is EstG.A every used for?  (certainly nothing in imu.c.   If the smoothed acc data is ignored, this preceding line calculates EstG.V:

rotateV(&EstG.V, &deltaGyroAngle);

I'm guessing that this is a gyro based surrogate for acc data.  I don't know how EstG.V is different from EstG.A or if it is smoothed in any way.

The primary filtering of the gyro data is at the mpu6050 at gyro_lpf rate.   Is that the only filtering applied to  'EstV'?  If so, whenever the quad is accelerating, acc data isn't used at all, and presumable EstV is used, and may oscillate or be subject to aliasing.

It is now apparent that the offset I saw in my acc_smooth motor tab data could not have come from the acc.  The accSmooth offset in the motors tab was more than 0.15G so the acc would have been 'neutralised' (ignored) as soon as that offset appeared.  It's hard to imagine significant DC aliasing occurring in the mpu6050 because it was sampling at 8kHz and almost certainly doing a very good job of filtering before downsampling to 256Hz.  While it's theoretically possibly the aliasing could have arisen due to aliasing by the STM as its 800Hz loop time was getting samples from the mpu6050 with my gyro_lpf of 256, but it should be OK, we are less than Nyqvist of 400Hz - and anyway I found it made no difference if I dropped gyro_lpf as low as 42 or increased loop time to 300Hz, the aliasing issue was unchanged.

I'm starting to understand the imu code a lot better.  It seems that the acc data gets very little mileage in imu.c, only when the quad is not accelerating (or perceived to be accelerating) does acc data get used at all.     

I sure don't like the abrupt step between using acc data and discarding it entirely, that seems like asking for trouble whenever accMag gets close to the on/off points.

At this point I'm not any closer to the ideal values for gyro_cmpf_factor though I'm now quite clear on gyro_lpf and acc_lpf_factor and what they do.  

Thanks for all your patience.

ledvinap, 2015-05-19:
@ctzsnooze 
as for internal MPU filter - using 188Hz means  that unwanted signal at 400Hz is approximately reduced to 50% (and 1kHz signal is reduced to 25%). 
Also MPU6050 is still updating data at 1kHz and CPU is reading them at 800Hz, this will probably affect signals around 200Hz.
So proper question is how low can be the filter set without compromising performance or what is best compromise. I don't know the answer ... 

The attitude estimation basically starts with vector pointing downwards.
On each loop this vector is rotated by gyro signal(in opposite direction of measured rotation), so if gyro data are bias and noise free, this vector will always point down, no matter how the quad is rotated.
Due to drift and noise, the vector will drift away eventually. 
To compensate it, it is assumed that gravity is pointing downwards;
and drone acceleration (that is summed  with gravity in accelerometer) is relatively small when heavily filtered (this is direct consequence of limited drone speed - acceleration can't point in one direction for long).
This measured signal is very slowly mixed with 'down' vector. And when acceleration is large, it is ignored altogether. This mix does compensate gyro inaccuracies...

ctzsnooze, 2015-05-20:
Great info!

I think 98Hz is sounding a pretty good compromise. My 'problem' FC is now on such soft gel that I'm sure it has an Fc below that anyway!  :-)  But others are still hard mounted.

Thanks for explaining the attitude estimation concept. I now understand that the acc is really only fine-tuning the estimation of 'what the level vector is', which is actually maintained moment to moment from the gyros.  Therefore the 'angle of the quad' is in relation to the 'estimate of the level vector' and is primarily determined moment to moment from gyro data, not acc data.  That explains why the 'level vector' estimation process ignores high estimated acceleration values - not much point fine-tuning 'level' while the quad is accelerating hard.  

On that basis one could smooth the acc data quite a lot and not make much difference to handling at all (which is exactly what I found in practice).

It all makes sense!

I need to experiment, though, with some known noise inputs and test the behaviour of the system.

PS - This makes me realise that someone really needs do proper performance testing of FC mounting methods.  i.e. to find out how effective those O-ring methods or gel mounts really are, what their Fc value is.  One could put a speaker on a motor thrust test arm - or on the quad itself - and see what the FC receives from 0 to 1hKz.  

Of course to do this we need to see 1kHz quad or acc data somehow.  That's why it would be great to have a realtime FFT in the configurator with response up to 1kHz (or maybe 500Hz can't recall FFT maths properly.    :-)

One could also test mounting methods using an audio amp, a piezo transducer attached to a dummy FC and a speaker on the frame....  :-)

Thanks again,

Chris

ctzsnooze, 2015-05-20:
Actually there is one good reason to leave gyro_lpf at 256, that being the delay gets appreciable - 3ms if set to 98Hz:

[image]

MasterZap, 2015-05-20:
Isn't the default 42 hz??

/Z

MasterZap, 2015-05-20:
I'm still thinking the acc_lpf_factor hurts more than it helps. It makes the acc vector sluggish vs. reality. THEN i filters this sluggish vector, heavier against the estimated vector (which is just the old vector, rotated by gyros).

See it like this: 

CASE #1: If everythign is ideal (in a magical world), and no filters exist:

- The acc vector is perfect.
- The estimated vector gets rotated by the gyros
- If we compare the acc vector and the rotated estimated vector, they should be identical (because this is the ideal world :)) 

So even if we add a filter to step 3, it doesn't matter how much delay or slowness that filter introduces, because in the ideal world, the two vectors are identical the whole time... so we are just filtering a difference of zero, so nothing will happen.

CASE #2: Not take a slightly less ideal world: Perfect acc, but with gyro drift.

- the acc vector is perfect
- the estimated vector gets rotated by gyros. This is *slightly* of (a tiny hair)
- If we compare the acc vector and the rotated estimated vector, they are almost identical but not quite - there will be a tiny drift.

If we now filter in step #3, we are only removing a very tiny deviation, just the drift. It will be fine.

CASE #3: Now imagine the original "ideal world" (where everything is perfect) but we filter *before* the rotation:

- the acc vector is perfect
- ...but we smooth it, so we get a sluggish, delayed vector, that doesn't actually follow the real acc vector
- the estimated vector gets rotated by gyros. The gyro rotation is perfect.
- ...but since we are now comparing this against a sluggish smoothed vector dragging behind reality, we will have differences... the faster rotation, the bigger the difference!

By filtering first we made things WORSE... everything  will break, necessitating a 2nd filter, which kinda put a band-aid on things, but doesn't truly fix it back to the ideal case. There is no way we can get back to perfect case #1, no matter how much filtering we add. The fact that we have a deviation between "filtered" and "estimated" introduces wobbles and problems.

CASE #4: Less ideal world: acc noise, but no gyro drift

- the acc vector is "okay but noisy"
- the estimated vector gets rotated by gyros. 
- the estimated vector will be off from the acc vector (but only due to the noise)
- due to the noise, the acc vector will jump around, but around a "real" center. That center should coincide with the estimated vector.
- with heavy enough filtering against the estimated vector, the post-rotation smoothed acc vector should be identical to the estimate

This would be IMPOSSIBLE with the pre-filted acc-vector (case #3)

CASE #5: Real world, but no-pre-filtering. Acc-noise and drift

- the acc vector is "okay but noisy"
- the estimated vector gets rotated by gyros (but adds tiny drift). 
- the estimated vector will be off from the acc vector (since there is noise, and drift)
- the noisy vector is filtered against the estimate
- the estimate is updated (to correct drift)
- we now have a good eastimate, with very little lag. EVEN if we filter heavily and there IS lag, it only slows down the drift-correction a tiny bit. For the most part, things will be correct... 

CASE #6: Real world, with Cleanflight code (pre AND post filtering)

This is disastrous, here's what happens:

- the acc vector is "okay but noisy"
- we smooth it into a mushy, delayed mess
- the estimated vector gets rotated by gyros (but adds tiny drift). 
- the estimated vector will be off from the acc vector a LOT.. not due to noise, but due to being a delayed mushy mess
- the noisy vector is filtered against the estimate. This filter won't deal with noise, but will deal with the ugly offsets introduced by the first filter. It will try to fight the error introduced by the first filter by covering it up with another error
- the estimate is updated incorrectly (beacuse it is based on a skewed, delayed vector)
- we now have a HORRIBLE eastimate, based on a laggy vector. EVEN if we filter heavily it doesn't help, because we are literally comparing to the wrong thing. 

There is NO SET of filter values that will get us the right thing.

Need I reiterate I think we should be doing #5, but we are doing #6?

Basically, acc_lpf_factor should ALWAYS be zero, and gyro_cmpf_factor should deal with the noise.

/Z

ctzsnooze, 2015-05-21:
Great analysis!  Thank you.  I hadn't thought about the issue of getting the difference between an acc value filtered by acc_lpf_factor and the unfiltered gyro data.  Yes!  Comparing filtered with unfiltered wouldn't be good at all!  

So.... let's say we forget about the hardware filtering, and assume that for practical purposes the delays and filter effects on both the acc and gyro signals are nearly equal.  There will be noise (some) in both acc and gyro signals.  With luck the noise vectors will be very similar over time in both acc and rotated gyro values.  There may be more high frequency noise in the gyro signal above 1kHz, assuming gyro_lpf is 256 and the gyro is being sampled at 8kHz, but lets for now ignore this (I think we can).

Let's then, maybe, rotate the realtime incoming (unfiltered by FC) estimated gyro vector, and calculate the difference between it and the realtime incoming (unfiltered by FC) acc vector.  (This is what happens in the code right now if acc_lpf is 0, I think).  

Ideally that difference should be zero, but in reality, as you say, there will be some random noise component.  

If we were to smooth that difference, to remove the noise component, any DC offset indicates 'drift' of the gyro-estimated attitude value, and can then be corrected.

it wouldn't matter much if such a smoothing factor was long or short; so long as it was significantly longer than the noise component, and shorter than the drift rate by say 2-4 times.  A value of 1000 would be 4 times longer than noise components arriving at the fastest possible gyro_cal rate, given their filtering out above 256Hz, but would probably still be quick enough given the normally very slow rate of gyro drift.  And by filtering the correction value so smoothly, the estimated attitude value would be as stable as the actual gyro data itself, and as responsive.

If for some reason we actually wanted a 'smoother' attitude vector signal, I guess we could smooth the resulting drift-correct estimated attitude vector to some user-controlled extent.  However I wonder if in practice that should ever be needed.  If there was noise in the attitude signal, it would be the same as that in the gyro signal, and could be dealt with physically or by lowering gyro_lpf.  If we wanted to control the time-constant of the return to level following a perturbation in level or Horizon modes (Multiwii2.3 level D value) there are better ways of achieving that kind of thing.  

So potentially there would be two or maybe three kinds of adjustment for attitude filtering/smoothing:

1.  gyro_lpf, to set hardware filtering in the mpu6050 for both acc and gyro
2.  a value smoothing out the per-axis difference between realtime accADC and rotated gyro vector EstG
3.  (optional, probably unnecessary) a value smoothing out of the resulting estimated attitude

It would be great to try a re-build of the IMU.c code, re-using acc_lpf_factor to achieve the smoothing in (2) above?

If someone could code that I can test it.

This is quite exciting!

Chris.

ctzsnooze, 2015-05-21:
I've now been looking at the code carefully... and what you say again makes absolute sense.

If all I do is set acc_lpf_factor to 0, accSmooth will contain the instant acc value, and accMag the instant absolute acceleration value.  The gyro_cmpf_factor code will then only be active when the instant accMag value, not the smoothed acc value, is close to 1.  And if gyro_cmpf_factor is say 1000, it will be doing the smoothing of any noise in the difference between the two.

So yep I agree, just set acc_lpf_factor to 0 and gyro_cmpf_factor as high as possible, and use gyro_lpf or other means to block noise in the incoming data.

I will try this asap.

MasterZap, 2015-05-21:
YES, I agree with you both 2000%, and I will try this the minute I get time to fly. Just testing acc_lpf_factor=0 and gyro_cmpf_factor=1000 is obviously trivial, and I will do so as fast as I can.

As for changing the code, a couple of changes come to mind:

#1:  Weighting the vector based on "closeness to G=1.0" (or some other heuristic... I have some thoughts about this, see below)... right now it's a hard step function within a range
#2: Lower G sensitivity in the 6050... if we only care about a band around 1.0G, why even measure beyond 2.0G!?!

As for the horizon e.t.c. issue... doesn't the PID values for leveling handle that issue? I mean, the copter isn't returning *instantly* to level, but at some particular speed defined by the levelling PID values... that's what you want to be slow... not the "down vector". Down should always be down... it's the copters desire to have it's "down" down that should be slowed (to the degree you want).

/Z

MasterZap, 2015-05-21:
As for dealing with non-1.0 vectors, I've been thinking....

The G vector is only LESS than 1.0 if we are dropping. So we could probably quite safely ignore quite a lot less-than-1.0 vectors (but of course in a smooth way, i.e. 

0.5G => 0% impact
1,0G => 100% impact

The question is, what do we do with a vector greater than 1.0?

Well, this occurs every time the copter starts or takes off. But remember, the original G vector is a component of this, the copters acceleration or retardation (as long as it is not acceleration straight down) will ADD to the G vector. There will "always" be a component of length 1.0 in this vector.

So... what I propose.... rather than filtering the EstG vector to the direction of the accelerometer vector, you instead, you compute if it is possible for the EstG vector to be a component of the accelerometer vector.

Assume we are accelerating 0.5G forward. Acc will measure 0.5G in X and -1.0G in Z. This will give us a tilted ACC vector. But down is not tilted, down is still down!

But we have EstG around (and assume the gyros do their job) it will be pointing down.

All we need to do is compute if EstG is still close enough to the measured acc value to be a component of that vector. This should be possible with some dot product and.... something else. (I need to draw this out on paper to figure out the details). :)

Only IF EstG is further away from the acc vector direction than where it could be a component of it, do we nudge EstG closer to the vector. The shorter the acc vector will become, the nearer this will be, until the acc vector is exactly 1.0 long, when only exactly one vector will match, and we nudge EstG towards it.

I do vector math all day (computer graphics shading guy) so I'll do a napkin scribble to figure out the math, but it should be trivial. And with the wonder of vector math, you won't need any sine, cosine, or other stuff... dot- and cross-products ftw :)

/Z

ctzsnooze, 2015-05-21:
Well at least one of us understands vector math! :-)

In MultiWii 2.3 there are true Proportional and Integral values for level and horizon, but not a classic Derivative value, instead D acts as a function that simply delays the rate of return to level.  Usually with gyro I well tuned, Level I can be zero (always is, for me).  

In Luxfloat CF there is no classical P I or D either.  P is the proportional factor for Level mode, I the proportional factor for Horizon, and D, in Horizon mode, is the angle of the sticks at which there is zero levelling.  

In neither of these modes is there any suitable D factor.

ledvinap, 2015-05-21:
@ctzsnooze : ACRO P + D is still used i most controllers ...

ctzsnooze, 2015-05-21:
@ledvinap yes definitely for acro/gyros, but where would I find the P I or D settings for the levelling in say PIDC2?  It looks like only P.

ledvinap, 2015-05-21:
https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/pid.c#L362

[39 more replies]

closed 2018-03-07 by hydra