cleanflight/cleanflight #793: feature request: BLHeli 1-wire config from within CF configurator

csurf, 2015-04-19:
Has the possibility been considered and/or discussed to add access to BLHeli ESC configuration settings (and possibly other 1wire-configurable ESC's as well) from within CF/CF configurator? 

I think this would be an awesome addition to CF, and would make it amazingly simple to get all of your ESC's and FC setup in one fell swoop.

sjtrny, 2015-04-19:
Great idea. This would provide a cross platform BLHeli configurator since it is currently limited to Windows.

dkisselev, 2015-04-20:
It's been discussed in #216, and I posted some comments on it in #687

barthchris, 2015-04-21:
Im just a mere mortal here, I have no coding ability besides 1979 era TI-99/4A BASIC! Wish I could contribute, but to those who have the ability I think this would be much used option, .

Without you guys we wouldn't be enjoying this awesome hobby. Thanks in advance if it does come to fruition.

Chris

csurf, 2015-04-21:
duplicate of issue <a href=https://github.com/cleanflight/cleanflight/issues/216>#216</a>. closing this out, will follow the existing issue instead. All those interested should do the same.

closed 2015-04-21 by csurf