cleanflight/cleanflight #796: No telemetry over bluetooth UART1

finalquest, 2015-04-21:
Hi.
Im trying to get telemetry using MSP protocol. I have the bluetooth connected to UART1, and cleanflight configurator can connect to it, so I think that is working.
When I set telemetry output in the port tabs, upon arming the quad, I see no output in the UART1 where the bluetooth is connected.

am I missing something? Should I query cleanflight with MSP commands to see telemetry, or the output should be always there.

Regards

stronnag, 2015-04-21:
If you have telemetry set as a feature, you should get telemetry automagically on arming. If you poll MSP (as mwp, ezgui do), then you should get a response without "feature telemetry". Works for me over BT (MSP polling).

hydra, 2015-05-03:
Isolate bluetooth from the problem, use an oscilloscope, logic analayzer or ftdi adapater to verify that telemetry data is being sent as expected.  MSP telemetry does not work in the same way as MSP. It just sends a sequence of MSP command responses one after the other.

Please re-open if there is indeed an issue with MSP telemetry.

closed 2015-05-03 by hydra