cleanflight/cleanflight #803: Tricopter Yaw servo Kick Naze32 board

Redshifft, 2015-04-23:
Video of exact problem..
http://www.rcgroups.com/forums/showthread.php?t=2385708
This problem occurs with throttle above 1100us and maybe reduces near full throttle, when yaw (rudder) stick is  moved slowly.
Sensor data shows no kicks or spikes
RC monitor shows no kicks or spikes
Servo PWM rate makes no difference
Servo filtering  makes no difference
Fault is found in cleanflight 1.7.1 and newer (older versions not tested)

thenickdude, 2015-04-23:
Which PID controller are you using?

This is most likely due to the way that the PID controller handles the "I" term. Since the "I" term becomes counterproductive during large yaws (it's mostly useful for maintaining position during hover), when the yaw stick exceeds a certain position the yaw I error will be reset to zero and the correction strength will suddenly drop:

```c
if ((ABS(gyroData[axis]) > (640 * 4)) || (axis == FD_YAW && ABS(rcCommand[axis]) > 100))
    errorGyroI[axis] = 0;
````

If you set your yaw "I" setting to zero you'll probably see this problem go away.

I don't know why it works in the case where the throttle is zero, though.

closed 2015-04-23 by thenickdude

Redshifft, 2015-04-23:
If you are right maybe yaw pid loop is not correctly processed below min check then ?
I did ask the question on the other "no yaw" issue  if Tricopters have always had unarmed active control loops as they correct yaw from power on ?

This was on the original Multiwii pid controller

thenickdude, 2015-04-23:
Yes, tricopters have always have unarmed active control loops. This is because CF implements stopping the PID controller when disarmed or at zero throttle by forcing all the motors to be set to zero. However, it never forces the servo to be set to zero.

PID loop is fine for tricopters, there's no min_check threshold where the PIDs for the tricopter yaw stops working.

Redshifft, 2015-04-23:
Well I'm confused with this one Nick.
If it was the "I " term error build surely if you simulate flight and yaw inputs it would at least change ?
It occurs exactly at the same TX input with or without motors running and waving the Tri around in the Air, Also exact problem exists with Baseflight so this is nothing anyone has introduced.

You may recall I asked about servo filtering ? that was because I felt I was getting strange spurious kicks back then, sadly we had the no yaw control unarmed get in the way. I think this is a  bug worth investigating. Whilst Quads are obviously more common maybe this problem exists for them also albeit barely noticeable because of the weak yaw control. Your thoughts ?

thenickdude, 2015-04-23:
You're running with the motors disarmed, so the tricopter cannot meet the yaw rate goal you are setting for it. So the error builds up in the "I" term until it's extremely high, way higher than it will ever get during flight. So the portion of the correction being applied to the yaw servo from the "I" term is unusually high while disarmed. 

As soon your yaw stick crosses that +100 deflection threshold, the PID loop decides that including the yaw "I" term will be counterproductive, so it clears the "I" error ("I" terms aren't useful during high speed yaw, they only end up making it hard to brake to a stop at the end). So suddenly that "I" correction, which was a large part of the servo position, gets cleared to zero. With that "I" portion missing and only the "P" correction remaining, the servo jumps back toward the origin. 

This is irrelevant during flight because the "I" error never builds up like this. As soon as the motors are spinning and you're in the air, the FC can meet the yaw rate goals you are setting for it, so the error doesn't accumulate.

Redshifft, 2015-04-23:
No, motors are armed, If it's "I" build up why is it not slightly different when I physically spin the Tricopter around, motors spinning etc ? but it seems the same.
Like I said I have picked up this problem previously when flying.
Looking at the sensor outputs, moving the Tri in yaw only gets response from gyro and not accelerometer ! I thought they were 3 axis ? I have strong response in pitch and roll.. strange. The TRI stabilizes fine in all axis when flying.
I will have to fly this thing in FPV to get more understanding of this issue and if indeed if it's fully repeatable.

thenickdude, 2015-04-23:
Yaw isn't an acceleration, if you rotate about the accelerometer then it
doesn't record any movement (by design).

If you want to see that it's definitely caused by the I term then just set
your Yaw I setting to zero.

Redshifft, 2015-04-23:
I will try that.
:D of course they don't self auto correct yaw position....to much thinking here !
I realise also you have just been pretty deep into all things Yaw with that floating point pid controller yaw overshoot issue ;)

Redshifft, 2015-04-23:
You will soon be elevated to sir dude :)
Well I'm still mystified rotating the quad (so changing the rate demand) did not influence the problem,unless it's not like a more conventional pid loop as such. It must act as a linear rate demand with no reference to what is actually happening at any point in time ?
As you rightly said the problem can be reduced by decreasing "I"
Thanks again :)