cleanflight/cleanflight #810: The latest cleanflight development builds 1.9.0 look to have isues booting on some STM32F1 based boards

MJ666, 2015-04-24:
There are quite some reports that CF will not properly boot on Naze boards. Sometimes it helps to unplug and reapply the power to get it booting. I personally experience this with an Lulfro Brushed FC. An build a couple of days ago was not even booting at all. An actual build will require to reapply power sometimes an second time to boot. The AfroMini32, AlienWii F1 and Flip32+ is not really affected by this problem. I guess the has to do with the introduction of the interrupt code for the MPU6050 (I may be wrong with this).

Arakon, 2015-04-24:
The CC3D also seems to have no problem with it. I only tested 3 or 4 times, but it seemed to boot up fine every time.

SteveAmor, 2015-04-24:
It would appear that this issue became present in commit 6d5b44df7af32d194448c57a99f2d0af02d89f27 on a Naze 32 acro (AfroFlight 32 Rev5).

It appears to be a timing issue.  If the board has been powered down for more than 10 seconds, connecting 5V to the receiver pins will result in a failed boot.  If you disconnect the 5V and reconnect within a couple of seconds, the board will boot correctly.  If you have a receiver connected, you have to power cycle really quickly to get it to boot and it usually takes several attempts.

I can't replice this issue when powering via the USB port.  It boots correctly every time.

Reverting to commit d3d9721e9195f40752d7aa2cdefe6923fb1c5426 (which is the first, earlier commit that enables the board to boot at all) does not exhibit this issue.

dascons, 2015-04-25:
I have a CC3D and it seems to be affected by this issue as well.
Also a potentially related issue is that it doesn't connect to the
configurator unless powered by battery (This is not an issue with stock FW)

MJ666, 2015-04-25:
This is alse the behavior I also have seen with the Lulfro board.

hydra, 2015-04-25:
@St7ven when your board fails to boot what do the leds to, a video would be helpful here.

hydra, 2015-04-25:
Can you also please use git biset to find the EXACT commit that introduced the problem.

http://git-scm.com/docs/git-bisect

SteveAmor, 2015-04-25:
In this case, the video won't be very exciting and the power LED  is so bright it over exposes the camera so it's difficult to see the other LEDs.  On power up, the power LED lights immediately and less than a second later, L1 (my green led) lights and stays solid.  L0 does not light up (my red led) at all.

If you need me to, I can connect a logic analyser or analogue scope to L1 but I'm pretty sure it's permanently lit (i.e. 100% duty cycle).

I'm struggling with git bisect as there is the bug that stopped my board booting up in the middle of the commits it's asking me to check.  I'll keep working on it.

SteveAmor, 2015-04-25:
It's impossible to find the EXACT commit that has introduced thisbug as I am having to git bisect skip whenever I hit bug 774 that prevents the board from booting up at any time.   The git bisect algorithm isn't closing in on the issue as I have to keep skipping.

nebbian, 2015-04-25:
I know that I had problems with versions that were compiled just after 1.8.1 was released.  I was trying to use it to do testing on a new proximity system, and I know that it wasn't booting at all.  I tried several versions, but going back to 1.8.1 fixed the problem.
Just tonight I tried the latest build from source, which also booted fine.  
I'd suggest that the bug has now been resolved.

SteveAmor, 2015-04-25:
The problem where the board won't boot at all has been resolved (bug 774).  However, this is a different problem and it is still present in the latest build.  The board boots ok from USB but takes several attempts from external 5V supply.

ethomas997, 2015-04-27:
Tonight when I first powered up my Naze32 Acro it exhibited this issue -- froze during boot with green LED lit. After a power cycle it booted OK, and doing a few more power cycles did not reproduce the issue. Was running with power from 5V on the pins (not USB), transmitter off; code was at commit 1d069ad.

--ET

SteveAmor, 2015-04-27:
I can repeat this issue by waiting at least 10 seconds before applying power.

JurikRistjouw, 2015-04-27:
I also have this issue.
- when connecting to USB (beeper starts to chirp like a cricket indicating low voltage and no tx)
- when connecting usb+power, everything boots and is good
- when connecting power only after disconnect, not booting / no tx response, not arming

Spirre, 2015-04-28:
Also got the problem on Naze32, running the latest build with MJ666 G-tune code. When it happens for me the ESC tunes comes out of sync and blue and green led are solid. A few more tries and it works. Only get this when I connect the battery, never happens when I just use USB.

SteveAmor, 2015-05-01:
I ran a git bisect but had to skip all the commits between commit d3d9721e9195f40752d7aa2cdefe6923fb1c5426 (good) and commit 6d5b44df7af32d194448c57a99f2d0af02d89f27 (bad) so it reported that a shed load of commits were possible first bad commits.  I decided on a different approach given I was convinced that the issue was introduced in 6d5b44df7af32d194448c57a99f2d0af02d89f27.  So I commented out 
#define USE_MPU_DATA_READY_SIGNAL https://github.com/cleanflight/cleanflight/blob/master/src/main/target/NAZE/target.h#L74 and my AfroFlight rev5 now boots every time when connecting 5V.
Just to reiterate, there is never a boot problem when powering via USB.  It only happened after leaving the board for down powered for more than 10 seconds and then powering up with an external 5V.

Spirre, 2015-05-01:
Seems like a timing issue, so it's the extra delay that mess things up, or?

SteveAmor, 2015-05-01:
It does look to be a timing issue.  When I was in the field, the issue was 100 times worse.  With a receiver and escs connected, I really had to be very quick with my power cycles to get it to boot correctly.  I put it down to the Spektrum rx and/or BEC at the time but it's clearly not that (and 1.8.1 didn't have this issue with my fully built tricopter).

Spirre, 2015-05-01:
I was doing some tests today and i got it almost every time I connected the battery for the first time. Quite easy for me to notice if it didn't initialize, the LED's wouldn't go on. Unplug directly and plug it in again and it usually worked. I not 100% sure but I believe that I also got it when I just connected the USB.

JurikRistjouw, 2015-05-02:
@SteveAmor thanx, i will test this and report.

Spirre, 2015-05-02:
I've comment out the delay(50); from line 81 in hardware_revision.c and did some tests and no problem so far :)

hydra, 2015-05-03:
are you guys using naze full or acro boards?

SteveAmor, 2015-05-03:
Acro.

stronnag, 2015-05-03:
This only happens for me on the Mullet Acro board. Full naze and flip32/flip32+ do not exhibit the problem.

MJ666, 2015-05-03:
Have problems with the Lulfro Bushed board. AlienWii F1 and Flip32+ looks to be not affected. My Naze, MW32 are not tested yet.

Spirre, 2015-05-03:
Acro board. Seems like the temperature makes a difference. When I tested on the bench I usually don't have the problem, but when I got out last evening it was almost impossible to get it to initialize. But today   it was warmer and it only happened a couple of times. I can get it also when I plug in the USB, but that is not very often.

SteveAmor, 2015-05-03:
I just freezer sprayed my board and it made no difference.  However, I think I have found the problem.  Comment out the following i2cWrite (to a read only register so this line needs to be removed anyway):

https://github.com/cleanflight/cleanflight/blob/a5c2c705417c0f771edd0312a8ef9a72466a1c82/src/main/drivers/accgyro_mpu6050.c#L434

stronnag, 2015-05-03:
With the Acro Mullet, consistently requires power on / off / on to boot. Just a momentary initial connection (to the first ESC chime), disconnect, reconnect is enough. If I don't go through that dance sequence, this particular board will never boot first time on 1.9.

hydra, 2015-05-03:
Its possible the int handler is configured and enabled to early after telling the 6050 to generate data ready signals on the int pin.  I will try and find a board that exhibits the same behaviour.

hydra, 2015-05-03:
@SteveAmor the register is Read/Write. See page 27ofthe mpu6050 reference manual.

SteveAmor, 2015-05-03:
Ooops, I was looking at the INT_STATUS register.

I moved configureMPUDataReadyInterruptHandling() to the end of mpu6050GyroInit and it made no difference.

SteveAmor, 2015-05-03:
Adding a delay doesn't work either.  However, it does exhibit a slightly different problem, only the power light comes on now.  

#ifdef USE_MPU_DATA_READY_SIGNAL
    i2cWrite(MPU6050_ADDRESS, MPU_RA_INT_ENABLE, MPU_RF_DATA_RDY_EN);
#endif
    delay(100);
    configureMPUDataReadyInterruptHandling();
}

SteveAmor, 2015-05-03:
I would do a PR but it's late and as this is a critical bug, I believe the fix is as follows (essentially, moved the chip id read to the beginning and return false is no ack).

bool bmp085Detect(const bmp085Config_t *config, baro_t *baro)
{
    uint8_t data;
    bool ack;

    if (bmp085InitDone)
        return true;

    delay(20); // datasheet says 10ms, we'll be careful and do 20. 

    ack = i2cRead(BMP085_I2C_ADDR, BMP085_CHIP_ID__REG, 1, &data); /* read Chip Id */
    if (!ack)
        return false;

#if defined(BARO_XCLR_GPIO) && defined(BARO_EOC_GPIO)
    if (config) {
        EXTI_InitTypeDef EXTI_InitStructure;
        NVIC_InitTypeDef NVIC_InitStructure;
        gpio_config_t gpio;

        bmp085InitXCLRGpio(config);

        gpio.pin = config->eocGpioPin;
        gpio.mode = Mode_IN_FLOATING;
        gpioInit(config->eocGpioPort, &gpio);
        BMP085_ON;

        registerExti15_10_CallbackHandler(BMP085_EOC_EXTI_Handler);

        // EXTI interrupt for barometer EOC
        gpioExtiLineConfig(GPIO_PortSourceGPIOC, GPIO_PinSource14);
        EXTI_InitStructure.EXTI_Line = EXTI_Line14;
        EXTI_InitStructure.EXTI_Mode = EXTI_Mode_Interrupt;
        EXTI_InitStructure.EXTI_Trigger = EXTI_Trigger_Rising;
        EXTI_InitStructure.EXTI_LineCmd = ENABLE;
        EXTI_Init(&EXTI_InitStructure);

        // Enable and set EXTI10-15 Interrupt to the lowest priority
        NVIC_InitStructure.NVIC_IRQChannel = EXTI15_10_IRQn;
        NVIC_InitStructure.NVIC_IRQChannelPreemptionPriority = NVIC_PRIORITY_BASE(NVIC_PRIO_BARO_EXT);
        NVIC_InitStructure.NVIC_IRQChannelSubPriority = NVIC_PRIORITY_SUB(NVIC_PRIO_BARO_EXT);
        NVIC_InitStructure.NVIC_IRQChannelCmd = ENABLE;
        NVIC_Init(&NVIC_InitStructure);
    }
#else
    UNUSED(config);
#endif 

    bmp085.chip_id = BMP085_GET_BITSLICE(data, BMP085_CHIP_ID);
    bmp085.oversampling_setting = 3;

ledvinap, 2015-05-04:
@SteveAmor : Yes, good catch.

When baro is not detected, unregisterExti15_10_CallbackHandler is called and handler removed, but EXTI is not disabled. After first rising edge on Baro pin EXTI interrupt is called indefinitely... Input is floating, everything is OK if it gets high before EXTI is setup ...

SteveAmor, 2015-05-04:
Doing PR right now

SteveAmor, 2015-05-04:
https://github.com/cleanflight/cleanflight/pull/836/files

stronnag, 2015-05-04:
Tested. Fixes my Acro Mullet, thanks.

hydra, 2015-05-06:
Thanks for the update @stronnag

hydra, 2015-05-06:
Fix merged, closing.  Please re-open if there are further issues.

closed 2015-05-06 by hydra