cleanflight/cleanflight #818: Smartport telemetry when sharing with MSP fails (times out)

Netsplits, 2015-04-26:
When MSP and SmartPort telemetry is shared on the same UART, telemetry fails to work.

After some testings I've detected that the telemetry times out if don't arm before 5 seconds.

I've made a workaround, not sure if correct enough but don't know how to send it GitHub to get it approved or tested. It works for me.

On smartport.c function void configureSmartPortTelemetryPort(void) add:

smartPortLastRequestTime = millis();

so this way we reset the time out timer as soon as the port is open.

at the end of function, so should be:

void configureSmartPortTelemetryPort(void)
{
    portOptions_t portOptions;

    if (!portConfig) {
        return;
    }

    portOptions = SERIAL_BIDIR;

    if (telemetryConfig->telemetry_inversion) {
        portOptions |= SERIAL_INVERTED;
    }

    smartPortSerialPort = openSerialPort(portConfig->identifier, FUNCTION_TELEMETRY_SMARTPORT, NULL, SMARTPORT_BAUD, SMARTPORT_UART_MODE, portOptions);

    if (!smartPortSerialPort) {
        return;
    }

    smartPortState = SPSTATE_INITIALIZED;
    smartPortTelemetryEnabled = true;
    smartPortLastRequestTime = millis();
}

Hope it helps.

hydra, 2015-05-03:
Hi @Netsplits thanks for the feedback.  What do you mean by 'Forgot to say that the side effect, is that can't connect to the GUI when ARMED, so I suppose I defeated the timer'?

Can you also provide a config dump so we can see your settings that you're using for testing.

Netsplits, 2015-05-08:
In fact after posting I saw its a normal behaviour, I mean, that once ARMED telemetry kicks in so UART is no more configured for MSP GUI, so just forget the PostData comment, I thinks it's working as expected.

In my testing bed I'm using a Naze32 board and a home made adapter circuit to split X8R Smarport bidirectional signal to TX and RX unidirectional. Also edited the code to open the port in SERIAL_UNIDIR instead of SERIAL_BIDIR, this way I'm using the X8R receiver without modifications.

Here is the dump:

```
# version
Cleanflight/NAZE 1.9.0 Apr 22 2015 / 00:05:12 (ebd5475)
# dump master

# mixer
mixer HEX6

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature TELEMETRY

# map
map AETR1234

# led
led 0 0,0:::0
led 1 0,0:::0
led 2 0,0:::0
led 3 0,0:::0
led 4 0,0:::0
led 5 0,0:::0
led 6 0,0:::0
led 7 0,0:::0
led 8 0,0:::0
led 9 0,0:::0
led 10 0,0:::0
led 11 0,0:::0
led 12 0,0:::0
led 13 0,0:::0
led 14 0,0:::0
led 15 0,0:::0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
led 22 0,0:::0
[... 251 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
```

kilrah, 2015-05-22:
AFAIK the way it's supposed to work is that if the board does not receive smart port requests from the receiver in the first 5 seconds (either because it's disconnected or not powered) telemetry gets disabled and the port falls back to MSP.

> Also edited the code to open the port in SERIAL_UNIDIR instead of SERIAL_BIDIR, this way I'm using the X8R receiver without modifications.

Not sure what you're trying to do... but as TX and RX need to be "shorted" for the bidirectional smart port communication you can't really use MSP on the same UART without physically disconnecting your adapter circuitry from the Naze first.

Netsplits, 2015-05-22:
It's supposed to work as you described, but in my testings telemetry got disconnect even if receiving telemetry data if not armed in 5 seconds, impossible for me to do as the boot up sequence already takes more than that.

I'm already using MSP shared with Smartport telemetry on UART 1 and works. I do this way so I can free up one UART for other purposes.

To make it to work I'm using SERIAL_UNIDIR connection mode with a custom circuit to convert to one wire smartport requires (it's a dual inverter circuit), and as I said it works.

USB takes priority on my setup, so if I don't enable telemetry via switch or arm (I'm using switch to enable telemetry now, I've made another PR which I use to fix the telemetry switch enable setting #913), I can use configurator. I must disconnect USB cable if I want to use telemetry, but except for testings, I do not use this when at home, so no problem for me.

In fact I made this PR because I saw in the docs that softserials support only 19200 bauds, and I'm using Serial RX on UART 2, GPS on Softserial 1 and in the future I wanted to put the OSD on Softserial 2. Later yourself told me that softserials supported Smartport's 57600 bauds, but I already did this...

commit 13d923a9 (2015-06-24): Smartport telemetry fixes. Closes #818 and #877.
closed 2015-06-24 by hydra