cleanflight/cleanflight #819: ledstrip.h does not take colors defined using 'color' command.

ExplodingFist, 2015-04-26:
It would be nice if we could change the colors of the armed state LEDs, so that we can have the safety feature as well as flight colors being definable (instead of blue). At a minimum, you could make it so that ledstrip.c uses the colors defined in the 'color' command instead of hard coding the RGB values separately. Right now the only way to make this change is changing the hard coded values.

Arakon, 2015-04-28:
Agreed, that would be nice. Maybe combine it with the mode screen, i.e. have a drop down box next to the aux box to select from a choice of colors.

JurikRistjouw, 2015-05-06:
Agreed. I am working on ledstrip.c privatly. I would have to learn the interaction between the configurator and firmware in terms of variables/constants so it can be made congifurable. I am thinking in terms of a flightmode color configuration per mode as an enhancement on the led page. Any ideas?

closed 2017-02-25 by hydra