cleanflight/cleanflight #822: Invert Buzzer pin output state via mode switch setting?  Buzzer pin high at audio frequency?

ctzsnooze, 2015-04-26:
Hi

I was wondering if there could be an option in the Modes section titled 'inverse beeper', so that with a switch I could invert the logical output of the buzzer pin?  I would find this very useful.

Originally I think the intention for buzzer pin was that high = alarm (LEDs flash, buzzer on) and low = normal (LEDs and buzzer off).

For many quad flyers, especially those who fly at night, in low light, or indoors, the more common usage is to invert the behaviour so that LEDs are normally on but flash when battery is low etc.  Typically a small FET is used to ground the LEDs when the pin is high.

However there are times when a quad flyer prefers to be 'stealthy' i.e. have the inverse operation, LEDs off unless there is an issue - without changing hardware etc.  I made a hack in MultiWii to do this, and found it very useful.  I'd be grateful if this could be considered in cleanflight.

Additionally, I was wondering if it would be possible to provide a CLI option that sets the buzzer pin output to oscillate with a 50% duty cycle at a user specified frequency when active?

This would allow the use of simple 'passive' piezo buzzers, which can be much more compact than active buzzers (though typically not as loud as active buzzers with tuned LC circuits).

Passive piezos often have a natural oscillation frequency of 4kHz - very high - so maybe it could be possible that the user can set a frequency in the CLI - zero means normal behaviour (constant), any other value sets a frequency up to some practical limit.  

The buzzer_pin_frequency range could include very low frequencies, say 5Hz for example, so that when active the LEDs flash like a strobe, which could be fun with LED only installations.  One issue at the moment is that when there is an alarm and you rely on LEDs and its dark, the off time is very dark; if the off time was a flashing behaviour you could still detect the alarm and also know where the quad was all the time.

I even though that if the CLI had a non-zero value for buzzer_pin_frequency, then there could be a Mode switch option to enable it or not.  This way you could switch the buzzer on continuously, or have it flash or beep or whatever you like in almost any way imaginable, with switches.

So to summarise, it would be great to:

1.  Have 'Invert Beeper' Mode switch

2.  Have buzzer_pin_frequency setting in CLI, default 0 (i.e. normal not flashing behaviour)

3.  If 2 is implemented, include a 'Beeper Constant' option that overrides the buzzer_pin_frequency (i.e. forces it to zero, overrides it)

With these options people would suddenly have a huge amount of flexibility.  They could use passive piezo buzzers, they could have their LEDs on all the time or off all the time in normal flight, they could move a switch to have the LEDs flash or not flash, or buzz or not buzz, whatever.  

I'd be so grateful if this could one day be built into cleanflight.  

Thanks for considering this.

Chris

hydra, 2015-04-28:
feature requests should be consice and to the point, there are too many topics for one request in here which will result in fragmented discussion and an issue that is not easy to follow.  They should request one thing and describe why the thing is needed.

I will close this issue with the following comments:

1. yes that's posible, open a new issue for this.
2. that requires the use of a timer on the pin the buzzer is connected to but there are no free timers for this.  there may be other ways of doing it but not that would fit very well with the current code, so not possible at this time.
3. an LED overhaul is due, it could work similar to how the current buzzer code works, you could open an issue for this.

closed 2015-04-28 by hydra

ctzsnooze, 2015-04-29:
OK cool thanks, good points, will do.