cleanflight/cleanflight #824: no rudder with power off

Buzzliteyear, 2015-04-27:
Is this a setting change or bug

Airplane mode there is no rudder control when you power off coming in for landing

there is no rudder movement without motor running

cheers 
Brett

hydra, 2015-04-28:
can you try with 1.7.x

thenickdude, 2015-04-28:
This is due to the changes to support pid_at_min_throttle, which added code that zeros out the yaw RC channel when the throttle is low. That probably shouldn't happen on the airplane mixer.

Buzzliteyear, 2015-04-28:
will try 1.7  in a couple of days (weather) thanks

yes same as the yaw on a tri-copter when motors aren't spinning

no its hard to land with no rudder when you need to power off

Buzzliteyear, 2015-04-29:
tested with 1.7.2 rudder works as it should normal movement armed or disarmed

While testing it reminded me that with 1.8 loaded changing flights modes between Angle and Horizon motor was dis arming in Flight but with 1.7.2 it does not do this

PR #826 (2015-05-04): Allow yaw at low throttle for airplane mixes
Fixes #824

This restores the 1.7.2 yaw behaviour for aircraft.
commit 766f409b (2015-05-04): Allow yaw at low throttle for airplane mixes

Fixes #824
closed 2015-05-04 by hydra