cleanflight/cleanflight #825: Naze32 frsky telemetry

AintheT, 2015-04-27:
The voltage telemetry for frsky is not working on v1.8.0 and v1.8.1 . The latest firmware that supported that was 1.7.2. It would be great if you bring this back.

JurikRistjouw, 2015-04-27:
no problem here, how have you connected the wires and setup in cleanflight config?

AintheT, 2015-04-27:
Hey,  thanks for the response.  I followed the instructions on the rcgroups
page,  many people posted trouble shooting, but it did not solve the
problem. I can get rid off the problem just by changing the firmware and
repeating the configurations.  If you want I can repeat it tomorrow and
make a video with all the details.

thenickdude, 2015-04-28:
Works fine for me. Run "dump" on the CLI tab and paste the output here. Also you didn't say what port your telemetry is connected to.

JurikRistjouw, 2015-04-28:
please post your CLI dump and show a foto of how you connected the RX to the Naze.

Spirre, 2015-04-28:
Have you set set? telemetry_inversion = 1

AintheT, 2015-04-28:
Hi, 
I want to point out the difference between versions 1.7.2 and 1.8.1. I triedset serial_port_1_functions = 5 and  telemetry_inversion = 1  also but did not make a difference. 

To test the difference I followed this protocol.
For each firmware following is done
1) flash firmware with full chip erase
2) Configure Motor stop, RX_PPM,VBAT TELEMETRY from configuration tab and save
3) Change channel map to TAER1234 from receiver tab and save
4) board disconnected from cleanflight
5) battery connected and quad armed
6) vfas cells cell values are checked

Here is the setup 
[image]

Telemetry reading after test with V1.8.1
[image]

Telemetry reading after test with V1.7.2
[image]

#Dump from Naze1.7.2
###################################################################
###################################################################
# version
Cleanflight/NAZE 1.7.2 Feb 23 2015 / 14:21:30 (afe44d1)
# dump master

# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature TELEMETRY

# map
map TAER1234

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
[... 589 lines cut ...]
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

#

thenickdude, 2015-04-28:
You've forgotten to choose a port for FrSky telemetry on the "Ports" tab.
You want UART1 if you're using the "FrSky" pin header.

AintheT, 2015-04-28:
Thanks a lot, it works now. I did not have to do that in the previous firmware obviously.

JurikRistjouw, 2015-04-29:
I think that would correspond with the setting:
set serial_port_1_functions = 5

hydra, 2015-05-03:
Please use the forums and/or irc for support.  Issue tracker is for confirmed bugs and feature requests.

closed 2015-05-03 by hydra