cleanflight/cleanflight #827: MOTOR_STOP / disarm after x seconds seems not to work

Vittel, 2015-04-29:
Not really sure if this is a bug, or i just missunderstood the feature.

I recently enabled MOTOR_STOP, which worked fine.
Then the checkmark for "disable after x seconds" appeared, and i tried.
But still, as soon as i reduce the throttle to minimum, the motors immediately disarm. there is no delay or something.
My copter didn't take of yet (still configuring), but i raised the throttle to a reasonable amount for a while.

So, did i get it wrong, or is it broken?

thenickdude, 2015-04-29:
Do you mean that the motors immediately stopped spinning (which is the goal of enabling MOTOR_STOP) or did the flight controller actually disarm (requiring you to rearm it before you can throttle up again?). 

The automatic disarm is intended so that you don't accidentally bump the throttle when you go to pick up your quadcopter whose motors gave stopped spinning. It's not needed when MOTOR_STOP isn't used, because you have a clear visual indication that the craft is still armed.

Vittel, 2015-04-29:
Ah now i understand this!
I expected a different behaviour when Reading the Text.
I Think it would Better iff the Motors do stop, but not instantly. It's a
bit of a security issue that you pull down throttle, then pull up again and
the motor doesn't come up again while in the middle of a flip.
So: do not Spin on arm, but once they are running, do not instantly turn
off when there is no throttle.
Guess it would be great as additional checkbox

Buzzliteyear, 2015-04-30:
two boxs below " Motor_Stop" is a set delay in seconds and you can adjust that to disarm from 0-60
set to 10 and try

tricopterY, 2015-05-02:
If Feature MOTOR_STOP is enabled i.e ticked in the Configurator - Configuration Tab. The motors will not start spinning when Armed (default behaviour is to spin at Min-Throttle i.e idle speed when MOTOR_STOP is Not Enabled, visual indication that craft is armed and do not approach). There is an inbuilt safety feature for enabled MOTOR_STOP (motors do not spin - no visual indication) that kicks in and disarms the motors after that "disable after x seconds" parameter. The default is 5 seconds, and if it's elapsed motors get disarmed, and you need to arm them again. Five seconds is fairly long enough and you would never want to be in a flip/roll situation with your throttle at Min-Throttle for 5 seconds. The suggestion of increasing that to 10 and larger is good. But, I would check the value of my throttle channel, when at its lowest stick position, on the Configurator - Receiver Tab, and set my Min-Throttle to that value. I normally set mine at 1050 (I use Futaba 14SG, non-standard counts eg. centre stick is always 1520). The default of 1150 may be a bit too large for your setup.

hydra, 2015-05-03:
As I understand this there is no issue here.  If I'm mistaken please open a new issue and reference this one from it.

closed 2015-05-03 by hydra

dipeshpatidar, 2016-08-17:
my quad automatically disarm as i increase throttle.......i m using kk2.1 FC board with D2826-6 2200kv outrunner with 30A ESC and 4s BATTERY and 10*4.5 props .......my quad dosent takeoff plz suggest me props