cleanflight/cleanflight #828: Horizon mode won't activate

btimby, 2015-04-29:
I am using an AfroMini with the Naze firmware loaded (1.8.1). I have configured modes such that HORIZON should be activated, but the quad flies as if in ANGLE mode. I have verified using the configurator that HORIZON mode is configured and becomes active.

[image]

I also double-checked the CLI aux command and it looks correct.

    # aux
    aux 0 2 0 900 1600
    aux 1 3 0 1200 1600
    aux 2 0 0 900 900
    aux 3 0 0 900 900
    aux 4 0 0 900 900
    aux 5 0 0 900 900
    aux 6 0 0 900 900
    aux 7 0 0 900 900
    aux 8 0 0 900 900
    aux 9 0 0 900 900
    aux 10 0 0 900 900
    aux 11 0 0 900 900
    aux 12 0 0 900 900
    aux 13 0 0 900 900
    aux 14 0 0 900 900
    aux 15 0 0 900 900
    aux 16 0 0 900 900
    aux 17 0 0 900 900
    aux 18 0 0 900 900
    aux 19 0 0 900 900
    aux 20 0 0 900 900
    aux 21 0 0 900 900
    aux 22 0 0 900 900
    aux 23 0 0 900 900
    aux 24 0 0 900 900
    aux 25 0 0 900 900
    aux 26 0 0 900 900
    aux 27 0 0 900 900
    aux 28 0 0 900 900
    aux 29 0 0 900 900
    aux 30 0 0 900 900
    aux 31 0 0 900 900
    aux 32 0 0 900 900
    aux 33 0 0 900 900
    aux 34 0 0 900 900
    aux 35 0 0 900 900
    aux 36 0 0 900 900
    aux 37 0 0 900 900
    aux 38 0 0 900 900
    aux 39 0 0 900 900

Otherwise it flies great, any ideas?

btimby, 2015-04-29:
Also, I tried re-flashing whilst clearing the EEPROM and reconfiguring to no effect. I have several other quads with the same configuration (although using Acro Naze boards rather than the mini) so I am fairly confident that the configuration is correct.

thenickdude, 2015-04-29:
Hm, on the receiver tab do your channels reach the appropriate ranges (1000 to 2000)? If they are substantially short I could see you getting stuck in the self levelling portion of the stick throw. 

Which PID controller are you using?

btimby, 2015-04-29:
In the screenshot, you can see I don't have any ranges defined for self-level. The yellow box on the HORIZON range shows the value of the channel when my flight mode switch is in it's default position. This screenshot was taken with the transmitter switched ON.

In other words, HORIZON is the default mode and should be active for two of the three positions of the switch (the last position is RATE). 

I am using the default PID controller, 0 - MultiWii.

btimby, 2015-04-29:
That is what really gets me, ANGLE should never be active, but when trying to flip, the quad just goes full bore in the stick direction, the attitude is limited as it is in ANGLE mode.

thenickdude, 2015-04-30:
Hm, what is your rc_rate set to? The stick throw scaled by rc_rate is used to decide between self-levelling and gyro flight, so if rc_rate is small then you will get stuck in self-levelling. The default is 90 which corresponds to a stick scaling of 0.9, which should be fine. You can set it to >1.0 to ensure you're definitely in the gyro-only portion of the stick at full throw.

tricopterY, 2015-04-30:
Use CLI dump and double check the value of 'sensitivity_horizon', it defaults to 75. You may want to reduce that, or set it to zero.

btimby, 2015-04-30:
rc_rate is 0.90, the default.
sensitivity_horizon is 75 the default.

I have not tried changing them yet. The settings are bone stock, I erased EEPROM and changed only the following:

SERIAL_RX and channel mappings
Flight modes
-VBAT
+SERVO_TILT
Calibrate acc

And now that I think of it, perhaps SERVO_TILT affects HORIZON mode? I changed this because the AfroMini uses pads rather than pins for motor outputs otherwise. This change is suggested in an RCGroups thread regarding this FC. Not something I can easily disable to test, googling does not suggest this would have an effect.

http://www.rcgroups.com/forums/showpost.php?p=28328731&postcount=3

btimby, 2015-04-30:
Dump, just in case:

    # version
    Cleanflight/NAZE 1.8.1 Apr 2 2015 / 17:06:11 (988ae2d)
    # dump master

    # mixer
    mixer QUADX

    # feature
    feature -RX_PPM
    feature -VBAT
    feature -INFLIGHT_ACC_CAL
    feature -RX_SERIAL
    feature -MOTOR_STOP
    feature -SERVO_TILT
    feature -SOFTSERIAL
    feature -GPS
    feature -FAILSAFE
    feature -SONAR
    feature -TELEMETRY
    feature -CURRENT_METER
    feature -3D
    feature -RX_PARALLEL_PWM
    feature -RX_MSP
    feature -RSSI_ADC
    feature -LED_STRIP
    feature -DISPLAY
    feature -ONESHOT125
    feature -BLACKBOX
    feature RX_SERIAL
    feature SERVO_TILT
    feature FAILSAFE

    # map
    map TAER1234

    # led
    led 0 15,15:ES:IA:0
    led 1 15,8:E:WF:0
    led 2 15,7:E:WF:0
    led 3 15,0:NE:IA:0
    led 4 8,0:N:F:0
    led 5 7,0:N:F:0
    led 6 0,0:NW:IA:0
    led 7 0,7:W:WF:0
    led 8 0,8:W:WF:0
    led 9 0,15:SW:IA:0
    led 10 7,15:S:WF:0
    led 11 8,15:S:WF:0
    led 12 7,7:U:WF:0
    led 13 8,7:U:WF:0
    led 14 7,8:D:WF:0
    led 15 8,8:D:WF:0
    led 16 8,9::R:3
    led 17 9,10::R:3
    led 18 10,11::R:3
    led 19 10,12::R:3
    led 20 9,13::R:3
    led 21 8,14::R:3
    led 22 7,14::R:3
    led 23 6,13::R:3
    led 24 5,12::R:3
    led 25 5,11::R:3
    led 26 6,10::R:3
    led 27 7,9::R:3
    led 28 0,0:::0
    led 29 0,0:::0
    led 30 0,0:::0
    led 31 0,0:::0
[... 242 lines cut ...]
    set i_alt = 0
    set d_alt = 0
    set p_level = 90
    set i_level = 10
    set d_level = 100
    set p_vel = 120
    set i_vel = 45
    set d_vel = 1

    # dump rates

    # rateprofile
    rateprofile 0

    set rc_rate = 90
    set rc_expo = 65
    set thr_mid = 40
    set thr_expo = 75
    set roll_rate = 55
    set pitch_rate = 55
    set yaw_rate = 40
    set tpa_rate = 50
    set tpa_breakpoint = 1500

    #

tricopterY, 2015-04-30:
Generally feature SERVO_TILT is used for camera stabilisation (Gimbal), but in your case it enables certain motor outputs. As you can fly in other modes without any difficulty, the Horizon mode remains a mystery. Your dump looks ok, and your roll and pitch rates of 55 are ok too i.e should flip with ease. So, hopefully others can shed some light on this issue. 
Raise this issue at irc://irc.freenode.net/#cleanflight.
Intuitively if some craft is carrying a gimbal, it should not flip and roll, may be that SERVO_TILT is confusing matters for FC. See, if there is a workaround.

tricopterY, 2015-04-30:
I just realised SERVO_TILT moves motor pin-outs for a quad, two pins forward. Disable that feature and use the normal pins for the motor outputs.

tricopterY, 2015-05-01:
My comment about 'sensitivity_horizon' is only valid for pid_controller = 2. I just noticed in your dump, you are using pid_controller = 0.

btimby, 2015-05-02:
I am going to try without SERVO_TILT, but I have to find time to take the quad apart and do some soldering.

tricopterY, 2015-05-02:
:smile: Cool

btimby, 2015-05-03:
No dice. I disassembled the copter soldered a pin to the M2 pad on the bottom. I then reassembled, moved all the ESC signal wires, entered:

    feature -SERVO_TILT
    save

But yet when I jam the stick right the copter shoots right but won't flip.

btimby, 2015-05-03:
I tried again. The copter rotates past 90 degress, so the top is pointing right and a bit downwards. I would say it is being limited to 100 degrees or double 50 degrees (max_angle_inclination).

btimby, 2015-05-03:
Sorry, I  hit the close button accidentally on my tiny phone screen.

hydra, 2015-05-03:
Servo tilt is unrelated.  A blackbox log would help.  Horizon mode is clearly activating (as per the screenshot).

hydra, 2015-05-03:
I would suggest for asking for help on rcgroups forums and posting exact specs of your quad too, maybe someone there can help isolate the issue.  Once you've isolated the issue please re-open this if there's still a problem.

closed 2015-05-03 by hydra

nivaldin, 2015-07-17:
I have the same issue... did you get the solution?

chipko, 2015-08-23:
I'm having the same issue on a Naze32 rev4 board...