cleanflight/cleanflight #830: CC3D channel 1 input dead with PWM Rx

redbeardpete, 2015-05-01:
Hey guys, thanks for your help on the other issue I posted. I feel pretty dumb I didn't plug in the battery, but I did not know it was a soft-serial. Anyway, I've got another CC3D weirdness I'm hoping is super easy as well. 

I have one input on the CC3D that is dead. It used to be the throttle input on Openpilot, now it does nothing. I tried different channel maps on the RC page but I tried several different Rx channels and all my switches and controls and no combination would ever show anything on this input. I also tried both Spektrum settings on setup (1024 and 2048 was it?). Are there any other possible inputs than T, A, E, R, 1, 2, 3, 4? I rearranged the wires and have TAER1 all working fine.

The dead input is wire #1, white, on the same plug with the power and ground wires.
I'm using plain ole PWM channel inputs from the (Frsky) Rx. 

I'll be happy to fill in any info holes, just give me a shout.

dascons, 2015-05-01:
I have had this issue but thought it was a case of my particular board just
being baaaad, this is in fact something that needs fixing, I'm willing to
test fixes for people.
The way I got around it was getting a ppm encoder.

redbeardpete, 2015-05-01:
Interesting, thanks for sharing! Somehow I lucked out and the only thing I'm missing is Aux2 which I can live without. I didn't even change the channel map, just moved the sticks, saw what moved, and swapped wires around until it was right (much different from the openpilot input order). Tx on TAER1234 and same on the FC.

dascons, 2015-05-01:
Hrmm, I think I had a different pin that did not take input, it was one
that meant that if I change the channel map it still wasn't fly-able, like
CH 3 or something, heh I cant really remember since I have been using ppm
for a while

projectkk2glider, 2015-05-01:
This might be related to my incident.

I have an Flip32+ board (naze32 clone) and all RC inputs are tied directly to the CPU. Unfortunately the designer used IO pins that are **not 5 volt tolerant**.  The board also does not have any series resistor from inputs to the CPU IO pins. 

How this affects things? Well in my case I mistakenly connected +5V on first RC input (that I use as PPM input). After realizing my mistake I corrected it and the board worked, but behaved strangely, for example the mid RC value has moved from 1500 to 1525us for all channels!

The after couple of flight thing worsened. The PPM input was not recognized anymore and CPU was getting hot. This board still boots and led blinks, but CPU gets extremely hot. It is toast.

You might have as similar problem, you connected +5V volts to one of RC inputs and have burned it. That is why it is not working anymore.

dascons, 2015-05-01:
I really don't think this is the case as running stock FW it behaves as
expected

redbeardpete, 2015-05-01:
Interesting problem, got me this came about immediately following CF firmware load from openpilot and no plugs were yet moved, also never plugged 5v in the wrong place when I did move plugs. It does happen the dead channel is on the same plug with VCC and GND wires that are powering the rx but I think that's a coincidence

hydra, 2015-05-03:
The pin is not dead, it's unused by cleanflight in Parallel PWM input mode. See cleanflight documentation for the CC3D board.  Sharing timers used by inputs and outputs requires code re-work, see timer.c pwm_output.c pwm_rx.c pwm_mapping.c, this is why PIN3 (white) is not used in Parallel PWM input mode.  PR's accepted that fix this.

closed 2015-05-03 by hydra

bbqflyer, 2015-09-14:
I'm completely new to cleanflight, I'm having same issue.  So what's the fix here?

redbeardpete, 2015-09-14:
the input is not used by CF in PWM input mode... use a different input pin
and re-map as necessary in cleanflight

acdi01, 2016-04-29:
I'm stilling having the same problem with my cc3d atom. I can't get any aux channels to work. Any idea's?
Thanks,
Alex

dascons, 2016-04-29:
It is likely that it is a physical issue with your board.

acdi01, 2016-04-29:
Hmmm maybe. I had it working with Libre pilot though.
Thanks,
Alex

TsPCs, 2016-12-14:
same problem everything works just fine in dronin but not on your firmware whats wrong here

Sympster, 2017-01-28:
Same problem here.  Have a band aid solution.  Moved CC3d wire that normally went to throttle to the battery/bind slot on my Spektrum receiver.  Then move the last wire to the throttle slot on my receiver, then rearranged channel map within betaflight so the channels mapped correctly.  This effectively only uses 5 of the 6 channels on my receiver.  4 for directional control, 1 for flight mode.  None for switch arming or beeper or anything else.