cleanflight/cleanflight #832: Buzzer for all controllers

snakescb, 2015-05-01:
On some controllers, Sparky or CC3D, I am really missing the buzzer. It would be quite easy to connect a buzzer on any board on PWW out 6 for example, if unused. We have GND and VCC and SIGNAL present there, so it would be easy to build a cable with the buzzer, transistor and resistor and a servo plug, to connect this directly to an unused PWM output.

Would it be possible to add an option to reroute the buzzer signal to an unused PWM output with CLI?

Regards
Christian

Arakon, 2015-05-01:
It's still possible to connect a buzzer by going directly to pin 15 using a small circuit. But that's not exactly easy to solder, so if there was a "pretty" way of doing it, that would be great.

Pierre-A, 2015-05-02:
same as issue ≠653.
I did the mod and it works fine ; so ready to open the appropriate PR.
But what other users think of that ? means that output 6 could no longer be used for anything else. Except if we introduce a new CLI parameter specific to CC3D in order to choose the beeper output.
**hydra** what do you think ?

hydra, 2015-05-03:
Im not sure about this, maybe adding BUZZER as a feature is the right thing to do here..

The problem comes about due to pin/timer allocation.  Perhaps pwm mapping needs to happen first then if there's a spare PWM pins use one for the buzzer.

I'm reluctant to accept PR's for this until the timer/pin allocation for existing features is done.  See #357

Pierre-A, 2015-05-03:
OK , we put that on hold for the moment.
adding BUZZER as a feature requires some reflexion : what for under the on/off or with/without function? and consequences.

hydra, 2015-05-03:
@Pierre-A exactly. Its not straightforward and needs some thought.

JohnieBraaf, 2015-05-08:
Would it be an idea to use the CAN bus connector on the Sparky? You can do more with it like passing info to a gumball unit, but currently its unused for most people I think.

hydra, 2015-05-15:
The long term solution is likely to use resource allocation and configurable buzzer pin.

snakescb, 2015-05-15:
I was able on Sparky to use a PPM controlled Buzzer on RC Output 5, gimbal_flags = 4 and passing an AUX channel to it. It is working like this. For me, it is very essential to have a buzzer for small FPV flyers. I would have lost more than one without this.
Soldering as on CC3D was fine for me too, unfortunately Sparky does not deliver the signal on an unused pin.
Would be great to have the long term solution as hydra propsed in future. Thanks for thinking about it.

closed 2017-02-25 by hydra