cleanflight/cleanflight #833: Angle mode Problems With Damped Light on BLheli

Tabsy123, 2015-05-03:
This pitching back or sideways issue is more noticeable of crafts with high power/thrust  to weight ratio.
I think it may be something to do with the Angle mode code/pid not being able to recover some motors if they slow down too much from the active braking .  And being on a craft with very high thrust to weight ratio then the motors are at low RPM when at a hover etc already.
But im not too sure.   All I know is this issue is starting to pop up here and there from other people more often now.   And as soon as I turned of Damped light in esc on my craft, Angle mode became back to normal with no issues at all.

hydra, 2015-05-03:
Can you describe the symptoms a little clearer?
Can you provide a blackbox flight log?

Tabsy123, 2015-05-04:
I am just confirming some things today,   as it might be a blheli version issue combined with active braking on.   Just for more info even with active braking off on blehli 13.2 , fault is still there. Im about to go out and try active braking off and blheli 12.2 soon.

TYhe issue is that when you flick the switch to go into angle mode, the craft will pitch back in my case up to about 45 degrees (bad) ,  other people have reported to the side roll and pitch drift that is real bad also.   mne however is just leaning right backwards,  it exagerates if you add throttle ,  sometimes this will bring on the fault if it did not present for you when you first flicked into angle mode etc..

Tabsy123, 2015-05-04:
Ok now im stummped,  Maybe theres just a bad batch of mpu6050's out there or something.
Or can vibes cause this,  but only in angle mode?

As just then went back to 12.2 blheli with damped light off  and fault is still there......
Which is so starnge as before when i did this and only this,  I then went out to fly and tryed the craft more than 10 times and it never faulted (10 times flicking angle mode on and off and flying around and some punch outs also).   
So maybe its just my luck and some others out there that some faulty mpu6050 ????  
As I purchased a new Mullet32 Naze and it had the same fault as my genuine naze32 that was also just new out of packet.......
Im clutching at strings here now ....

ledvinap, 2015-05-04:
@Tabsy123 : it would be best if you can provide blackbox log. There is SPI flash on Mullet32 Naze, so it should be possibly to log without external hardware.

Tabsy123, 2015-05-04:
Ok thanks Ledvinap,  Im new to logging,  is there anywhere that you could point me to to help guide me on what to choose/select to try this out?

ledvinap, 2015-05-04:
@Tabsy123 : There is something called 'documentation' ;-)

https://github.com/cleanflight/cleanflight/blob/master/docs/Blackbox.md

Tabsy123, 2015-05-04:
Cheers, I think i will order the open log microsd...

ledvinap, 2015-05-04:
In long term it's good idea, blackbox logs are great for lots of things. But using onboard flash is enough  for this issue ...
The onboard flash should hold about 4-5 minutes of recording...

Tabsy123, 2015-05-04:
Yeah my acro version board does not have any flash memory  :(   
I did the order tonight for the sparkfun openlog.

In the mean time I have transfered one of the naze32 boards that was doing this fault into another Quad with different layout of electronics and differnt esc's and motors etc,  So while im waiting on the openlog I will see if the fault is there with that FC in another craft.  If it follows the FC then It is looking like it might just be that I have 2 different brand new naze32's with the same issue.....
This time I will not post up nothing until ive at least put 2 batteries through it.

borisbstyle, 2015-05-04:
I know exacly what this problem could be, cause I just experienced it. There is this issue with signal being sent to the motors during boot. I think that can calibrate the ESC as I had the same problem today that my quad was suddenly pitching back. I read the throttle values in blhelisuite and I saw maxthrottle of 1350 configured. Disable programming by TX in blheli and set the proper values manually.

dascons, 2015-05-04:
Is the max throttle the same on all or just one esc?

borisbstyle, 2015-05-04:
In my case it was even 2 ESC's, but a guy I know had it on one.

I noticed it as my quad was starting to pitching back while flying. I also heard a bit of rattling noise, which was clipping of the ESC's.

Tabsy123, 2015-05-05:
Thanks Boris for the sugestion :)
But in my case I dont think its it,  as acro mode even to full throttle is flying perfect, also I calibrate them manually in blheli suite  and have checked the values after wards and they have not changed  Aslo I am prety sure that unticking tx programming does still not disable the end point calibration by tx as i was told by one of the dev's.

An Update:
Totally different craft with one of the offending naze's put in it,  tested for 2 battery packs, unpluging battery many many times and re-powering up,  no fault at all.

So it must either be Vibes effecting Accel sensor only,  or some kind of interferance on the frame effecting accel sensor only.  As manual mode is perfect.

You know how the Gyro lpf works ,   does the Accel have this feature ?   does accel_lpf_factor do a similare this to gryo_lpf?   could i lower this value lower than 4 to see if vibes might be effecting the angle mode?

Another thought is if the frame is like tweaked/twisted just slightly by around 2 or 3 degrees  could this cause the fault?  I would have thought that the FC angle mode code is smart enough to just adjust each motor enough to compensate for just slight frame imperfections that would only be noticable with some sort of measuring device etc.

ctzsnooze, 2015-05-06:
There is something definitely wrong with acc handling in CF.  There are serious issues with Level and Horizon in PIDC 2 and 3.  I too am running OneShot, maybe that's the problem. It is so bad that Level mode is nearly completely useless. I have been 'hand maiden' testing and find on my quad that it will go to nearly 45 degrees left after a few moments.  This happens after trimming acc.  None of the CLI features appear to make any difference.  It just does not level like any other quad I have.  I need to post some videos.  I don't have blackbox capability.

ledvinap, 2015-05-06:
@ctzsnooze : You can configure Blackbox to use same port as MSP and use PC to save blackbox data (http://realterm.sourceforge.net/ is one of many possibilities .. ). Main disadvantage is that your quad needs to be connected for recording ...

ctzsnooze, 2015-05-06:
OK thanks I made a video on http://www.rcgroups.com/forums/showpost.php?p=31545774&postcount=7347 and will see if I can get blackbox recording to my mac.  The FTDI might not be fast enough but I only need sticks, motors and acc I guess.  The issue is totally reproducible on two separate quads.  Both run OneShot and Damped Light and people are wondering if that's why the issue is coming up for quite a few people as those modes become more common.  I don't have a quad without those modes but will test this later too.  Thanks for your help.

Arakon, 2015-05-06:
I can confirm that when holding the quad in my hand and switching from Horizon to Acro, I get a slight tilt to the right. In my case, Horizon mode actually feels right/level, though, and Acro knocks to the right. I'm using a CC3D. Blackbox Log:
http://wikisend.com/download/236552/acro-horizon.txt

Note that this was done holding it in my hand, and switching back and forth between acro and horizon. Only the 4th log in the file is relevant (49 seconds long).

ctzsnooze, 2015-05-08:
Hi guys - this thread is a sort of duplicate of https://github.com/cleanflight/cleanflight/issues/791

I don't want to double post... should we somehow merge the threads?

Tabsy123, 2015-05-11:
Update:  My craft is all fixed now,  it turns out it was exactly as others in #791 issue have been chatting about (vibes to the ACC sensor causing offset) ,  Gyro lpf seemed to do its magic so issue was not there in acro etc.

I have mounted my FC now between to soft silicone shock orings from my rc cars , and bored out the holes in naze a bit bigger to get the naze to not be physicly connected or touching the Fc mounts in anyway other than the soft silicone orings.  And its fixed.

Also note that my build had no vibes i could feel by hand, craft had never been crashed at all, nothing loose on craft at all, Brand new cobra 2204 motors, HQ 5x4 props brand new.
In other words the frame didnt show any vibes by hand or any noticable in the motor test tab or sensor tab, well not with the standard sensor tab graph settings anyway.

closed 2015-05-11 by Tabsy123

ctzsnooze, 2015-05-11:
Good news!  I also could not feel anything much in the frame itself.  Glad to hear yours is good now!