cleanflight/cleanflight #843: RTH failsafe shutdown motors after 20 seconds

andrenoites, 2015-05-04:
20 seconds is the maximum delay for the failsafe_off_delay and the quadcopter disarm in the middle of the flight (RTH)

The same applies to hold position fail safe, after 20 seconds it shuts off the motors

It would be great if I can set larger delays or an option to disable failsafe_off_delay

Thank you

hydra, 2015-05-06:
Really, the failsafe system should be updated to include RTH facilities.  Not sure if it's worth hacking it to work around issues with RTH flight mode at this time.

digitalentity, 2015-05-06:
Currently failsafe is not capable of RTH, probably GPS navigation should be improved first (#197). But I like the general idea of failsafe RTH instead of simple landing.

Jackuzy, 2015-05-06:
Well, as baseflight and from it cleanflight are based upon multiwii software maybe You should check the MultiWii by Mis version? It has lots and lots of improvement to use failsafe GPS RTH, PosHold and even baro landing ;)

It all actually is working (checked it my self on old boards some time ago) on 8bit's Guys! And yet not much of improvement in the 32bit world ;) 

http://rc-fpv.pl/viewtopic.php?f=46&t=10002
http://romek_by.republika.pl/MK/MultiWii_2_27_MIS.zip

digitalentity, 2015-05-06:
@Jackuzy, thanks for the feedback, I'll have a look at Mis's code.

Tinkyy, 2015-05-08:
The RTH feature work's well enough. The main problem with GPS hold and RTH is magnetometar. It is very hard to use internal board mag on larger drones because of large current's that ESC's draw cause havoc on measurement. The mag offsets on my drone from 0 to + 90° from idle to full throttle, standard 2212 980kv x4 motors. 2 simple solutions:
1. Map the mag changes in relation to throttle range and compensate in software via error table.
2. Cut the SDA trace pin 1 of internal mag and use $3 mag board from Ebay affixed to the bottom of landing skid, connected to i2c bus (the solution I use and works like charm)

My drone returnes to +/- 2m of takeof position every time now.

Tinkyy, 2015-05-08:
Failsafe proposition:
 If signal lost:
1 Check baro if  <30m if yes climb to 30m
2. RTH till distance from home is <3m
3. Set failsafe throttle to descend value
4. wait for accelerometar spike (touchdown) and then disarm
5. beep :)

ProDrone, 2015-05-09:
In PR #852 i added a failsafe flightmode with things like POS Hold and RTH in mind. What i really like to end up with, is a configurable failsafe action script. The current simple landing could be one of these actions.

Argotera, 2015-05-09:
I would like to see too rth as a failsafe option.
I do not know if it's of any use, there is a implementation of rth as failsafe in this thread in Multiwii forum:
http://www.multiwii.com/forum/viewtopic.php?f=8&t=4672

Final version is several posts down the thread.

digitalentity, 2015-05-11:
@ProDrone, I'm working on new navigation framework with better PH/RTH/WP. I am also having RTH on failsafe in mind, so adding a "failsafe" mode will come in handy.

hydra, 2015-05-11:
Yes, I will be merging in the FAILSAFE flight mode soon.

Tinkyy, 2015-05-14:
Nooo, you will put DJI out of business :)
Keep up the good work. I will gladly test (make a video) and report the results.

Argotera, 2015-05-26:
Just a note that I think affects Gps:

The default configuration for a Ublox GPS (neo-6m,m8n etc) which is the majority of what people use) is to set Dynamic Model to Pedestrian.
Using gps_auto_config=1 will also configure the gps module to Pedestrian mode.
 
Would it not be better to use Portable mode or one of the Airborne modes? 
Pedestrian mode offers 2d position fix while the other do not, but it is targeted for applications with low speeds, low acceleration, and not much vertical movement in general.
This is ok for slow AP platforms, but not for faster aircrafts.
 PatrikE of Multiwii commented recently on this in a forum

Rockson-slo, 2015-07-23:
Hi guys, I can see you made a great progress with cleanflight. Congrats to developers...
Are there any news about using RTH as a failsafe option?
No landing or such, just a possibility to regain control if out of rc range.
I guess it shouldn't be that hard, because I can allready activate RTH manually?
I am not trying to be smart, just thinking.

Best regards to all.

ProDrone, 2015-09-07:
> I'm working on new navigation framework with better PH/RTH/WP. I am also having RTH on failsafe in mind, so adding a "failsafe" mode will come in handy.

@digitalentity #852 and some supporting PR's are now merged to 1.10 master. Currently there is only a simple auto-landing procedure in place. It would be very nice if this could be extended with RTH,

There is a (little?) problem though. On F1 targets people tend to disable the ACC to get faster looptimes. Without a working ACC it is impossible to auto-land as it uses level mode. Is it correct that RTH and PH both need a working ACC?

To overcome this i first thought of this: https://github.com/cleanflight/cleanflight/pull/1287, But that cannot be done.

I am working on a different approach. I divided ACC processing over 10 parts in a state machine, so that it is processed with less time consumption per loop. With this modification i achieve looptimes of min/max: 303-840 with `set looptime = 0`.

The quad will at least react must slower, but it may be enough. This needs to be tested and is still work in progress.

digitalentity, 2015-09-07:
@ProDrone 
I'm keeping an eye on your work as I intend to add ability to RTH on failsafe event - IMO this is critical for far fpv flying.

>There is a (little?) problem though. On F1 targets people tend to disable the ACC to get faster looptimes. Without a working ACC it is impossible to auto-land as it uses level mode. Is it correct that RTH and PH both need a working ACC?

You are correct, NAV is heavily using accelerometer for speed estimation. There is one more problem - navigation code is rather taxing on CPU, I was only able to go as low as 2000us looptime with navigation modes active. This might be a problem for non-looptime-aware pid controllers that were tuned at looptime=1000.

digitalentity, 2015-09-08:
@ProDrone 
>I am working on a different approach. I divided ACC processing over 10 parts in a state machine, so that it is processed with less time consumption per loop.
Maybe better solution would be to use less time-consuming IMU? AFAIK @bdoiron74 is working on quaternion controller (#1192) and @MauroMombelli suggested using a DCM filter (#1253). Both could be significantly faster than current attitude estimation logic.

ProDrone, 2015-09-08:
@digitalentity Nice! Faster processing is of course a better option. I was playing with #1192 and added PR #1107 to it, replacing occurrences of standard functions with the *_approx functions. Next i `set looptime = 0` and show debug output: like current, min. and max. cycle times.
 (min. and max. are reset every 20 seconds).

Debug output showed me these min..max values, using:
~~~
standard math:  865..1012
fast math:      741...934
very fast math: 745...928
~~~

So it is less improvement then i hoped for. Also the upper limits are close to or exceed 1000, which may cause trouble with gyro-synced loop at 1KHz sample rate.

I am going to try and split processing like i did with the old IMU calculations...

MauroMombelli, 2015-09-09:
Please note that MahonyDCM without fastmath WITH magnetometer take 147us, WITHOUT magnetometer is take 85us(!), this is without sensor reading (see#1253 to get more information and data, please take attention that timing are for 5000 loop to get a better estimation)

IMHO having reading code inside DCM logic is bad and evil, and should be moved

digitalentity, 2015-09-12:
@ProDrone 
Can you give some advice on how to implement "RTH on failsafe" functionality?
I am thinking on having an option to override autolanding and modify this line (https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/failsafe.c#L115) to skip FAILSAFE_LANDING phase and jump to FAILSAFE_RX_LOSS_MONITORING phase. Actual RTH activation will be handled by navigation code depending on result of failsafeIsActive() function. What do you think?

ProDrone, 2015-09-12:
@digitalentity
I prefer to to keep control (initiation, monitoring and abort) at a central place, and add states to the state-machine that currently only handles auto-landing.

It is cleaner to let the failsafe handler call something like `activateRTH()` and `abortRTH()`, instead of monitoring `failsafeIsActive()` at another place in the code (unless used for functions that do not attempt to control the failsafe procedure, like display, beeper, LED control). 

Selection of the desired failsafe procedure can be done with a CLI variable, which should be interpreted by the failsafe handler (and exposed in the GUI).

The failsafe handler also needs a function like `getStateOfRTH()` to monitor progress.
I expect RTH has several states, like IDLE, IN_PROGRESS, LOST_GPS, HAS_LANDED, etc.

Btw. what happens if GPS is lost while RTH (or PH) is in progress?

I did not yet look at your code and like the idea of separation, with a well defined API in between.

If you want me to, i can make these changes and add stubs for the API functions. You should fill in these functions to make it work (?).

digitalentity, 2015-09-12:
@ProDrone 
Your way looks even better to me - cleaner and more modular. This way if failsafe or navigation ever gets replaced by a (better) one, the other module would be kept completely intact as long as then both use common API.

GPS loss currently is not handled at all. Thanks for reminding me of that, I'll see to it. I think the best way to handle GPS loss during RTH is to land immediately.

I would be grateful if you would modify failsafe code to handle RTH on failsafe, I'll then merge your commit to my branch and eventually it'll get merged as a part of nav-rewrite or a separate PR. I would suggest the following function naming: `void activateForcedRTH(void)`, `void abortForcedRTH(void)`, `int getStateOfForcedRTH(void)`.

ProDrone, 2015-09-12:
@digitalentity
 
The functions `void activateForcedRTH(void)` and `void abortForcedRTH(void)` are clear.

I like to change `int getStateOfForcedRTH(void)` to `rthState_e getStateOfForcedRTH(void)`

and then define your required states *with a description* in an enum in your RTH.H code like:
~~~C
typedef enum {
    RTH_IDLE = 0,               // RTH is waiting
    RTH_IN_PROGRESS_OK,         // RTH is active
    RTH_IN_PROGRESS_LOST_GPS,   // RTH is active but has lost GPS lock
    RTH_HAS_LANDED              // RTH is active and has landed.
} rthState_e;
.
.
.
rthState_e getStateOfForcedRTH(void);
void activateForcedRTH(void);
void abortForcedRTH(void);
~~~
and use it in your RTH.C code like:
~~~C
rthState_e rthState = RTH_IDLE;
.
.
.
rthState_e getStateOfForcedRTH(void)
{
    return rthState;
}
~~~

If you supply the states/description then I can modify failsafe code to use it like above.

digitalentity, 2015-09-13:
My code already has a state machine for autonomous navigation - RTH is part of that state machine. I've made up some code to provide the interface you describe in 29ec999fadc2e9e04a23eac0190e9a4099ce3162 . Internals may change, but this API will be kept in place.

ProDrone, 2015-09-17:
@digitalentity I have the modified failsafe handler code in:
https://github.com/ProDrone/cleanflight/tree/add_rth_support_to_failsafe_handler
You may check it out and discuss it further here...

digitalentity, 2015-09-17:
@ProDrone thanks! Branch pulled. As soon as I get RTH code working relibaly, I'll start testing "RTH on failsafe" feature.

digitalentity, 2015-10-24:
Added preliminary support for RTH on failsafe event to new navigation. 
Regression: When RTH can't activate (no GPS etc) and failsafe requests RTH, `getStateOfForcedRTH()` called from here https://github.com/digitalentity/cleanflight/blob/f897a0e55a6d5b1608e6bc36ddcf9e57a0839816/src/main/flight/failsafe.c#L258 will always return `RTH_IDLE` and the machine will enter `FAILSAFE_LANDED` state, disarm mid-flight and drop out of the sky.

closed 2017-02-25 by hydra