cleanflight/cleanflight #846: Implement RC Yaw Expo

borisbstyle, 2015-05-05:
I never considered it as really necessary before, but with the latest yaw improvements and very snappy yaw performance on some multirotors and agility to do yaw acrobatics I highly see a need to make Yaw expo available in Cleanflight

here is a blackbox demonstration of how well yaw works on my Mini H setup. You can clearly can see that my yaw input is very minimal when using pid controller 2 with these rate settings:
RC Rate 1.0
Roll rate 0.5
Pitch rate 0.5
Yaw rate 0.4
RC Expo 0.65
https://www.youtube.com/watch?v=zQtE0nrz2Sk

borisbstyle, 2015-05-05:
Ok I did an attempt to implement it. Is there anybody who can help testing/reviewing the code with me before I make a PR?

https://github.com/borisbstyle/cleanflight

borisbstyle, 2015-05-05:
And I made a PR for this

https://github.com/cleanflight/cleanflight/pull/847

hydra, 2015-05-06:
Done

closed 2015-05-06 by hydra