cleanflight/cleanflight #850: modes (aux settings) not restored when restoring CLI on naze32 rev5 acro

JurikRistjouw, 2015-05-06:
When i restore a CLI on a Naze 32 acro rev5, somehow allways the aux modes are not restored. I do a flash with full erase, paste values and save. All is restored but not the aux settings. When i try to paste again, same result. When i paste only the config i want:

aux 0 1 0 1900 2100
aux 1 2 0 1400 1625
aux 2 12 1 1725 2100
save

then it will restore the correct modes again.

stronnag, 2015-05-06:
This is a support question (and  a FAQ) , not an issue. Please try not to waste developer's time with support questions.

You are pasting faster than the FC can process and getting data overrun. Either paste in smaller chunks, or use an external cli backup application like cf-cli <http://www.daria.co.uk/cf-cli/> or the python one described on RC Groups that I can't for the life of me recall.

JurikRistjouw, 2015-05-06:
Even when i use a diff only, it will not get it, i am not trying to waste anyones time tho

stronnag, 2015-05-06:
The python one http://www.rcgroups.com/forums/showpost.php?p=30950883&postcount=4155

JurikRistjouw, 2015-05-06:
thanx for the workaround, but i would rather see it fixed either in CLI restore or configurator restore. I do not mind pasting small chuncks but find it disturbing that the last setting is restored, but those settings in de middle are not. Who knows what settings in the future will not come through. I will try the diff only restore and will update my results. Let's hope i am wrong then. I only want a reliably restore option.

hydra, 2015-05-06:
@stronnag thanks.

@JurikRistjouw the CLI is not a long term solution. Really a better backup/restore process is needed that doesn't require msp or CLI.

closed 2015-05-06 by hydra