cleanflight/cleanflight #851: Plugin-Interface possible?

IljaSchumacher, 2015-05-06:
This is more of a Feature Request.
is it possible to make an interface to override functions by 'plugin'?
A dynamic loading procedure on boot to incorporate stuff overriding system functions placed in a plugin directory? pre compiled stuff against the used source?

stronnag, 2015-05-06:
Unlikely to be practical on a resource limited platform.

It would be less disruptive of developer time to first discuss such ideas on RC Groups or IRC rather than here.

hydra, 2015-05-06:
@stronnag thanks. Agreed.

@IljaSchumacher Not right now, it would be nice to be able to compile and test modules independently for sure, but it would require extensive architectural rework and there are other things to do before we contemplate this.

closed 2015-05-06 by hydra