cleanflight/cleanflight #859: Unable to use Sparky.bin

snakescb, 2015-05-07:
Hi, for the 1.8.x versions for Sparky, there is just a .bin file, no .hex anymore. Unfortunately, also using the 3.0.2 version of the DfuSe File Manager, I am not able to produce a valid dfu file to load to sparky. There is a dfu produced, but this will not download. If possible, please release the Sparky.hex also for 1.8.x versions

defritonc1, 2015-05-08:
I am having the same problem.

defritonc1, 2015-05-09:
The .bin file works.
When converting to dfu, you have to make sure you put the start address of the loaded file as 08000000.

[image]

snakescb, 2015-05-12:
This solves the issue, thanks for the quick help

closed 2015-05-15 by hydra