cleanflight/cleanflight #860: Use modified version of Gimli to fix words_with_underscores in PDF manual

thenickdude, 2015-05-08:
Currently in the manual PDF, when there is a word with multiple underscores in it (like "servo_center_pulse" in the CLI listing) the middle words gets put in italics and the underscores are removed, so the manual can't really be used as an accurate CLI reference.

This is because the default Markdown dialect in Gimli treats underscores as emphasis, even when they occur inside words. GitHub Flavored Markdown, on the other hand, does not. We can run a tweaked version of Gimli to bring its dialect closer to GitHib Flavor.

My fork of Gimli has the small patch applied to it to fix this problem:

https://github.com/thenickdude/gimli/tree/no-intra-emphasis

You can clone it, then enter its directory and run:

```
gem build gimli.gemspec
sudo gem install gimli
```

And you should now have the fixed version of Gimli installed and ready to generate a corrected manual.

Before:

[image]

After:

[image]

pulsar256, 2015-05-11:
Any chance you could integrate it into https://github.com/cleanflight/cleanflight/blob/master/.travis.sh? I have been also experimenting with an alternative pdf generation pipeline: https://github.com/pulsar256/cleanflight/blob/pdf_manual_tryouts/support/manual/build_docs.sh which produces a bit nicer looking pdfs and does not suffer from that underscore issue: http://cleanflight.memoryleaks.org/builds/testing/Manual.pdf

Perhaps it is worth moving to the generate-md -> wkhtmltopdf  approach?

thenickdude, 2015-05-12:
I've added a PR which fixes this now by using a patched version of Gimli!

PR #881 (2015-05-14): Patch Gimli to fix underscores_inside_words in PDF manuals
Previously the words in the centre of a underscores_inside_words compound would be wrapped in &lt;em&gt; tags and the underscores would be removed, this fixes that by tweaking the Markdown dialect.

Fixes #860
closed 2015-05-14 by hydra