cleanflight/cleanflight #861: problems with arming my naze32

kaefert, 2015-05-08:
Hi there! I've got two issues both about arming my motors, maybe their connected:
(And I'm sorry if this is just me being stupid, this is the first quadcopter I'm building myself, please forgive me)

1.) I can't arm using `LOW HIGH CENTER CENTER` for `Throttle Yaw Pitch Roll`.

-) I've checked all those channels in the "Receiver" tab of the cleanflight configurator and adjusted the "Minimum Command" value from 1000 to 1050 (since the lowest I get with my receiver-setup is 1049)

-) I checked that the motors + escs themselfs are okey by testing them in the "Motors" tab.

2.) When I reboot my FC through the configurator (by changing some settings) it always makes the motors arm (or at least they spin slowly). I can't increase their speed by increasing throttle on my controller and I can't disarm by using `LOW LOW CENTER CENTER` for `Throttle Yaw Pitch Roll`. The only way I can make my motors stop is by disconnecting the battery.

The only combination for `Throttle Yaw Pitch Roll` from this page
https://github.com/cleanflight/cleanflight/blob/master/docs/Controls.md
that does "look like" it actually does something for me is Calibrate Acc `HIGH LOW LOW CENTER` which makes the green (=middle) LED on my controller flash for a few seconds.

About my Setup:
I've got a Naze32 or Flip32+ (it looks closer to Naze32 but not exactly and what I've ordered was a Flip32+ )
I've flashed it with Cleanflight "v1.8.1 NAZE 2015-4-2 16:16 (stable)" using the "Load Firmware [Online]" button in the Cleanflight Configurator

And this is what the log in the configurator looks like after a config-change-reboot that causes my FC to "arm" the motors:

```
2015-05-08 @ 10:47:21 -- Serial port successfully opened with ID: 6
2015-05-08 @ 10:47:21 -- MultiWii API version received - 1.8
2015-05-08 @ 10:47:21 -- Flight controller info, identifier: CLFL, version: 1.8.1
2015-05-08 @ 10:47:21 -- Running firmware released on: Apr 2 2015 17:06:11
2015-05-08 @ 10:47:21 -- Board: AFNA, version: 2
2015-05-08 @ 10:47:21 -- Unique device ID received - 0x66fff485755776767100922
2015-05-08 @ 10:49:40 -- EEPROM saved
2015-05-08 @ 10:49:40 -- Device - Rebooting
2015-05-08 @ 10:49:43 -- Device - Ready
2015-05-08 @ 10:49:58 -- Serial port successfully closed
```

thenickdude, 2015-05-08:
`mincommand` is the "idle" value sent to the ESCs when the craft is disarmed and doesn't relate to your Rx throttle in any way. You're causing your ESCs to spin all the time by increasing that value. Set it back to 1000.

`min_check` (settable on the CLI) controls the throttle threshold below which you can perform stick commands. The default is 1100, so if your throttle goes to 1049 at the minimum position you should be able to arm fine.

kaefert, 2015-05-08:
oh, okey. so that explains the spinning after a reboot (while it seems strange to me that it doesn't do that on a "first boot" when connecting the battery)

thenickdude, 2015-05-08:
Please take a screenshot of the Receiver tab when you are trying to arm.

kaefert, 2015-05-08:
[image]

thenickdude, 2015-05-08:
Your yaw doesn't reach high enough to clear the max_check threshold of 1900, so it's not being detected as a stick command. On the CLI, reduce max_check to 1850 or 1825 or something:

```
set max_check=1825
```

kaefert, 2015-05-08:
that worked! thank you very much for the help! and also thanks a lot for contributing to this awesome FC-software! :)

closed 2015-05-08 by kaefert