cleanflight/cleanflight #864: custom mixer, different front to back pitch values will not pass sanity check

SinMike, 2015-05-09:
When inputting values for an sk450 deadcat the pitch settings for the engine 1 and 3 are different from engines 2 and 4, when inputting the values into cli the sanity check will not pass on pitch
example:

# cmix
Custom mixer: 
Motor	Thr	Roll	Pitch	Yaw
#1:	1.000	-0.767	0.837	-0.800
#2:	1.000	-1.000	-0.428	0.769
#3:	1.000	0.767	0.837	0.800
#4:	1.000	1.000	-0.428	-0.769
Sanity check:	OK	NG	OK	

after some experimenting I found out that this is probably because the sanity check does not take into account that for pitch settings work differently from the other settings that will become 0 when entered correctly.
as an example when i enter the following it will pass the sanity check

Custom mixer: 
Motor	Thr	Roll	Pitch	Yaw
#1:	1.000	-1.000	1.080	-1.000
#2:	1.000	-1.310	-0.560	1.310
#3:	1.000	1.000	-1.080	1.000
#4:	1.000	1.310	0.560	-1.310
Sanity check:	OK	OK	OK	

however this is not the intension of this mix.

the issue does not occur when all pitch value are equal as then the calculation will by accident not throw the error.

thenickdude, 2015-05-12:
All the sanity check does is see that the sum of values for roll, the sum of values for pitch, and the sum of values for yaw each equate to zero. As far as I can see, there is no meaningful sanity check possible for asymmetrical craft.

hydra, 2015-05-15:
Should the sanity check be removed then?

SinMike, 2015-05-15:
Hi,

The sanity check for yaw and roll is valid as in principal the value combined should always be 0. However for asymetrical frames, pitch does not have to be 0 and so the sanitycheck currently does not provide a valid result. To do a valid sanitycheck on these mix values is almost imposible, this can in my opinion only be done with all values necessary to create the mix. 
The sanity check could give some confusion to if a mix will work, i had to look through the code to find that it was just a simple sum of all values. 

On a side not, maybe you know the answer;
When calculating a mix, should you use center of thrust as a basis for the measurements or the center of the flight controler?

Kind regards Michel

thenickdude, 2015-05-15:
@hydra I think someone mentioned that they were working on a GUI display/editor for the motor values (i.e. show the motors positioned according to the mix). That would be way better than any FC sanity check.

FWIW the advice I've seen on the forums has always been to ignore the sanity check because it isn't useful. If your craft is completely symmetrical then you probably don't need a custom mix.

borisbstyle, 2015-05-19:
Just ignore the sanity check. It tells you here that your pitch is not symmetric so it fails. Your front motors are affecting pitch differently than rear motors. 
There is really no way to detect this better. Perhaps the name "sanity check" is not the most suitable name.

closed 2017-02-25 by hydra