cleanflight/cleanflight #865: Incorrect GPS longitude value in HoTT telemetry

Q-Branch, 2015-05-09:
Hi,
I have been experimenting with the HoTT telemetry using my Naze32 board, a Graupner MZ-24 transmitter, a Graupner GR-12S receiver, a UBLOX based GPS, and Cleanflight 1.8.1.

The Latitude shown on the TX's screen is correct, however the Longitude is shown as being W 655'31.6670  (i.e. 655 degrees, 31.6670 minutes).  Something seems to be wrong here and my first guess would be that it's an issue relating to the formation of the telemetry data when longitude is -ve from 0 degrees (i.e. anything "West").  The GPS position reported in the GPS tab of Cleanflight Configurator is shown correctly as -0.0983 degrees (W).

I have had a quick look through the hott.c code but couldn't see anything glaringly obvious.  Without first being able to check whether this bug also occurs for telemetry other than HoTT, it's a bit difficult to know where to start looking for the cause.

Does anyone live West of 0 degrees and have the ability to check their telemetry from GPS for a non Graupner / HoTT perspective please?  This would give us a starting point.

Kind regards,
Rich

hydra, 2015-05-12:
you could try hardcoding some values in the HoTT code for the GPS values and seeing what the transmitter shows on the screen and working from there.  There's also some unit tests for the HoTT code which might help.

Q-Branch, 2015-05-12:
OK, that's a good idea.  It will have to wait for a while though because sadly I don't currently have a board with a debug header on it, and no current budget to purchase one specifically for dev purposes.

I like the look of the newly announced boards though.  Great job!  :-)

R

closed 2017-02-25 by hydra