cleanflight/cleanflight #867: Porting to other arm architectures. (MINI54ZAN/MINI54ZDE)

forkoz, 2015-05-09:
I've been tooling around with bradwii on tiny quads using MINI54ZAN/DE. Unfortunately its a bit buggy and has very little development. Stumbling onto cleanflight it looks to be much more robust and supports most of the hardware already. Everything a bit different is already spelled out in the bradwii port for the most part.

So why am I here? Well I don't quite know where to start and if this is a worthy endeavor. The core itself only has 16kb of flash, would cleanflight fit with compass/gps/etc pulled out? Finally, just how much of the code base would have to be altered to work with the mini51 libraries. Experienced devs, what say you?

thenickdude, 2015-05-10:
The current smallest target is CJMCU, which has 64kB of flash and a very stripped down feature set. It only supports the quad mixer, and there's no support for GPS etc. The fanciest feature supported is Serial RX. Even then the total used flash size is 62208 bytes.

Trying to fit Cleanflight on a 16kB target would be a waste of development time.

forkoz, 2015-05-10:
On the hubsan board I'm using I dont have a serial port to do runtime configuration. It may be smarter to compare flying code from bradwii to cleanflight by what you're saying, however.

hydra, 2015-05-15:
Support for non STM32 platforms is not on currently on the road-map.

Also, when CPU's as powerful as the F3 are available for such a small amount it makes no sense to try and fit Cleanflight on targets that have < 64k of storage.

I would welcome porting cleanflight to other more capable platforms for which good cheap development and commercial hardware already exists.  However the codebase and build system would need a higher level of hardware abstraction otherwise the resulting merge would likely be a mess of `#ifdef`s.

closed 2015-05-15 by hydra