cleanflight/cleanflight #870: GPS on soft serial

James205mann, 2015-05-10:
Forgive me if I'm being a muppet but I've looked every where to Che k I'm not but haven't found a solution. I'm trying to connect a GPS to softserial 2 for osd use as I'm using a spectrum satellite on uart 2. This is a full naze32.  The gui implies this will work which it does immediately after the GPS is configured on u-center. However after leaving it an hour or so it will no longer find any satellites.  GPS passthrough in cli returns gibberish. It does not matter what baud rate I configure in u-center or cleanflight. It will work temporarily but only find 4 or 5 sats in cleanflight and then stop finding any. Works fine in just u-center.  Any help much appriciated.

tricopterY, 2015-05-10:
By definition SoftSerial is inefficient and should only be used for slow updates like Telemetry etc. I would only try to operate at 19200 Baud Rate. See, if that improves things for you. If you are not out in open space, you will not get a good fix/lock. Also, have your GPS module mounted away from any wiring or Carbon/Metal material. Normally, on a pedestal is the best. My two cents worth.
...BTW, if all fails test using your UART-2, just to establish some datum for comparison.

hydra, 2015-05-12:
@James205mann please provide a configuration dump

James205mann, 2015-05-12:
Thanks for the help.

@ tricopterY; Ive tried combinations of baudrates from 9600 to 19200 setting the same baudrate in u-center and cleanflight. No joy. I'm only hoping for the GPS info on my OSD so not worried about a higher than 9600 baudrate, but it would be nice.

@hydra;here is my entire dump, didn't want to miss something.

# dump

# version
Cleanflight/NAZE 1.8.1 Apr 2 2015 / 17:06:11 (988ae2d)
# dump master

# mixer
mixer TRI

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature ONESHOT125
feature BLACKBOX

# map
map TAER1234

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
[... 253 lines cut ...]
set i_alt = 100
set d_alt = 250
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 120
set rc_expo = 65
set thr_mid = 60
set thr_expo = 30
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 45
set tpa_rate = 50
set tpa_breakpoint = 1600

#

James205mann, 2015-05-19:
Sorry for the delay in getting back, I have now tried this with a FrSky PPM receiver and the GPS on UART 2 with the same behavior. Re flashed Cleanflight but that had no effect. I have tried a GPS analyzer and don't appear to be getting interference from how my components are sited. My next step will be to try a new GPS module even though the module works fine in u-center. Am I missing something else that might resolve this? Thank you again.

tricopterY, 2015-05-19:
As per your statement, after an hour or so no satellites are found. What is your supply voltage reading after that? 
And if you disconnect and reconnect using the same power pack, after one hour or so, does the GPS get a fix and operate normally again?

tricopterY, 2015-05-19:
How do you power your gear? (if Lipo only - do you have 12V LEDs attached)
Does any component on the FC feel hot, when touched?
I am wondering if it's power related (voltage regulation), rather than firmware.

James205mann, 2015-05-19:
Hi tricopterY, I now have the GPS wire on uart 2 with a cppm receiver. No change in behaviour. I have a pdb with 5v and 12v bec, 5v powering the naze and the GPS powered from an unused motor output. No less and nothing is getting warm. 

Sorry, a more accurate description of the problem is that the GPS works fine on u-center when connected via a uasp (the USB to 6 pin connector?) and it's still ok when I then solder it to the naze. I then take it outside and get a flight or two with GPS signal. That's also ok. But if I leave it off for an hour or so, no GPS. Connect it to u-center or cleanflight and I'll get another flight or two with GPS signal. As I said my next step is to try another GPS module in case that is corrupt.

tricopterY, 2015-05-19:
@James205mann :smile:

tricopterY, 2015-05-19:
@James205mann thanks for the info.

James205mann, 2015-05-28:
Sorry for the delay in getting back to you. I have been away and still waiting on a replacement GPS module. 

My ESCs are all opto.

When I use the command gpspassthrough in CLI i get a string of non sensible data. I.E. not nmea sentences. Is this right?

James205mann, 2015-06-03:
I have tried this with a new GPS module, a ublox-m8, on uart 2 with the same issue as before.

I connect the GPS to u-center and configure it, connect to the tricopter and verify it is working on Cleanflight. Take it outside and it works fine. Try it again the next day and it no longer finds sats. I'm at work now but I will try a bit more later, although I am totally lost now.

closed 2017-02-25 by hydra