cleanflight/cleanflight #874: SmartPort current multiplied by 10 on Taranis

ledvinap, 2015-05-11:
Current reported over SmartPort is wrong, displayed value is 10x larger than actual.

There was some discussion about current resolution - is this intended behavior?

hydra, 2015-05-11:
What version of OpenTX are you seeing this on?  And is this already addressed in the 'smartport fixes' PR?

hydra, 2015-05-11:
This is the other PR with some SmartPort (S-Port) fixes #821

ledvinap, 2015-05-11:
Tested on 2.0.17, the displayed values is 10x larger ... 

#821 does not address this

kilrah, 2015-05-17:
No it would be the Curr value. 

I don't have a current sensor to connect to my Naze so didn't test that. I use a separate FrSky current sensor that is hooked together with the Naze on the smart port.

closed 2017-02-25 by hydra