cleanflight/cleanflight #877: Incorrect GPS coords with SmartPort telemetry

Netsplits, 2015-05-11:
I'm getting incorrect GPS coordinates when using SmartPort telemetry. I've made some changes to the code and now works OK for me. Tested on OpenTX 2.0.15 and 2.0.17 using GitHub master code from (2015-05-03) aprox.

I'm using a U-Blox 6 GPS module connected to soft serial 1 at 9600 bauds with UBLOX protocol.

The code I've edited is on smartport.c function "void handleSmartPortTelemetry(void)", search for case FSSP_DATAID_LATLONG replace with this:

```c
#ifdef GPS
            case FSSP_DATAID_LATLONG    :
                if (sensors(SENSOR_GPS) && STATE(GPS_FIX)) {
                    uint32_t tmpui = 0;
                    // the same ID is sent twice, one for longitude, one for latitude
                    // the MSB of the sent uint32_t helps FrSky keep track
                    // the even/odd bit of our counter helps us keep track
                    if (smartPortIdCnt & 1) {
                        tmpui = (abs(GPS_coord[LON] / 100) * 6) | 0x80000000;
                        if (GPS_coord[LON] < 0) tmpui |= 0xC0000000;
                    }
                    else {
                        tmpui = abs(GPS_coord[LAT] / 100) * 6;
                        if (GPS_coord[LAT] < 0) tmpui |= 0x40000000;
                    }
                    smartPortSendPackage(id, tmpui);
                    smartPortHasRequest = 0;
                }
                break;
#endif
```

Also I noticed GPS altitude was 10x greater than real, change this line:

```c
smartPortSendPackage(id, GPS_altitude * 1000); // given in 0.1m , requested in 100 = 1m
```
for
```c
smartPortSendPackage(id, GPS_altitude * 100); // given in 0.1m , requested in 100 = 1m
```
Complete case should be this way after edit:
```c
#ifdef GPS
            case FSSP_DATAID_GPS_ALT    :
                if (sensors(SENSOR_GPS) && STATE(GPS_FIX)) {
                    smartPortSendPackage(id, GPS_altitude * 100); // given in 0.1m , requested in 100 = 1m
                    smartPortHasRequest = 0;
                }
                break;
#endif
```

hydra, 2015-05-12:
Can you have a go at creating a pull-request for these changes?  Check the developer documentation and/or ask in the IRC channel if you need help.

Netsplits, 2015-05-12:
Done! It's the first time I'm using GitHub, so I hope I've done it correcty...

Please feel free to close this comment if all is correct. Thanks.

closed 2017-02-25 by hydra