cleanflight/cleanflight #878: Add sonar values to blackbox recording

alvins82, 2015-05-12:
Currently it looks sonar values are not recorded as part of blackbox logs. Would be good to see sonar values in blackbox for debugging alt-hold issues.

thenickdude, 2015-05-12:
Do you have a sonar? If so, please test this new build out for me and post the resulting log, it should add logging of sonar data:

https://github.com/sherlockflight/cleanflight-dev/releases/tag/blackbox-sonar

alvinpb, 2015-05-12:
Yes - will test and report back shortly.

alvins82, 2015-05-12:
Looks like it works nice. Screenshot and log attached.

Based on the blackbox log looks like I may have an unrelated issue with Sonar in - values seem to be jumping around.

https://dl.dropboxusercontent.com/u/7607766/blackbox_log_2015-05-12_174111.TXT

[image]

thenickdude, 2015-05-12:
Hm, those are some pretty big jumps. Do you see jumps like that in the Configurator's readout of the sonar?

From the code, it almost looks like if it sends out one pulse but hears two echos, it'll report two distances (first the shorter distance and then the longer distance). If you're indoors then it could be hearing a secondary echo off a wall. If so, this is probably a bug.

alvins82, 2015-05-12:
I am indoors and on carpet.

I just checked the configurator readout and no jumps. I am guessing this is due to the low resolution of 10ms of the configurator whereas the blips in blackbox are only for a few loop iterations.

thenickdude, 2015-05-12:
I've now merged this feature with e6a40d732d8f303599b56f2848f3d262d3a7218b. I'll open a separate ticket to investigate the sonar blips.

closed 2015-05-12 by thenickdude