cleanflight/cleanflight #879: Enable BlackBox logging by switch?

NL-Splinter, 2015-05-12:
Would it be worthwhile and possible to enable/disable Blackbox logging via a Tx switch? For tuning, I mostly need just some inflight data, not a full logged run. I'm logging on the full Naze, so also for storage reasons I want to be able to log just short parts of some flights.

On RCG this is requested too, though not replied to as I could find.

Thanks in advance!

thenickdude, 2015-05-12:
I can definitely see value in such a switch. However I think it's probably not advisable to allow logging to begin while in the air without further testing. The blackbox initialisation code attempts to break its workload up into many small chunks in order to minimise disruption to the looptime, but this was never intended to be executed in the air and wasn't tested for that situation, so could cause flight instability.

ledvinap, 2015-05-12:
@thenickdude : Maybe it would be better (and easier) to use switch to enable blackbox recording - emit header when arming as usual, but skip writing frames until switch is activated.
Then just make sure that I frame is emitted when logging resumes

NL-Splinter, 2015-05-12:
Hi Nick, fully agree about the worries over inflight disruptions! And yes please, you/we/me should thoroughly test it. In general I think any inflight switch use goes with some level of attention or caution. Can we define 'disruption' in terms of maximum allowed timeframe? Would a 100 msec disruption for this specific switch be reasonable for example?
If there's something I can do or test, pls let me know.
Cheers!

thenickdude, 2015-05-12:
Looptimes are around 2-3ms, so any delay of 8ms or longer makes me concerned (as I don't think many of the flight algorithms expect 3 loop iterations to suddenly go missing). 

Blackbox startup takes quite a while (mostly because it spreads its workload out into so many pieces) but any one of those pieces could be unexpectedly large (because it's never been examined that closely). Or it could all be perfectly fine, but it needs testing first!

@ledvinap that's quite an interesting approach and would indeed be easy to implement.

hydra, 2015-05-12:
yes, my first thought was the same as @ledvinap 's

fedorcomander, 2015-06-23:
@hydra @ledvinap  +1... Mode switch - BB_IDLE, BB_ERASE_FLASH (When NOT_ARMED)... When arming - always write header, after that, when IDNE - do not accept new events/do not log.

borisbstyle, 2015-06-29:
Perhaps the first safe implementation would be to only check when arming! I think most people would like to choose flights they want to log and not in flight changing. Well at least I don't see a point enabling inflight, but rather before you take off.

fedorcomander, 2015-06-29:
Well. You can keep it turned off... So BB will log header and start beep - end of the log events. But if you enable it (on the ground or inflight) the log will be stored. By default its ON, so keep old logic ;)

NL-Splinter, 2015-06-29:
The practical background from my initial request: I go to the field to fly, testing the feel of my tune and collecting some data. If I dont bring my laptop, I can only fly half a battery with logs enabled and I have to go back home. Which is a waste of traveltime and the other 5,5 batteries!
Anyway, I'm fully confident any solution like discussed here gets to a better way for logging, so please continue!

borisbstyle, 2015-06-29:
You can also get openlog for example if you want to do some serious tuning. Than you can compare all the flights during the tune.

But that's not the point here. It's good to have a possibility to enable/disable.

borisbstyle, 2015-07-08:
Any progress with this?

fedorcomander, 2015-07-08:
not here. u?

borisbstyle, 2015-07-08:
No. I am kind of missing it :D

fedorcomander, 2015-07-08:
I will take a look then. :)

NL-Splinter, 2015-07-08:
Much appreciated!

fedorcomander, 2015-07-09:
Ok... I added the switch, to enable BB logging. Its being checked during arming. If switch in active position, blackbox will start logging as usual. If switch inactive, there will be no log :) During flight switch is not being checked in any way.

[image]

I wait till @borisbstyle  will check it in flight and approve and  make PR out of it.

ledvinap, 2015-07-09:
@fedorcomander : Is there any reason to abandon blackbox switch gating  approach (produce headers as usual, log frames only when switch is active)?

fedorcomander, 2015-07-09:
@ledvinap: Just to make it conditional for now... So internal flash will be not used for every flight, but only when needed. If this is not enough i will add gating too.

NL-Splinter, 2015-07-09:
Nice! To retain backward compatibility, shouldn't "active" be the default state, and the switch be used to disable logging?
Anyway, once this goes to production I'm willing to update blackbox.md documentation.
Thanks!

fedorcomander, 2015-07-09:
Modes are enabling something, and i think its wise to keep it consistent. If somebody wants to log, some actions needs to be taken anyway (feature, denominator, device, ports etc), so extra mode config will not do extra troubles. :)

I will check later on today gating approach as @ledvinap suggested. Seems nicer that way after all.

thenickdude, 2015-07-09:
It seems like this makes recording harder, because people would either have
an AUX channel spare, or figure out that an unused AUX channel probably
sits at 1500 so they can add a range for BLACKBOX in the centre of it in
order to start recording. I'd prefer a BLACKBOX_PAUSE switch.

fedorcomander, 2015-07-09:
But idea is to enable bb when needed in place of disabling it when not needed? I use one channel for beeper and blackbox... falls in nicely imho. 

If we check the switch only once, we can REUSE range of beeper, and turn on beeper when arming to get bb rolling... But it will record whole flight.

Ok, @thenickdude , final word is yours. :)

ledvinap, 2015-07-09:
Isn't it enough to include full range as default? Or check if at least one range for this mode mode is defined?

fedorcomander, 2015-07-09:
What we are trying to achieve? To force BB log always by default? It might be good thing for developers, but most of us have small onboard flash which is almost unusable because its full in 5 minutes. Adding openlog device is a nice thing, but also requires quite a lot tinkering, flashing, soldering, formatting in the process. So, initial request from @NL-Splinter was to have logging conditional, so he (and actually me) can choose flight to log by flipping the switch. Don't forget, downloading 2/8/16mb (chip takes quite some time, so erasing it. Its totally worth it, but still i would prefer to have log of what i need... My problem now also that logging to openlog with very low looptimes needs to be with higher denominator... Seems SPI chip is faster (may be i am wrong). I understand that for developer, having whole logs is more interesting for analysis of the failures. But for me, if i just wanna tune and see 1 minute of changing in behaviour of the quad while flipping inflight adjustments is more then enough.

Of course, its also possible to just disable feature from cli for that ;)

borisbstyle, 2015-07-09:
I still would like to have enable/disable option. Even when I use openlog I sometimes end up having 50-100 files after day of testing, while I just needed a few of them to analyze. Guess how much time I spend looking for the needed file!
Perhaps a stick command can be used as well to enable or disable with beep pattern as confirmation? That won't use any AUX channels.

fedorcomander, 2015-07-09:
@ledvinap Isn't it enough to include full range as default? Or check if at least one range for this mode mode is defined?

Actually its a good idea. If no range for BB defined, have CURRENT functionality - log always. If at least one range defined, log ONLY WHEN range is active ;)

thenickdude, 2015-07-09:
>Actuallyi think its a good idea. If no range for BB defined, have CURRENT functionality - log always. If at least one range defined, log ONLY WHEN range is active ;)

+1

fedorcomander, 2015-07-09:
And i think logging whole flight is nicer then bits and pieces.. Well, in viewer it looks kinda weird.

[image]

I was flipping switch blackbox on and off and spinning quad in hands ;)

ledvinap, 2015-07-09:
@fedorcomander : There is already mechanism in viewer to render skipped (broken) frames correctly. All that is needed is to trigger it when frame delta time is large.

fedorcomander, 2015-07-09:
Thanks Petr! I will look into that!

thenickdude, 2015-07-10:
@fedorcomander I don't think logging slow frames will be useful during pauses, they're mostly information that isn't useful to know about without the rest of the log being recorded as reference. 

Events like in-flight adjustments will naturally continue to be recorded even if the Blackbox's main loop stops, and these will indeed be useful in order to keep track of what PIDs you are using when recording begins again.

fedorcomander, 2015-07-10:
@thenickdude Yap... Thats the idea. Should events also get loopIteration parameter now??? If there is no I or P frames, there is no way to determine event timestamp.

thenickdude, 2015-07-10:
Eh, I don't think that a timestamp for those events is even important if no other data is being collected at the same time. They're mostly useful for arriving at known PID settings once recording begins again.

fedorcomander, 2015-07-10:
True... But current model expects them to get time from previous frame ;) And exist within the chunks.. Which are 4xI frames, which aren't there... Oh my. Don't think we can log them when BB paused. And add timestamp is extra bytes which is not desirable. Am i correct?

thenickdude, 2015-07-10:
But that's fine, they'll all just end up with the same timestamp and get processed in order.

fedorcomander, 2015-07-14:
Just added #1122

fedorcomander, 2015-07-14:
@thenickdude Basically for rendering i removed time and loopiteration checks and added gap starts marker to some frames. It works, but i am not sure its right thing to do. Can you take a look?

fakatwork, 2015-07-22:
I'd like if you could keep the on-demand enable/disable in order to just record flight segments (vs always whole flights) It's particularly useful for those having limited memory (ex: Naze full)

I don't mind if the switch is reversed to keep backwards compatibility though (ie it's a pause switch and not a enable switch) this way if you don't configure the switch it would be always recording as usual.

fedorcomander, 2015-07-22:
Its implemented this way, If no range defined - its always ON... If range defined, it will log only if range is active... u can do segments... @thenickdude adding new events to have it rendered better/proper now.

rscarawa, 2015-07-22:
Cannot wait for this to come out.

thenickdude, 2015-07-23:
I've added #1157 to implement this now!

closed 2015-09-09 by thenickdude

fedorcomander, 2015-09-09:
Hi Nick!

From MY real world experience... Using the on off switch is kinda not handy... every arm takes 64k of space on the internal flash... When you need to log it, usually its already full... May be if on arm all empty logs will be deleted will help.  Still liked it more when it was ON/OFF before arming...

my 2 cents.

bdoiron74, 2015-09-10:
Could a stick command (when disarmed) that erases the log be useful?

thenickdude, 2015-09-10:
It's every poweron / arm / poweroff cycle that causes 64kB to be consumed, rather than on every arm. This is because on powerup a binary search is executed to find the start of the free space on the flash, and this is configured to look in 64kB chunks.

I just benchmarked the search process and it looks like it takes about 35 microseconds per search. Searching 2MB of flash at 64kB blocksize requires at most 5 searches, so about 175 microseconds. That's a tiny amount of time, so it should be fine to reduce the blocksize to 2kB, pushing the maximum startup overhead up to a still-tiny 350 microseconds. Searching an 8MB flash chip increases this to 420 microseconds.

I should be able to do final testing and commit that tonight.

fedorcomander, 2015-09-10:
that is great improvement!!!  thanks!

thenickdude, 2015-09-11:
The wasted space reduction has now been committed as 6c231e189b5f02a30167dcef97eed0a3d441b69c.