cleanflight/cleanflight #880: Investigate sonar problem(?) using new Blackbox logging support

thenickdude, 2015-05-12:
Blackbox logging support for raw sonar data has been added in the latest commit, e6a40d732d8f303599b56f2848f3d262d3a7218b. @alvins82 recorded this log:

https://dl.dropboxusercontent.com/u/7607766/blackbox_log_2015-05-12_174111.TXT

[image]

Notice that their sonar has periodic spikes of a large measured distance, which then fall back to the baseline height. This test was recorded while flying indoors. It could just be a problem with this sonar setup, or it could be a general problem with the sonar driver. We need more logs to find out for sure! (I don't have a sonar myself to investigate with)

hydra, 2015-05-12:
I'll try and get some more data.  I have a hunch that the driver needs a bit more work.

alvins82, 2015-05-12:
Suspecting it might be EMF interference from TBS core (which was directly above the sonar) I put foil below the TBS core but this didn't resolve the issue.

Another log linked.

https://dl.dropboxusercontent.com/u/7607766/blackbox_log_2015-05-12_211408.TXT

hydra, 2015-05-15:
@thenickdude cant remember of the top of my head, but are we also logging the output of the alt-hold pid controller?  Not strictly related to the sonar driver but would probably be very useful.

thenickdude, 2015-05-15:
No, we're not currently logging that. I haven't seen a single log posted in altitude hold mode so there's been no call for extra data logging yet. It could well be that all the problems are due to crazy sensor readings and we don't have to look any further.

thenickdude, 2015-05-18:
@alvins82, if possible please retest with the new 1.9.0 release candidate:

https://github.com/cleanflight/cleanflight/releases/tag/v1.9.0

The b739a57ea23f68a0e554bc6507638f657ab1b7a1 commit that's included in this release fixed a Blackbox bug that could have impacted logging of Sonar data.

alvins82, 2015-05-18:
@thenickdude give me a day or so. Will update this comment.

anees042, 2015-06-30:
I also faced the same problem, i didn't got time to switch off as i was nearly on 2 meter height and the quad crashed to the ground.

alvins82, 2015-07-18:
I have a new sonar module which I attached and the same issue exists. I have also added a baro to my naze32 acro now and interestingly it shows the same issue but I'm not sure if that is because some issue with logging/interpretation of the baro var (e.g. is calculated from sonar input with small delay). Anyway - new blackbox log attached.

https://dl.dropboxusercontent.com/u/7607766/blackbox_log_2015-07-18_103145.TXT.zip

MauroMombelli, 2015-08-25:
can you test the module using directly another board like arduino just to exclude a sensor fault?

also would be interesting a log of the tension from the battery, or giving power to the electronics from a dedicated battery.

Another test is to try without battery but only USB (should be enough) and see if the problem is still here.

Please note that if you use sonar on thing like clothes, for example, it will see "infinite" distance

alvins82, 2015-08-25:
I will have to rewire/rejig the power for it as it gets its power from the ESC rails on the naze32 at the moment. First thing I will try is to change it's position and also change how its wiring is routed to rule out any interference issues with other wiring/components. If that still shows issues will rewire the power also and report back. 

On the note above with the baro - once I disable sonar, baro doesn't exhibit any problems so issue isolated to sonar issue.

MauroMombelli, 2015-08-26:
The fact that sonar also influence baro  is giving more clue to power issue.

Anyway do a test using only USB power and moving the quad in the limit of
the cable, to see it the error can be easily reproduced that way

alvinpb, 2015-08-26:
Mauro - are you available for a quick chat - can use Skype? Email me at alvins82 at gmail dot com

alvins82, 2015-08-26:
Ok ran some more tests. Tried moving cable to completely outside frame. Also tried powering from separate 5v power source. No go - same issue.

I suspect it could be something with my particular naze board so unless other people are also experiencing this it might be time to close the issue?

Ravid824, 2015-10-04:
Same issue on my Naze32. Spikes show frequently regardless of the SONAR locations of power supplied. It looks like a problem with the driver which start a ping before last echo is completed.

martinbudden, 2015-12-07:
Yes, I believe there is a problem with the driver, in that the echo pin can get 'stuck' high and needs to be reset. I have an idea of how to do this and will look into it.

ZMR250, 2015-12-31:
Hello all. I have also same issue with adding sonar. Im using ppm reciver so pins 7,8 1K resistors all as in the video mentioned above. I do have 5 sonar modules HC-SR04 and they all do work with my arduino but not with the naze32 rev5, 1.11.0 firmware. Tilt the naze board and you get spikes in sensor window. Im pretty shure problem isnt in hardware or interfearence it is somewhere in code.  
Best regards.

Ravid824, 2015-12-31:
Your Naze board is not hooked to the drone? meaning when you tilt Naze board the Sonar doesn't move with it?
My Naze and Sonar are hooked to the drone. When I lift the drone I can see good reading results. I will try to tilt the drone and see if there are spikes.

ZMR250, 2015-12-31:
Noup i took it off. I have just reciver 5v supply and sonar connected. Ravid824 what firmware you run on your board and what board you use?

Ravid824, 2015-12-31:
1.11.0

Ravid824, 2016-01-05:
After verify that the signals and 5V are clean of noise coming from the motor, I made few tests with the HC-SR04 driven from a standalone Arduino. The Sonar worked quite well while the sonar was on the table. Then I placed the Sonar  on the drone and start the motors (still driven from the Arduino).
I found very interesting results. On a small detection range (30cm) it worked well, while when I increased the range to about 100CM , false reads showed frequently.
I believe that it is because of the Soanr vibrations generated by the motors. The sonar has two transducers (which are similar to speaker diaphragm which convert mechanical vibrations to sounds). 
Motor Vibrations cause the transducers to vibrate and generate false echo results.
I think that CF set the MAX range to 300 cm which increase the false echo reading dramatically.
Do anyone has an idea on how to eliminate the soanr vibrations?

Ravid824, 2016-01-08:
Well, after softening the vibrations the drone is keeping height below 3 meter and it quite stable.
Still there is a small problem when it transition from Sonar to Baro, but it is reasonable.
The pictures below shows the sonar with rubber dumber to reduce vibrations.
[image]
[image]

closed 2017-02-25 by hydra