cleanflight/cleanflight #882: custom mix: arm doesn't work

kaefert, 2015-05-12:
Hi there!

I've seem to got a short-circuit on my first motor output - I can't find anything on the outside, but as a workaround I thought I'll use a output 2 to 5 instead of 1 to 4 for my "Quad-X" configuration.

I've used these commands:
```
mixer custom
cmix reset
cmix 1 0, 0, 0, 0
cmix 2 1.0, -1.0, 1.0, -1.0
cmix 3 1.0, -1.0, -1.0, 1.0
cmix 4 1.0, 1.0, 1.0, 1.0
cmix 5 1.0, 1.0, -1.0, -1.0
```
with and without the cmix 1 .. line. It does pass the sanity check, and all the ESCs are made to boot when connected as configured. But when I arm my FC it only turns on the green LED but doesn't start spinning the motors.

Is there something more I need to enter into the command line to get this working, or why won't it work?

hydra, 2015-05-12:
i'm running a cmix on my vtail with no issues. please provide a full config dump

kaefert, 2015-05-12:
hi there! thanks for the reply!
I've already started anew (with "full chip erase" checked)
What I need to config to make my setup work (mostly)
-) Tab Ports: Check UART2 GPS
-) Tab Configuraton: Activate GPS UBLOX

--> with that state 3 motors work, I can arm them and everything works.
then I switch to the CLI and enter what I've posted in the first post.

After a reboot by entering `save` in the CLI arming will turn on the green LED instead of starting to spin the motors.

This is what the CLI outputs when I enter `dump`
http://pastebin.com/r6T8LavN

kaefert, 2015-05-12:
ah okey I think I got the problem now
it seems he ignores the command `cmix 1 0, 0, 0, 0` and won't accept anything I enter from 2 to 5
since the reply to it is:
```
Custom mixer: 
Motor	Thr	Roll	Pitch	Yaw
Sanity check:	OK	OK	OK	
```

when I instead enter `cmix 1 1.0, -1.0, 1.0, -1.0` then I get
```
Custom mixer: 
Motor	Thr	Roll	Pitch	Yaw
#1:	1.000	-1.000	1.000	-1.000
#2:	1.000	-1.000	1.000	-1.000
#3:	1.000	-1.000	-1.000	1.000
#4:	1.000	1.000	1.000	1.000
#5:	1.000	1.000	-1.000	-1.000
Sanity check:	NG	NG	NG	
```
but I gess the "NG" thing tells me that it doesn't accept this config..
what could I enter instead that will be accepted for motor output 1 that doesn't work?

kaefert, 2015-05-12:
hmm, okey, this might be unrelated since this is the first time that I have all the parts neccesary to fly this quadcopter, but with this config it is really slow to react on my inputs and therefore impossible to get it off the ground more than a few centimeters without crashing it..
Maybe seeing it helps identify the problem: http://koega.no-ip.org/misc/quadcopter/MVI_3377.MP4

atomiclama, 2015-05-13:
Can you try
cmix 1 1.0, 0, 0, 0
cmix 2 1.0, -1.0, 1.0, -1.0
cmix 3 1.0, -1.0, -1.0, 1.0
cmix 4 1.0, 1.0, 1.0, 1.0
cmix 5 1.0, 1.0, -1.0, -1.0

this is your cmix from the top but with the throttle for motor 1 set. 
It passed the sanity check on my setup just now.
I used a similar config some time back on my quad for the same reason as you. it worked fine.

kaefert, 2015-05-13:
yes! that worked! thanks!!!
koega.no-ip.org/misc/quadcopter/MVI_3444.MP4

closed 2015-05-13 by kaefert