cleanflight/cleanflight #883: Blackbox sync beep not correct anymore in refactored beeper code

Warudo, 2015-05-12:
The Blackbox feature provides a short sync beep when arming on 1.8.1, so you can line up the .png frames generated by blackbox_render and the flight video in a video editor.
This was very useful in a video editor that can show the audio waveform on the timelime. Takes only a few seconds to line them up.

But since the beeper code was refactored, as of commit e6a40d732d8f303599b56f2848f3d262d3a7218b, I'm not getting a clear beep to which I can sync anymore. Arming seems to produce a long beep follwed or overlapped by a short beep now. The blackbox and flight videos are in sync when the blackbox video starts somewhere in the middle of the long beep, but the exact position varies.

Examples:

[oldbeep.mp3](http://notaserver.org/pubhost/cleanflight/oldbeep.mp3) (Link to my server since github doesn't allow audio file uplading): How the beep sounds on 1.8.1
Annotated timeline of my video editor: [image] The blackbox and flight video are in sync when arranged like shown. The blackbox beep is almost exactly 1 frame long at 30 FPS.

[newbeep.mp3](http://notaserver.org/pubhost/cleanflight/newbeep.mp3): How the beep sounds as of commit e6a40d732d8f303599b56f2848f3d262d3a7218b
Annotated timeline of my video editor: [image] The blackbox and flight video are in sync when arranged like shown, they are not in sync when blackbox video starts with the second, short beep, then the blackbox video is late.

hydra, 2015-05-12:
Thanks for testing and this excellent bug report.  very clear.

thenickdude, 2015-05-12:
Blackbox no longer adds a beep of its own, instead it just records the time stamp of the regular beep which is emitted on arming with the new beeper code. It should be the time of the first beep sound but I need to retest to make sure.

Warudo, 2015-05-13:
It does align somewhere within the first beep, but it varies. Sometimes it's the beginning, sometimes it's in the middle. I start the blackbox video somewhere within the long beep, but then I have to seek around and look for sharp manouvers maneuvers like a roll, and then readjust the blackbox video otherwise it's always half a second off or so.  Whereas in 1.8.1 it always correctly matches to the beginning of the short beep. From my end user PoV, the beep is just too long now and is less accurate. Not a huge problem, but it is a regression.

thenickdude, 2015-05-18:
Ah yep, looks like the time of the sync beep isn't being recorded (it appears in my event log as time zero). It should be appearing on the rendered graph as a blue vertical align (like the red centreline) but it's absent. I'll get this fixed.

thenickdude, 2015-05-18:
Please try this version out and let me know how it goes:

https://github.com/sherlockflight/cleanflight-dev/releases/tag/blackbox-arm-beep-timing

It's based on the 1.9.0RC, so backup your configuration first before upgrading (it'll be erased).

hydra, 2015-05-20:
Could maybe add this fix to 1.9.0 stable if it comes soon.

Warudo, 2015-05-21:
Yep, that fixes it for me. There are some frames after the blue line appears but before the graphs start, which is the offset I was noticing.

PR #936 (2015-05-22): Fix Blackbox arming beep time logging
Closes #883
closed 2015-05-22 by hydra