cleanflight/cleanflight #884: New resource handling proposal

ledvinap, 2015-05-12:
@hydra: I'am playing with new IO,timer, EXTI (and ADC, etc) implementation. Can you quickly review it (local branch in my clone) and tell me your opinion?

It is work in progress, I would like to improve it in some respects .. 

The basic idea is:
- resources are described by `const  <resource>Def_t` and (optional)  runtime `<resourceRec>` structures.
 - `<>Def_t` structure contains all details necessary for resource allocation and access
 - `<>Rec_t` may not be unique (I'll see how it works out)
 - `<>Def_t` points to associated `<>Rec_t`
 - `<>Def_t *` pointer is all that is needed to identify resource (pin, exti, timer channel, timer, ADC channel)
 - `<>Rec_t`  is guaranteed to be unique for given resource (can be safely compared)
 - `<>Rec_t` contains runtime information, owner info etc
 - `<>Rec_t` or  `<>Def_t` will be probably collected by linker script into separate sub-section, so it can be enumerated (i am working on this)
- `<>Def_t` for some resources (maybe all) are defined in CPU specific file and accessible by name (linker symbol). 
 - you can use  `IO_PB8` to refer to pin PB8, it must only exist on used CPU.
 - records are separated into separate subsection, that way linker is smart enough to remove unused ones. Just use pin in code, it's definition gets linked ... 
- macro magic is used to define resources (quite a lot of magic is involved right now)
 - Using BOOST_PP may improve things, some external XML parser with template may be used if that fails
 - definitions should be usable even when implementation changes

This system may remove lot's of code duplication easily.
It will allow much easier configuration if required (set two pin indexes for sonar or softserial in config).

In medium-term perspective this may allow fully general resource assignment. Configurator may simply download available resources from board and create resource map to use. Onboard firmware will only check conflicts, but no collision-resolution logic will be necessary.

```
DEF_IO(GPIOA, PIN15);
DEF_IO(GPIOB, PIN0);
DEF_TIMER(TIM1, TIM1_CC_IRQn, TIM1_UP_IRQn, true);
DEF_TIMCH(GPIOB, PIN15, TIM15, TIMCH2); // AF multiplexer is missing now

    static const sonarHardware_t const sonarHardware = {
        .triggerIO = &IO_PB7,
        .echoIO = &IO_PB6,
    };

IOConfigGPIO(sonarHardware->echoIO, Mode_IN_FLOATING);

    EXTIHandlerInit(&hcsr04_extiCallbackRec, hcsr04_extiHandler);
    EXTIConfig(sonarHardware->echoIO, &hcsr04_extiCallbackRec, NVIC_PRIO_SONAR_EXTI, EXTI_Trigger_Rising_Falling); // TODO - priority!
    EXTIEnable(sonarHardware->echoIO, true);
```

JohnieBraaf, 2015-05-14:
Looks clean, but would it not be much better to define all the IO configuration in target.h file?

ledvinap, 2015-05-14:
@JohnieBraaf : I would like to reduce target.h redundancy - timer channel pin mapping, timer channel type, adc pin mapping, ... All these are only dependent on CPU, not target ...

ledvinap, 2015-05-14:
@JohnieBraaf : BTW - the ugly part is macro magic necessary ..

JohnieBraaf, 2015-05-14:
That makes sense ideed, so basically you want to map out all pins, and alternate functions to be able to use them more dynamically. With the increasing number of targets (sharing the same CPU) it would eliminate a lot of those conditional statements.

I guess it would be a first step in having a cleanflight-configurator based pin function selector.

hydra, 2015-05-14:
@ledvinap looking...

hydra, 2015-05-14:
Yeah @ledvinap anything we can do to cleanup the resource allocation would be an excellent addition and is required before we even think about F4 targets.

I'm aware that there is still the softserial improvement in PR #177 but i think this should be completed first, do you agree?

ledvinap, 2015-05-14:
#177 is already using similar interface (see my commit history, it was trivial to change it)

JohnieBraaf, 2015-05-18:
So what is the plan of attack here, and how can one help?

ledvinap, 2015-05-18:
Well .. I am coding it on the go, fixing design problem as they appear. I'll welcome any comments, ideas etc ...

I'll try to prepare some cosmetic PRs to remove unnecessary differences between my branch and master. 

When IO and timer stuff is stabilized, it is possible to start merging it. The changes should be quite small in first phase, then new features may be merged.

ledvinap, 2015-05-19:
@hydra: what do you thing about using BOOST_PP in cleanflight?http://www.boost.org/doc/libs/1_58_0/libs/preprocessor/doc/index.html

hydra, 2015-05-21:
Hmm..  So, thoughts.

1. I'm not a fan of pre-processor macros, they can hide the actual code and make debugging and fixing compilation errors much harder.
2. It is harder to work with pre-processor macros in the IDE - the IDE has to understand and parse not only the C source but also the pre-processor library.
3. It will likely confuse anyone not familiar with the library and perhaps may lead to less maintainable code.

A code generation step in the build is another option, many projects do this, i.e. take some definition and spit out a source file which can be compiled normally without any fancy tricks and is transparent when it comes to debugging and the IDE's will be able understand the code.

Perhaps using boost ONLY during the code generation step is a solution?

Any solution that doesn't require opaque pre-processor macros and/or a code generation step is preferable.

Thoughts?

hydra, 2015-05-21:
Pre-processor macros generate lots of WTFs.  This picture should be remembered at all times:

[image]

ProDrone, 2015-05-21:
```C
while (WTF/min) {
WTF--;
}
```

ledvinap, 2015-05-21:
Technically new system needs these structures correctly initialized and placed into sections (timers and IO now, ADC will be similar):
```
typedef struct ioDef_s {
    GPIO_TypeDef *gpio;
    uint32_t pin;
    struct ioRec_s *rec;
} ioDef_t;

DEF_IO(GPIOA, PIN0);
DEF_IO(GPIOA, PIN1);

typedef struct timerDef_s {
    struct timerRec_s* rec;
    TIM_TypeDef *tim;
    uint8_t irqCC, irqUP;
    uint8_t channels;        // number of channels allocated for this timer
    bool outputsNeedEnable;  // advanced timers
    rccPeriphTag_t rcc;
} timerDef_t;

DEF_TIMER(TIM1, TIM1_CC_IRQn,             TIM1_UP_TIM16_IRQn,      true,  RCC_APB2(TIM1));
DEF_TIMER(TIM2, TIM2_IRQn,                TIM2_IRQn,               false, RCC_APB1(TIM2));

typedef struct timerChDef_s {
    struct timerChRec_s *rec;
    const struct timerDef_s *timerDef;    // timer definition
    const struct ioDef_s *ioDef;          // io pin definition
    TIM_TypeDef *tim;                     // cache
    GPIO_TypeDef *gpio;                   // cache
    uint32_t pin;                         // cache
    uint8_t channel;
#ifdef STM32F303xC
    uint8_t pinAF;                        // pin Alternate Function multiplexer for this timer
#endif
} timerChDef_t;

    DEF_TIMCH(GPIOB, PIN15, TIM15, TIMCH2, GPIO_AF_1), // PWM1  - PB15 - TIM1_CH3N, TIM15_CH1N, *TIM15_CH2
    DEF_TIMCH(GPIOB, PIN14, TIM15, TIMCH1, GPIO_AF_1), // PWM2  - PB14 - TIM1_CH2N, *TIM15_CH1
const timerChDef_t timerQueueChannelDef = 
    DEF_TIMCH(NA, NA,  TIM2,  TIMCH2, GPIO_AF_1); // PWM10 - PA1  - *TIM2_CH2, TIM15_CH1N

```
Current macro implementation is intermixed with structs ... 

The macro approach has one big advantage - implementation details are separated from processor details. One can be easily changed without touching the other. Typical change will be caching more (or less) fields in timerChDef_s. If processor is defined directly, it will take enormous edits (number of targets x number of channels per target)

In current state it's mostly just pasting tokens together and then expands resulting macro:
```
# define DEF_TIMCH(gpio_, pin_, tim_, tim_ch_, pin_af_) {                \
        .rec = DEFIO_TIMERCH_REC(DEFIO_TIM_ID__  ## tim_, DEFIO_TIMCH_ID__ ## tim_ch_), \
        .timerDef = DEFIO_TIMER_DEF(DEFIO_TIM_ID__ ## tim_),            \
        .ioDef = DEFIO_IO_DEF(DEFIO_GPIO_ID__ ## gpio_, DEFIO_PIN_ID__ ## pin_), \
        .tim = DEFIO_TIM(DEFIO_TIM_ID__ ## tim_),                       \
        .gpio = DEFIO_GPIO(DEFIO_GPIO_ID__ ## gpio_),                   \
        .pin = DEFIO_PIN(DEFIO_PIN_ID__ ## pin_),                       \
        .channel = DEFIO_TIM_CHANNEL(DEFIO_TIMCH_ID__ ## tim_ch_),      \
        .pinAF = pin_af_                                                \
    }                                                                   \
/**/
```
It's not that pretty, but to possible to follow. And you can't supply PIN instead of TIMER, processor will catch it.

Problem arises when I want TIMER without IO pin. With Boost_PP I just use:
```
#define DEFIO_GPIO_ID__NA    0
#define DEFIO_GPIO_ID__A      1
#define DEFIO_IO_DEF_NAME(gpio, pin) CONCAT(IO_P, CONCAT(DEFIO_GPIO_LETTER(gpio), pin))
#define DEFIO_IO_DEF(gpio, pin) BOOST_PP_IF(gpio, &DEFIO_IO_DEF_NAME(gpio, pin), NULL)
```

Without it more magic will be necessary. Or define two macros - one for channel with PIN, one without.

ledvinap, 2015-05-21:
Also with BOOST_PP it is possible to define target/cpu/etc as list and then use construction like 
```
DEFIO_DECLARE(CPU_GPIO);  // declaration of PIN variables

...

DEFIO_DEFINE(CPU_GPIO);  // emit initialized structure definitions
```

leothehuman, 2015-09-10:
Is this a WIP or MIA? (:

PProvost, 2015-09-23:
Since this is blocking Quanton, Sparky2 and BrainFPV support, it would be good to understand status and intent for this plan. Thanks!

ledvinap, 2015-09-23:
I am working on it now. I did quite deep redesign, moving most necessary information into _Rec and using _Def mostly only on startup to initialize Rec.
Implementation for IO and EXTI should be ready for review quite soon.
IO definition is now based on BOOST PP library, it does considerably reduce code repetition, but some other method may be used in final version (Personally I quite like BOOST_PP ... it is fun to work with, but it removes lots of code repetition)

ledvinap, 2015-10-17:
IO redesign is working quite nice in my local branch.
IO interface is here: https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/io.h
Some semi-internal parts will need some polishing:
https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/io_impl.h
https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/io_def.h

Is is build around automatically generated file that defines all available pins on target. The file structure is quite simple (and surprisingly powerful):
https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/io_def_generated.h
https://github.com/ledvinap/cleanflight/blob/local/src/utils/def_generated.pl

Target definition is much cleaner: https://github.com/ledvinap/cleanflight/blob/local/src/main/target/SPARKY/target.h

In near furure I would like to replace `ioRec_t* ` with `uint32_t IO_t`, packing GPIO structure address and pin index into 32 bits (GPIOs are 4k apart, so there is lot of space in low bits). This should improve speed (mask+shift is faster than load) and reduce memory usage. Interface stays the same, so it is possible to replace internals without changing anything else ... 

@hydra: What do you think?

BTW: 
```
Entering CLI Mode, type 'exit' to return, or 'help'

# resources
IO:
A00:                FREE 00
A01:          PWOUT_FAST 06
A02:           SERIAL_TX 16
A03:           SERIAL_RX 10
A04:                FREE 00
A05:                FREE 00
A06:              SYSTEM 02
A07:                FREE 00
A08:          PWOUT_FAST 06
A09:                FREE 00
A10:                FREE 00
A11:                FREE 00
A12:                FREE 00
A13:                FREE 00
A14:                FREE 00
A15:                FREE 00
B00:          PWOUT_FAST 06
B01:          PWOUT_FAST 06
B02:                FREE 00
B03:                FREE 00
B04:              SYSTEM 02
B05:              SYSTEM 02
B06:           SERIAL_TX 10
B07:           SERIAL_RX 10
B08:                FREE 00
B09:                FREE 00
B10:         SERIAL_RXTX 10
B11:                FREE 00
B12:                FREE 00
B13:                FREE 00
B14:          PWOUT_FAST 06
B15:          PWOUT_FAST 06
F00:                FREE 00
F01:                FREE 00
F06:                FREE 00
TIMER:
0 : time: 0 prio: 0xd0
 - 0: 00000000 00000000 A08:6          PWOUT_FAST
 - 1: 00000000 00000000 ---:-                FREE
 - 2: 00000000 00000000 ---:-                FREE
 - 3: 00000000 00000000 ---:-                FREE
1 : time: 0 prio: 0x10
 - 0: 00000000 00000000 ---:-                FREE
 - 1: 20003560 00000000 ---:-               TIMER
 - 2: 00000000 00000000 A02:1          PWOUT_FAST
 - 3: 00000000 00000000 ---:-                FREE
2 : time: 0 prio: 0xd0
 - 0: 00000000 00000000 ---:-                FREE
 - 1: 00000000 00000000 ---:-                FREE
 - 2: 00000000 00000000 B00:2          PWOUT_FAST
 - 3: 00000000 00000000 B01:2          PWOUT_FAST
3 : time: 0 prio: 0xff
 - 0: 00000000 00000000 ---:-                FREE
 - 1: 00000000 00000000 ---:-                FREE
 - 2: 00000000 00000000 ---:-                FREE
 - 3: 00000000 00000000 ---:-                FREE
4 : time: 0 prio: 0xff
5 : time: 0 prio: 0xff
6 : time: 0 prio: 0xff
 - 0: 00000000 00000000 ---:-                FREE
 - 1: 00000000 00000000 ---:-                FREE
 - 2: 00000000 00000000 ---:-                FREE
 - 3: 00000000 00000000 ---:-                FREE
7 : time: 0 prio: 0xd0
 - 0: 00000000 00000000 B14:1          PWOUT_FAST
 - 1: 00000000 00000000 B15:1          PWOUT_FAST
8 : time: 0 prio: 0xff
 - 0: 00000000 00000000 ---:-                FREE
9 : time: 0 prio: 0xff
 - 0: 00000000 00000000 ---:-                FREE
```

hydra, 2015-10-19:
@ledvinap I need to dedicate some time to taking a look.  I'm happy you're making good progress with this!

hydra, 2015-10-20:
@ledvinap Mmmm, target file so clean. I like that.

hydra, 2015-10-20:
@ledvinap What additional software will developers need to install to automatically generate the files?

ledvinap, 2015-10-20:
The script i written in Perl, no other dependencies. Generated file may be committed into repository, there is no reason to regenerate it unless io internals are changed.

ledvinap, 2015-10-20:
Pushed new version to my local branch.
I'll start working on merge into master (IO part only).
Any ideas before I start? Refactoring, ideas, etc?

hydra, 2015-10-21:
no, not had time to look, won't get time until this weekend.
the sbus / failsafe stuff is taking priority, then this.

hydra, 2016-01-03:
ok @ledvinap we need this stuff ASAP.  Certainly a more declarative approach is preferred for target definitions.  What needs to happen to get this finished?

universam1, 2016-02-01:
Could be solving #1699 as well!?

closed 2017-02-25 by hydra