cleanflight/cleanflight #885: Support for Sparky 2

VirtualEnder, 2015-05-12:
Please, oh please! The I would REALLY love to see support for this board! At some point... If possible!

http://www.hobbiesfly.com/multi-rotors/flight-controller/taulabs-sparky2-0.html
[image]

readerror67, 2015-05-12:
lul

JohnieBraaf, 2015-05-12:
I think getting cleanflight to compile on a Discovery F4 would be a good starter.

sambas, 2015-05-13:
I've been playing with F4 targets in https://github.com/sambas/cleanflight, my own AnyFC, revo and colibri works at flyable level.

JohnieBraaf, 2015-05-13:
Good stuff, together with this repo https://github.com/treymarc/baseflight it should be possible to draft a  F4 PR.

hydra, 2015-05-14:
F4 support wont happen until hardware resource allocation is cleaned up.  Please help by commenting on and testing #884

brat002, 2015-11-16:
I am interested this feature too. TL for me is not interesting more.

VirtualEnder, 2015-11-16:
@brat002 Work is being done over here: http://www.rcgroups.com/forums/showthread.php?t=2533601

It's already flyable on the Revo, and Spark 2 isn't far behind! Hooray!

closed 2017-02-25 by hydra