cleanflight/cleanflight #887: Implement ETH Zurich safety algorithm

alexgorbatchev, 2015-05-12:
Search didn't show this was suggested earlier. I believe it would be great for Cleanflight and community overall to have this implemented. Perhaps a small Kickstarter campaign to sponsor this feature?

> — This paper presents periodic solutions for a
quadrocopter maintaining a height around a position in space
despite having lost a single, two opposing, or three propellers.
In each case the control strategy consists of the quadrocopter
spinning about a primary axis, fixed with respect to the vehicle,
and tilting this axis for translational control. A linear, timeinvariant
description of deviations from the attitude equilibrium
is derived, allowing for a convenient cascaded control design.
The results for the cases of losing one and two propellers
are validated in experiment, while the case of losing three
propellers is validated in a nonlinear simulation. These results
have application in multicopter fault-tolerant control design,
and also point to possible design directions for novel flying
vehicles.

References:

* [New algorithm makes quadrocopters safer](https://www.ethz.ch/en/news-and-events/eth-news/news/2013/12/new-algorithm-makes-quadrocopters-safer.html)
* [White paper](http://flyingmachinearena.org/wp-content/publications/2014/mueIEEE14.pdf)

https://www.youtube.com/watch?v=bsHryqnvyYA

orgua, 2015-05-12:
Yeah, that feature would be nice. BUT - this video was done by using motion capturing. Yes, it is possible even without - but you would need feedback from the motor or driver OR an internal state observer that detects the failing motor. If that wasn't hard enough to accomplish you would need more computationpower than the current cf-boards provide. The Paper states control loops at 1 KHz....

alexgorbatchev, 2015-05-12:
Thanks for the thoughtful response. Sounds like at this time this isn't feasible.

closed 2015-05-12 by alexgorbatchev