cleanflight/cleanflight #894: Calibrate Mag stick command doesn't work

Tinkyy, 2015-05-14:
I cannot get drone to Calibrate Mag/Compass using stick positions. Drone is disarmed and level, other commands like calibrate gyro and acc are working and drone beeps after calibration. I can provide CLI dump if nedded. 

Also what is difference between calibrate accelerometer and gyro, are they the same? Why is only cal acc in the cleanflight GUI ?

thenickdude, 2015-05-15:
The gyros calibrate themselves on every powerup, so there's no need for a distinct command for that (this is why you're not supposed to move the craft after connecting the battery, while it blinks the LED). The accelerometer normally only needs to be calibrated once, manually, from the Configurator. 

On the Receiver tab, check that all your sticks can reach the entire 1000 to 2000 range correctly. If they fall significantly short then they will not be able to reach the `min_check` or `max_check` thresholds required for stick commands.

When you calibrate the mag you have 30 seconds to rotate the craft completely on every axis. The configurator has a button for this and guides you through the process.

hydra, 2015-05-15:
Closing this issue, please re-open if there's still an issue after you have checked your configuration.

hydra, 2015-05-15:
@Tinkyy Please also see https://github.com/cleanflight/cleanflight/blob/master/CONTRIBUTING.md

closed 2015-05-15 by hydra

Tinkyy, 2015-05-15:
Ok, it works , but 30 second calibration time is what confused me. There is no indication that command was accepted, only beep after 30 seconds. If i calibrate Acc using stick command, LED's flash green and there is a beep.
Maybe it should be wise to add LED indication during calibration period, and also Beep indication, to get some sort of conformation that you entered calibrate mode.

hydra, 2015-05-15:
@Tinkyy During mag calibration one of the leds flashes while calibration is in progress.

https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/compass.c#L95

If you're not seeing this then please open a new issue and reference this one.

Tinkyy, 2015-05-15:
@hydra 
Nope no LED flashing. Only one beep after 30 sec. What is more strange Drone has all functions working during calibration, I can ARM the motors during calibration and start flying! This does not seem safe!
Maybe it should be smart to add a beep at the start of calibration.

Here is DUMP
https://www.dropbox.com/s/nh58i9dfsqby3ow/DUMP.txt?dl=0