cleanflight/cleanflight #902: Althold_fast_change

Fvk1960, 2015-05-15:
When using althold with Alt_hold_fast_change the moment I change stick to go up the copter goes down first before going up.
From the code I analyse: When throttle stick is in hoovering postion so no up/down command the rcCommand throttle is initial throttle + correction from pid. In case the correction for example is 
+200 and then change the throttle stick (move just above initial throttle by deadband) the throttle output changes instantly  from initial + 200 to initial + 2*deadband (or otherwise said: move just above initial throttle by deadband+deadband) which is actually a lower throttle output assuming a deadband of 50.
At point of code rcCommand[THROTTLE] is still unmodified rcData[THROTTLE] (at least not modified by ALT/VEL pid 
In Short what if altHoldThrottleAdjustment > 2*alt_hold_deadband at moment you want to ascent ?

    if (rcControlsConfig->alt_hold_fast_change) {
        // rapid alt changes
        if (ABS(rcData[THROTTLE] - initialThrottleHold) > rcControlsConfig->alt_hold_deadband) {
            errorVelocityI = 0;
            isAltHoldChanged = 1;
            rcCommand[THROTTLE] += (rcData[THROTTLE] > initialThrottleHold) ? -rcControlsConfig->alt_hold_deadband : rcControlsConfig->alt_hold_deadband;
        } else {
            if (isAltHoldChanged) {
                AltHold = EstAlt;
                isAltHoldChanged = 0;
            }
            rcCommand[THROTTLE] = constrain(initialThrottleHold + altHoldThrottleAdjustment, escAndServoConfig->minthrottle, escAndServoConfig->maxthrottle);
        }
    }
Do I see this correctly ?

ledvinap, 2015-05-16:
```
rcCommand[THROTTLE] += (rcData[THROTTLE] > initialThrottleHold) ? -rcControlsConfig->alt_hold_deadband : rcControlsConfig->alt_hold_deadband;
```
This actually removes deadband, so you get initialThrottleHold. 
This is probably by design - throttle command is almost the same as without althold mode (minus deadband) outside of althold range. Just enter althold at hovering throttle or do not use alt_hold_fast_change ...

Fvk1960, 2015-05-16:
Initially you use it in that manner. But after hoovering some time the I factor increases over time and battery dropping etc. Meaning I factor gets to a 200 value at some point. At this point changing altitude raises the problem of height drop while wanting to incline.
By the way I am missing a constrain in this part of the code.
But what if preceding the action with the modified RcCommand[THROTTLE] by althold and next do the addition?
rcCommand[THROTTLE] = constrain(initialThrottleHold + altHoldThrottleAdjustment, escAndServoConfig->minthrottle, escAndServoConfig->maxthrottle);
rcCommand[THROTTLE] += (rcData[THROTTLE] > initialThrottleHold) ? -rcControlsConfig->alt_hold_deadband : rcControlsConfig->alt_hold_deadband;
Still mssing the constrain here.
rcCommand[THROTTLE] = constrain(rcCommand[THROTTLE], escAndServoConfig->minthrottle, escAndServoConfig->maxthrottle);

ledvinap, 2015-05-17:
Whole throttle range will be shifted by `altHoldThrottleAdjustment`. for example, what will happen if you leave althold during fast forward flight? This may even result in minimum throttle being to high for hover ...

Fixing battery voltage dependency later in the control loop may great way to help with this problem (and will improve flight behavior and tuning in general)

Fvk1960, 2015-05-17:
In that case this part of software is no longer processed and rcCommand[THROTTLE] will follow rcData[THROTTLE] adjusted with any throttle curve.
Remember in althold throllestick is no longer a throttle stick but a vertical velocity stick and throttle range is dominated by initialthrottlehold dominated by whole copter layout weight,motor kv etc.
Leaving altholdmode while fast flying forward will rather set you back around initial throttlehold.
==========================================================================
But just for an idea I wrote in multiwii a code with different approach. Baromode can be on from start like in naze but f.baro.mode will only be set to true after arm+baromode+estalt>50cm. this will set initial throttlehold correctly (including limit for z-axis would be nice too truly recognize hoover) next the altholdcontrol is taken against MIDRC(1500) giving you the full stick range for up/dwn. When alt is < 50cm f.baro.mode is turned of after small timer. So you can have althold/baromode ON on forhand selected. Take off slow untill 50cm althold kicks in. Now just fly.
Land and althold disables so you can arm/disarm normally. 
Also my walkera RC transmiiter has auto center throttlestick like naze transmitter.
I can preselect MODE Acro, Attitude(Angle/Horizon) or Altitude before arming. ARM FLY LAND DISARM.
Just idea to think about for future design.

closed 2015-05-30 by Fvk1960