cleanflight/cleanflight #903: Sonar Baro Mode selection not working as aspected

Fvk1960, 2015-05-16:
It seems that sonar is always computed in the Baroalt value dispite the fact sonar is not selected in the mode setup.
I have a sonar connected.
In mode setup I have a 3 position aux switch to select Baro or Sonar or Neither.
When I select Baro Mode only Sonar is still used in the estalt determination when sonar is in range.
I expected from the Mode setup option that Sonar and Baro would be exclusieve.
I can imagine that only sonar would be a problem when the sonar gets out of range and baro should back it up. But when only BaroMode is selected I expect the Sonar being excluded.
This is usefull when terrain is not so good for clear sonar signal.
Is my finding correct ?

digitalentity, 2015-05-16:
Yes, if SONAR feature is enabled, sonar is always used in altitude calculations. I am working on new navigation (see #197) that will include reworked altitude hold code using all available altitude sources to maintail stable position), however `nav-rewrite` still have to go a long way before it will be flyable.

Fvk1960, 2015-05-16:
I fixed it rather simply with:
if (FLIGHT_MODE(SONAR_MODE)) {
    if (sonarAlt > 0 && sonarAlt < 200) {
        baroAlt_offset = BaroAlt - sonarAlt;
        BaroAlt = sonarAlt;
    } else {
        BaroAlt -= baroAlt_offset;
        if (sonarAlt > 0) {
            sonarTransition = (300 - sonarAlt) / 100.0f;
            BaroAlt = sonarAlt * sonarTransition + BaroAlt * (1.0f - sonarTransition);
        }
    }
}

closed 2015-05-17 by Fvk1960