cleanflight/cleanflight #906: 1.9.0 Prerelease issue

rscarawa, 2015-05-17:
I am unable to load this on my Naze.  It give the error "Communication with bootloader failed".  I have the 63.2 configurator as well.

rscarawa, 2015-05-17:
selected Full Chip Erase and things seem to be moving now.

rscarawa, 2015-05-17:
Just completed a 2 second flight.  Throttle works, as does yaw and pitch.  Zero yaw control.  WHen it lifts off the ground, it rotates to the left at about 1 turn per 8 seconds.

rscarawa, 2015-05-17:
# set yaw
align_board_yaw = 0
yaw_deadband = 6
yaw_control_direction = 1
yaw_direction = 0
yaw_jump_prevention_limit = 200
rc_yaw_expo = 0
yaw_rate = 50
p_yaw = 85
i_yaw = 45
d_yaw = 0
p_yawf = 8.000
i_yawf = 0.500
d_yawf = 0.050
yaw_p_limit = 0

rscarawa, 2015-05-17:
https://www.dropbox.com/s/w9o24jp1ir5n8dg/blackbox_log_2015-05-17_184827.TXT?dl=0

link to short black box.

hydra, 2015-05-17:
yaw defaults are:

```
# get yaw
align_board_yaw = 0
yaw_deadband = 0
yaw_control_direction = 1
yaw_direction = 1
yaw_jump_prevention_limit = 200
rc_yaw_expo = 0
yaw_rate = 0
p_yaw = 85
i_yaw = 45
d_yaw = 0
p_yawf = 8.000
i_yawf = 0.500
d_yawf = 0.050
yaw_p_limit = 0
```

hydra, 2015-05-17:
@rscarawa a full config dump might be better

rscarawa, 2015-05-18:
Hydra
See if this link works for you.  It should have the blackbox logs, cli dump and backup I took just a moment ago.

https://www.dropbox.com/home/rc/cleanflight/1.9.0/prerelease

hydra, 2015-05-18:
Can't access that link, can you use a different service to upload the files to?

JohnOCFII, 2015-05-18:
@rscarawa That DropBox link isn't working.  Did you not mark the files public?

rscarawa, 2015-05-18:
https://www.dropbox.com/sh/p5yo99t6z7g6rjj/AACLolr6pAkCGEKGfMtHmARUa?dl=0

See if this works better.

rscarawa, 2015-05-18:
https://www.dropbox.com/sh/weuw26ynl1l48ew/AAAGP-tpYshFmhSfi6mZNbiSa?dl=0

I his is better

JohnOCFII, 2015-05-18:
@rscarawa - That link works for me.

rscarawa, 2015-05-18:
Look in pre-release folder.

thenickdude, 2015-05-18:
`yaw_direction` of 0 causes yaw term to be zero in the mixer:

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L579

`yaw_direction` has to be 1 or -1.

rscarawa, 2015-05-18:
That old value worked before the upgrade.  I will make the change and try again tomorrow.  Any idea why the left yaw with no input?

thenickdude, 2015-05-18:
That code has been untouched since Dec 2014, `yaw_direction=0` has always caused no yaw control as far as I know. Are you sure that you were running `yaw_direction=0` previously? (do you have an old CLI dump?)

rscarawa, 2015-05-18:
I have the old cli dump.  WIll have to check that tomorrow.  I used it to do a copy paste into the new prerelease code.  

I gave the yaw_control=1 setting a try and the yaw drift went away and it yaws the proper direction.  Now I have the over rotation problem back.  I see that there is a yaw jump control variable and it is set to 200.  I assume this is new.  Will setting this to a lower number help with the over rotation problem?  Given that  the patched code fixed my problem, what do you advise I set this to?  SHould I adjust any other variables  to deal with the over rotation?

rscarawa, 2015-05-18:
if I read the comments in the code right, the closer to 500 the yaw jump correction is, the less it impacts yaw?

thenickdude, 2015-05-18:
The old limit on 1.8.0 was fixed at 100, the new default of 200 should allow for twice as strong of a yaw response, which allows the yaw to brake to a stop quicker. Try increasing it to 400 or something.

hydra, 2015-05-18:
Perhaps, checking yaw_direction==0 in validateAndFixConfig() might be useful.

Or maybe decoupling yaw_direction internal implementation details from the CLI command would be better?  e.g.  use 0 and 1 for the cli command and have the yaw implementation do what is right in the mixer.  that way it would be impossible to set an invalid value and then no code would be needed to fix invalid values.

rscarawa, 2015-05-18:
I have uploaded my old CLI and backup to the same folder.  These files predate the upgrade.

Changing yaw_jump_prevention_limit=400 did help alot with the over rotation.  Thank you.

rscarawa, 2015-05-18:
What is the difference between:

yaw_control_direction = 1
yaw_direction = 1
?

thenickdude, 2015-05-18:
Setting `yaw_control_direction` to -1 inverts your yaw stick input, this is useful if your transmitter is weird and doesn't offer the ability to flip this itself.

Setting `yaw_direction` to -1 tells Cleanflight that all your motors spin the opposite way from what the manual recommends. So it tells Cleanflight to use the opposite motors to affect a yaw that it normally would.

@hydra IMO using -1 and 1 is a more intuitive UI for describing flipping those control axes, I would prefer that `validateAndFixConfig()` approach.

rscarawa, 2015-05-18:
One new behavior I am seeing is a pitching or yawing movement when I do  hard 360 move to the left.  Seems to happen 50% of the time.  If in rate mode starting out level and I hit the rudder hard left, I can see a corner drop.  After the 360 is done, the copter drifts hard to that side.  That is hands completely off of the pitch/roll stick.  I need to get a log of that.

rscarawa, 2015-05-18:
Uploaded blackbox log of roll pitch issue when doing hard left pirouette.  It came off axis hard on the last one.

https://www.dropbox.com/s/7sw0oqaiwf6z3y9/roll_pitch_issue_blackbox_log_2015-05-18_184506.TXT?dl=0

ledvinap, 2015-05-19:
@rscarawa 
Main problem is probably output saturation - on maximum yaw command the motors are at maximum/minimum and roll/pitch command does not have any effect. Limiting YAW output will improve this (and decrease yaw responsiveness)

Also `PID_I[yaw]` seems to stay quite high after releasing stick and is compensated by P term ... Maybe the integration time constant is too high? @thenickdude ?

rscarawa, 2015-05-19:
In my test, I was in rate mode.  How come the quad did not return itself to the angle it was at when I centered the yaw?

borisbstyle, 2015-05-19:
@rscarawa 
Your motors work at max power during yaw stops. Your setup isn't able to cope with such a high yaw_jump_prevention_limit of 400 what you are using now.
Try bringing that down to default (200) and check again. If you still have the issue you can go even lower.

rscarawa, 2015-05-19:
Still not making sense.  It does not happen every time.  second, if I give zero pitch roll input and it goes off axis during yaw, why does it not correct itself when yaw is stopped?  Should the gyro not realize there is an error that needs correcting?

rscarawa, 2015-05-19:
Also, my yaw rate is only 50.  Seems I should not be hitting my motor limits at this setting.  This is the same setting I had preupgrade and I do not recall every having this issue.  Actual yaw rotation rate does not seem to have changed between versions.

borisbstyle, 2015-05-19:
blacbox still shows full power on motors during yaw on full stick input! So definately hitting the limit there with this setup. Check 0:57sec in the log file during full stick input and during motors again full power during yaw stop. 

You are asking more from your motors than those can deliver. Lower the jump limit to give more space for pitch and roll corrections. I think you should try default setup of 200 instead of 400 you are using now.

The old default hardcoded settings was 100 by the way. If you change it back to 100 you will have exact same behaviour as on previous firmware.

borisbstyle, 2015-05-19:
Full stick yaw:
[image]

Yaw stop:
[image]

rscarawa, 2015-05-19:
When set to 200, I had the over rotation problem.  Of 400 is a problem, then why does thin not happen every time?  When it happens, why is it not corrected when yaw is released?

borisbstyle, 2015-05-19:
When I read back in this issue I see you had different problems down the road after restoring backup. You also have been changing this value before you changed some other things and restored again. I do suggest to start all over with the correct default values and see if the drift issue is there again.

See above "yaw stop" screenshot where motors are using full power to stop. What also happens there is that you give some pitch input right during yaw stop. Why are you pitching up there during the yaw stop? In the screenshot isn't much, but during the stop pitch axis goes further to 155!

In case you want to test exact same yaw behaviour like in previous release set the yaw_jump_prevention_limit to 100. That's basically the only thing what changed as that was the hardcored value.

rscarawa, 2015-05-19:
I gave the pitch input to stop the hard drift that was introduced during the yaw.  The pitch problem didnot come in at the end of the yaw from what I remember.  It happened near the beginning.

I will go back and reset the yaw jump to 200 and post a log.  If the over rotation goes away, then good.  If not, I will set it back to 400, and redo the test.  When the pitch or yaw issue comes in, I will leave the quad in position for a second and then correct it to show how the gyro is not seeing an error when off axis.  Expect another log later today.

Thank you for your time.

rscarawa, 2015-05-19:
Performed a few more pirouettes with yaw_jump_prevention_limit to 200.  When I do full 360, it takes more than 90 degrees to stop but it does rotate back slowly to where I let go of the stick.  After doing about 4 or 5 of them, it did go off axis on the last one.  See the link below.  I could not leave the tilt in there since I am in a confined area.  You will see the correction I made.  Seemed it went off either roll or pitch axis after 90 degrees of rotation from what I could see.

https://www.dropbox.com/s/fkiuh2dfxfkc7b9/yaw_jump_200_blackbox_log_2015-05-19_171349.TXT?dl=0.

rscarawa, 2015-05-23:
I did some more testing today and became suspicious that the the props might not be in balance.  When I tried to hover in angle mode, I had to put alot of fore pressure on the stick to keep the quad from backing up.  I pulled the props to balance them and retried the test.  After a so-so prop balance, my pitch trim totally changed.  I did more 360s and the quad seems more resestant to falling off axis.  Out of about 10 360 turns, it fell off once and it was not as bad as it had been previously.  I probably need to find a better method of measuing their balance.  Even magnetic balancers do not appear to do the job well enough.

Are there gyro filter settings that help account for small motor or prop imbalances?

RClabTV, 2015-09-09:
Hi everybody, just wanted to report that I had the same problem after updating to 1.9.0 and restoring my Backup over the CLI commands.

No Yaw Control because of yaw_direction = 0
After setting yaw_direction = 1 problem was fixed...

closed 2017-02-25 by hydra