cleanflight/cleanflight #909: GPS incorrect time

kaefert, 2015-05-18:
Hi there!

I've connected a uBlox NEO M8N to the UART2 serial of my Naze32, and a Minim OSD to UART1.

I've configured the Minim OSD to display GPS Time alongside the GPS coordinates, though when GPS lock is aquired I only get the coordinates displayed, not the time.

I've also enabled Blackbox datalogging to the internal dataflash and when I download it through the configurator and use the blackbox_decode utility to get csv and gpx files, the gpx files contain timestamps like:
<time>2000-01-01T00:00:44.362081Z</time>

Does somebody know why this happens? Is my GPS Module at fault, or is this a problem somewhere in the Cleanflight code?

tracernz, 2015-05-18:
Neither UBlox (https://github.com/cleanflight/cleanflight/blob/master/src/main/io/gps.c#L874) nor NMEA (https://github.com/cleanflight/cleanflight/blob/master/src/main/io/gps.c#L562) parsers actually use the time information for anything.

kaefert, 2015-05-18:
okey. is it that way because we don't have the resources (processing or memory?) to deal with it, or is it because no developer has invested the time to code that yet?

tracernz, 2015-05-18:
Probably the latter.The official MSP protocol doesn't implement any fields for time (http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol) but MW OSD can read it if supported so I assume some other firmware does use a non-standard MSP_COMP_GPS packet format https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Serial.ino#L220, https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Config.h#L40. Cleanflight doesn't support it https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_msp.c#L1047.

kaefert, 2015-05-18:
thanks for all that info!
so as I understand it the scarab-osd code expects the time to be delivered as additional 32 bytes after the 8bit "GPS-update".
But for that to work we would first need to read and store the time from the gps interface in a variable in the cleanflight code.

For my needs the more important part would be to have the time stored in the blackbox logger so that the gpx files produced from that will have correct timestamps.

hydra, 2015-05-18:
Yes, i have thought for a while it would be good to use the time information from GPS.

It could be useful for at least:
1. Recording in blackbox logs, so you can match it up with your video files.
2. Displaying on OSD screen.
3. Displaying on field equipment, e.g. android phone, ground station
4. Storing in EEPRom when config is saved - this is less useful however, since time might not be available, e.g. when GPS RX is off.

Additionally I think allowing the time to be set via MSP or Telemetry systems could also be useful.

JohnOCFII, 2015-05-18:
I agree the time would be useful in the Blackbox logs as well on the OSD.

closed 2017-02-25 by hydra